zsmartsystems/com.zsmartsystems.zigbee #49: NPE in Karaf console (ZigBeeNetworkMeshMonitor.java)

openhab-5iver, 2017-07-12:
Originally posted [here](https://community.openhab.org/t/zigbee-binding/15763/231)

This NPE comes up in the Karaf console running OH2 and the Zigbee binding (2.2.0.201707042200).

```
openhab> Exception in thread "pool-35-thread-1" java.lang.NullPointerException
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor.getAssociatedNodes(ZigBeeNetworkMeshMonitor.java:305)
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor.access$500(ZigBeeNetworkMeshMonitor.java:59)
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor$2.run(ZigBeeNetworkMeshMonitor.java:233)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
        at java.lang.Thread.run(Thread.java:748)
Exception in thread "pool-35-thread-2" java.lang.NullPointerException
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor.getAssociatedNodes(ZigBeeNetworkMeshMonitor.java:305)
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor.access$500(ZigBeeNetworkMeshMonitor.java:59)
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor$2.run(ZigBeeNetworkMeshMonitor.java:233)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
        at java.lang.Thread.run(Thread.java:748)
Exception in thread "pool-35-thread-4" java.lang.NullPointerException
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor.getAssociatedNodes(ZigBeeNetworkMeshMonitor.java:305)
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor.access$500(ZigBeeNetworkMeshMonitor.java:59)
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor$2.run(ZigBeeNetworkMeshMonitor.java:233)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
        at java.lang.Thread.run(Thread.java:748)
Exception in thread "pool-35-thread-3" java.lang.NullPointerException
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor.getAssociatedNodes(ZigBeeNetworkMeshMonitor.java:305)
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor.access$500(ZigBeeNetworkMeshMonitor.java:59)
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor$2.run(ZigBeeNetworkMeshMonitor.java:233)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
        at java.lang.Thread.run(Thread.java:748)
Exception in thread "pool-35-thread-6" java.lang.NullPointerException
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor.getAssociatedNodes(ZigBeeNetworkMeshMonitor.java:305)
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor.access$500(ZigBeeNetworkMeshMonitor.java:59)
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor$2.run(ZigBeeNetworkMeshMonitor.java:233)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
        at java.lang.Thread.run(Thread.java:748)
Exception in thread "pool-35-thread-5" java.lang.NullPointerException
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor.getAssociatedNodes(ZigBeeNetworkMeshMonitor.java:305)
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor.access$500(ZigBeeNetworkMeshMonitor.java:59)
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor$2.run(ZigBeeNetworkMeshMonitor.java:233)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
        at java.lang.Thread.run(Thread.java:748)
Exception in thread "pool-35-thread-7" java.lang.NullPointerException
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor.getAssociatedNodes(ZigBeeNetworkMeshMonitor.java:305)
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor.access$500(ZigBeeNetworkMeshMonitor.java:59)
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor$2.run(ZigBeeNetworkMeshMonitor.java:233)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
        at java.lang.Thread.run(Thread.java:748)
Exception in thread "pool-35-thread-8" java.lang.NullPointerException
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor.getAssociatedNodes(ZigBeeNetworkMeshMonitor.java:305)
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor.access$500(ZigBeeNetworkMeshMonitor.java:59)
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor$2.run(ZigBeeNetworkMeshMonitor.java:233)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
        at java.lang.Thread.run(Thread.java:748)
Exception in thread "pool-35-thread-10" java.lang.NullPointerException
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor.getAssociatedNodes(ZigBeeNetworkMeshMonitor.java:305)
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor.access$500(ZigBeeNetworkMeshMonitor.java:59)
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor$2.run(ZigBeeNetworkMeshMonitor.java:233)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
        at java.lang.Thread.run(Thread.java:748)
Exception in thread "pool-35-thread-11" java.lang.NullPointerException
        at com.zsmartsystems.zigbee.internal.ZigBeeNetworkMeshMonitor.getAssociatedNodes(ZigBeeNetworkMeshMonitor.java:305)
[... 24 lines cut ...]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
        at java.lang.Thread.run(Thread.java:748)
Exception in thread "pool-35-thread-15" java.lang.NullPointerException
Exception in thread "pool-35-thread-16" java.lang.NullPointerException
Exception in thread "pool-35-thread-14" java.lang.NullPointerException
Exception in thread "pool-35-thread-18" java.lang.NullPointerException
Exception in thread "pool-35-thread-17" java.lang.NullPointerException
Exception in thread "pool-35-thread-19" java.lang.NullPointerException
Exception in thread "pool-35-thread-21" java.lang.NullPointerException
Exception in thread "pool-35-thread-20" java.lang.NullPointerException
Exception in thread "pool-35-thread-22" java.lang.NullPointerException
Exception in thread "pool-35-thread-24" java.lang.NullPointerException
Exception in thread "pool-35-thread-25" java.lang.NullPointerException
Exception in thread "pool-35-thread-26" java.lang.NullPointerException
Exception in thread "pool-35-thread-27" java.lang.NullPointerException
Exception in thread "pool-35-thread-28" java.lang.NullPointerException
Exception in thread "pool-35-thread-29" java.lang.NullPointerException
Exception in thread "pool-35-thread-23" java.lang.NullPointerException
Exception in thread "pool-35-thread-31" java.lang.NullPointerException
Exception in thread "pool-35-thread-30" java.lang.NullPointerException
Exception in thread "pool-35-thread-32" java.lang.NullPointerException
Exception in thread "pool-35-thread-34" java.lang.NullPointerException
Exception in thread "pool-35-thread-35" java.lang.NullPointerException
Exception in thread "pool-35-thread-33" java.lang.NullPointerException
```

cdjackson, 2017-07-12:
Can you take a look in the main debug log to see if you find some IeeeAddressResponse messages - ideally it would be good if you can correlate them with these errors, but even if not, if these errors are happening around the time of some IeeeAddressResponse message it would be good if you could post the debug file.

openhab-5iver, 2017-07-12:
Like this one? They come in pretty much continuously, but not currently getting the NPE.

`2017-07-12 12:51:32.664 [ZigBeeNetworkManager      ] - RX CMD: IeeeAddressResponse [52772/0 -> 0/0, cluster=8001, TID=NULL, status=SUCCESS, ieeeAddrRemoteDev=7CE524000014C420, nwkAddrRemoteDev=52772, numAssocDev=0, startIndex=null, nwkAddrAssocDevList=[]]`

cdjackson, 2017-07-12:
Ok - I'd need to see one with the NPE. This message looks ok - specifically, I'm wondering if the numAssocDev shows as null when there's an error.

openhab-5iver, 2017-07-12:
I kicked off a discovery, which I though might have triggered the NPEs last night, and saw one of the single line ones. So I grabbed the logs from the last hour. I'm not sure on the timing, but think it occurred within a 15 minute window. I'll try to find something...

openhab-5iver, 2017-07-12:
I don't think there is a direct correlation. There was only one NPE in Karaf but lots of them in the log. This is not all of them...

```
2017-07-12 13:14:32.159 [ZigBeeNetworkManager      ] - RX CMD: IeeeAddressResponse [33836/0 -> 0/0, cluster=8001, TID=NULL, status=DEVICE_NOT_FOUND, ieeeAddrRemoteDev=null, nwkAddrRemoteDev=null, numAssocDev=null, startIndex=null, nwkAddrAssocDevList=[]]
2017-07-12 13:14:32.163 [ZigBeeNetworkManager      ] - RX CMD: IeeeAddressResponse [15829/0 -> 0/0, cluster=8001, TID=NULL, status=DEVICE_NOT_FOUND, ieeeAddrRemoteDev=null, nwkAddrRemoteDev=null, numAssocDev=null, startIndex=null, nwkAddrAssocDevList=[]]
2017-07-12 13:14:32.167 [ZigBeeNetworkManager      ] - RX CMD: IeeeAddressResponse [21751/0 -> 0/0, cluster=8001, TID=NULL, status=DEVICE_NOT_FOUND, ieeeAddrRemoteDev=null, nwkAddrRemoteDev=null, numAssocDev=null, startIndex=null, nwkAddrAssocDevList=[]]
2017-07-12 13:14:32.174 [ZigBeeNetworkManager      ] - RX CMD: IeeeAddressResponse [35243/0 -> 0/0, cluster=8001, TID=NULL, status=DEVICE_NOT_FOUND, ieeeAddrRemoteDev=null, nwkAddrRemoteDev=null, numAssocDev=null, startIndex=null, nwkAddrAssocDevList=[]]
2017-07-12 13:14:32.202 [ZigBeeNetworkManager      ] - RX CMD: IeeeAddressResponse [47277/0 -> 0/0, cluster=8001, TID=NULL, status=SUCCESS, ieeeAddrRemoteDev=7CE524000012CC42, nwkAddrRemoteDev=47277, numAssocDev=null, startIndex=null, nwkAddrAssocDevList=[]]
2017-07-12 13:14:32.204 [ZigBeeNetworkManager      ] - RX CMD: IeeeAddressResponse [7798/0 -> 0/0, cluster=8001, TID=NULL, status=DEVICE_NOT_FOUND, ieeeAddrRemoteDev=null, nwkAddrRemoteDev=null, numAssocDev=null, startIndex=null, nwkAddrAssocDevList=[]]
2017-07-12 13:14:32.206 [ZigBeeNetworkManager      ] - RX CMD: IeeeAddressResponse [46214/0 -> 0/0, cluster=8001, TID=NULL, status=DEVICE_NOT_FOUND, ieeeAddrRemoteDev=null, nwkAddrRemoteDev=null, numAssocDev=null, startIndex=null, nwkAddrAssocDevList=[]]
2017-07-12 13:14:32.209 [ZigBeeNetworkManager      ] - RX CMD: IeeeAddressResponse [3117/0 -> 0/0, cluster=8001, TID=NULL, status=DEVICE_NOT_FOUND, ieeeAddrRemoteDev=null, nwkAddrRemoteDev=null, numAssocDev=null, startIndex=null, nwkAddrAssocDevList=[]]
2017-07-12 13:14:32.232 [ZigBeeNetworkManager      ] - RX CMD: IeeeAddressResponse [12203/0 -> 0/0, cluster=8001, TID=NULL, status=DEVICE_NOT_FOUND, ieeeAddrRemoteDev=null, nwkAddrRemoteDev=null, numAssocDev=null, startIndex=null, nwkAddrAssocDevList=[]]
2017-07-12 13:14:32.319 [ZigBeeNetworkManager      ] - RX CMD: IeeeAddressResponse [37647/0 -> 0/0, cluster=8001, TID=NULL, status=DEVICE_NOT_FOUND, ieeeAddrRemoteDev=null, nwkAddrRemoteDev=null, numAssocDev=null, startIndex=null, nwkAddrAssocDevList=[]]
2017-07-12 13:14:32.364 [ZigBeeNetworkManager      ] - RX CMD: IeeeAddressResponse [52772/0 -> 0/0, cluster=8001, TID=NULL, status=DEVICE_NOT_FOUND, ieeeAddrRemoteDev=null, nwkAddrRemoteDev=null, numAssocDev=null, startIndex=null, nwkAddrAssocDevList=[]]
2017-07-12 13:14:32.368 [ZigBeeNetworkManager      ] - RX CMD: IeeeAddressResponse [12203/0 -> 0/0, cluster=8001, TID=NULL, status=DEVICE_NOT_FOUND, ieeeAddrRemoteDev=null, nwkAddrRemoteDev=null, numAssocDev=null, startIndex=null, nwkAddrAssocDevList=[]]
2017-07-12 13:14:32.372 [ZigBeeNetworkManager      ] - RX CMD: IeeeAddressResponse [8916/0 -> 0/0, cluster=8001, TID=NULL, status=DEVICE_NOT_FOUND, ieeeAddrRemoteDev=null, nwkAddrRemoteDev=null, numAssocDev=null, startIndex=null, nwkAddrAssocDevList=[]]
```

openhab-5iver, 2017-07-12:
```
2017-07-12 13:14:32.201 [ZigBeeDongleEzsp          ] - Unhandled EZSP Frame: EzspSendUnicastResponse [status=EMBER_SUCCESS, sequence=63]
2017-07-12 13:14:32.202 [AshFrameHandler           ] - RX EZSP: EzspIncomingMessageHandler [type=EMBER_INCOMING_UNICAST, apsFrame=EmberApsFrame [profileId=0, clusterId=32769, sourceEndpoint=0, destinationEndpoint=0, options=[], groupId=0, sequence=32], lastHopLqi=245, lastHopRssi=-67, sender=47277, bindingIndex=255, addressIndex=255, messageContents=BA 00 42 CC 12 00 00 24 E5 7C AD B8]
2017-07-12 13:14:32.202 [ZigBeeDongleEzsp          ] - RX: EzspIncomingMessageHandler [type=EMBER_INCOMING_UNICAST, apsFrame=EmberApsFrame [profileId=0, clusterId=32769, sourceEndpoint=0, destinationEndpoint=0, options=[], groupId=0, sequence=32], lastHopLqi=245, lastHopRssi=-67, sender=47277, bindingIndex=255, addressIndex=255, messageContents=BA 00 42 CC 12 00 00 24 E5 7C AD B8]
2017-07-12 13:14:32.202 [ZigBeeNetworkManager      ] - RX CMD: IeeeAddressResponse [47277/0 -> 0/0, cluster=8001, TID=NULL, status=SUCCESS, ieeeAddrRemoteDev=7CE524000012CC42, nwkAddrRemoteDev=47277, numAssocDev=null, startIndex=null, nwkAddrAssocDevList=[]]
2017-07-12 13:14:32.204 [AshFrameHandler           ] - RX EZSP: EzspIncomingMessageHandler [type=EMBER_INCOMING_UNICAST, apsFrame=EmberApsFrame [profileId=0, clusterId=32769, sourceEndpoint=0, destinationEndpoint=0, options=[], groupId=0, sequence=2], lastHopLqi=254, lastHopRssi=-83, sender=7798, bindingIndex=255, addressIndex=255, messageContents=BA 81 F0 36 06 00 00 24 E5 7C AD B8]
2017-07-12 13:14:32.204 [ZigBeeDongleEzsp          ] - RX: EzspIncomingMessageHandler [type=EMBER_INCOMING_UNICAST, apsFrame=EmberApsFrame [profileId=0, clusterId=32769, sourceEndpoint=0, destinationEndpoint=0, options=[], groupId=0, sequence=2], lastHopLqi=254, lastHopRssi=-83, sender=7798, bindingIndex=255, addressIndex=255, messageContents=BA 81 F0 36 06 00 00 24 E5 7C AD B8]
2017-07-12 13:14:32.204 [ZigBeeNetworkManager      ] - RX CMD: IeeeAddressResponse [7798/0 -> 0/0, cluster=8001, TID=NULL, status=DEVICE_NOT_FOUND, ieeeAddrRemoteDev=null, nwkAddrRemoteDev=null, numAssocDev=null, startIndex=null, nwkAddrAssocDevList=[]]
2017-07-12 13:14:32.206 [AshFrameHandler           ] - RX EZSP: EzspIncomingMessageHandler [type=EMBER_INCOMING_UNICAST, apsFrame=EmberApsFrame [profileId=0, clusterId=32769, sourceEndpoint=0, destinationEndpoint=0, options=[], groupId=0, sequence=174], lastHopLqi=254, lastHopRssi=-78, sender=46214, bindingIndex=255, addressIndex=255, messageContents=BA 81 84 18 16 00 00 24 25 7C AD B8]
2017-07-12 13:14:32.206 [ZigBeeDongleEzsp          ] - RX: EzspIncomingMessageHandler [type=EMBER_INCOMING_UNICAST, apsFrame=EmberApsFrame [profileId=0, clusterId=32769, sourceEndpoint=0, destinationEndpoint=0, options=[], groupId=0, sequence=174], lastHopLqi=254, lastHopRssi=-78, sender=46214, bindingIndex=255, addressIndex=255, messageContents=BA 81 84 18 16 00 00 24 25 7C AD B8]
```

cdjackson, 2017-07-12:
Thanks - I think this captures it.  I’ll take a look at this in the next day or so.

PR #50 (2017-07-13): Fix NPE if IeeeAddressResponse provides null for number of devices
Fixes #49
closed 2017-07-13 by cdjackson