zsmartsystems/com.zsmartsystems.zigbee #52: Unable to join Phillips Hue dimmer switch

jimduchek, 2017-07-17:
I'm attempting to get a Phillips Hue dimmer switch to talk to a HUSBZB-1.  Obviously a pretty untested setup, but I'd love to get this working, as there are just so few battery-powered switches/dimmers out there.   This is less an 'issue' and more just opening a discussion.  TLDR on this: It doesn't seem, out of the box, the switch is replying to discovery with it's own sender network ID, but with 0 as the sender.  So the stack doesn't see the replies.

Where we seem to be stuck is, we send out an IeeeAddressRequest:
02:51:29.704  DEBUG  TX CMD: IeeeAddressRequest [0/0 -> 2858/0, cluster=0001, TID=5C, nwkAddrOfInterest=2858, requestType=1, startIndex=0]

And we do get a response -- with a sender of 0, not 2858.  So it never 'sees' the response (as it expects the sender of the response to match the destination of the request).

02:51:29.740  TRACE  <-- RX ASH frame: AshFrameData [frmNum=0, ackNum=0, reTx=false, data=5C 90 45 00 00 00 01 80 00 00 40 01 00 00 2C FF 00 00 00 FF FF 0D 00 00 32 6F 19 02 01 88 17 00 2A 0B 00]
02:51:29.742  DEBUG  RX CMD: IeeeAddressResponse [0/0 -> 0/0, cluster=8001, TID=NULL, status=SUCCESS, ieeeAddrRemoteDev=0017880102196F32, nwkAddrRemoteDev=2858, numAssocDev=0, startIndex=null, nwkAddrAssocDevList=[]]
02:51:39.707  DEBUG  Ieee Address for 2858 returned null

I'm unclear as to why the sender isn't set -- I think it might be possible that the dongle may be intercepting the request (since it knows the 'right answer'?  AFAIK it should treat this as any other unicast command, but... it is not)  At any rate, I changed this in zdo/command/IeeeAddressRequest.java (I'm not sure this is a good idea for everybody, but for my setup, it doesn't matter for the time being, as it should be correct on anything but a router I think):

//        return (((IeeeAddressRequest) request).getDestinationAddress()
//                .equals(((IeeeAddressResponse) response).getSourceAddress()));
        return (((IeeeAddressRequest) request).getNwkAddrOfInterest()
                .equals(((IeeeAddressResponse) response).getNwkAddrRemoteDev()));

Things continued well after that and other messages have the correct response (including sender).  I can use the console to query stuff on the device, but the buttons on the switch don't seem to do anything.  With that said, the extent of my testing there was pushing a button, seeing nothing in the logs, and deciding it is time for bed.

Perhaps a better solution is when: 

RX EZSP: EzspChildJoinHandler [index=0, joining=true, childId=25851, childEui64=0017880102196F32, childType=EMBER_SLEEPY_END_DEVICE]

Is received, start the discovery process on the second step and skip the IEEE address discovery, since we were already told.  Unfortunately, the childType is not passed up into the discovery stack, and I think the ieee address discovery step needs to happen for routers.

cdjackson, 2017-07-17:
> I'm unclear as to why the sender isn't set 

I suspect that because this is a sleepy end device, the router to which the device is currently connected is responding by proxy. So, this is saying that the response comes from the router (address 0 in this case) but the end device address is correctly set.

So, the matcher should match on  nwkAddrRemoteDev and not the source address.

> start the discovery process on the second step and skip the IEEE address discovery

Yes, you are probably right that we could be smarter here when the device is a sleepy end device. For non-sleepy devices we should make the request even if we know the IEEE address since the other reason for this call is to get the associated devices (which is not applicable to sleepy devices). I thought the child join handler was passed up in some form - I'll check, bit I thought I had an abstract method for this (ie to make it dongle independent) - if it doesn't include the type, I'll look to add it.

I'll look at the above changes tonight or tomorrow.

cdjackson, 2017-07-28:
@jimduchek are you able to confirm that the above change fixes the issue with the sleepy device (ie correlation of the IeeeAddressRequest message and response). If so I'll merge this change since it looks ok here.

cdjackson, 2018-02-12:
Closing this due to inactivity. If there's still an issue here, then feel free to reopen if there's further information...

closed 2018-02-12 by cdjackson