zsmartsystems/com.zsmartsystems.zigbee #76: Ember chip is sending XON and AshFrameHandler is not handling it.

cschwer, 2017-10-11:
On first startup, often the ember chip will send back a couple XON bytes (0x11)  AshFrameHandler does not exclude these and adds them to the input buffer, resulting in a bad packet.  ie:

ERROR com.zsmartsystems.zigbee.dongle.ember.ash.AshFrameHandler - <-- RX ASH frame: BAD PACKET 11 11 52 7B A1 E9 54 26 20

From the ASH protocol documentation, it specifies that certain bytes are reserved. Including:
0x11 XON: Resume transmission
Used in XON/XOFF flow control. Always ignored if received by the NCP.
0x13 XOFF: Stop transmission
Used in XON/XOFF flow control. Always ignored if received by the NCP.

cdjackson, 2017-10-11:
I'll look to handle (ignore) these, but what firmware are you using? The handler was designed to use the HW flow control - not XON/XOFF so I wonder if you are using HW or SW flow control?  I don't think I've ever seen these with the HW flow control firmware.

cschwer, 2017-10-12:
Thanks, I'll try it out.  I'm using the Nortek HUSBZB-1  ..   I've never seen the XOFF byte,, just usually those 2 XON bytes right when it first starts up and connects to the stick.  Do you know how I can tell what firmware it is using?

cschwer, 2017-10-12:
Yes, this fixed it for me.  Thanks!

cdjackson, 2017-10-12:
Great - thanks.

I'm not sure how to tell what firmware has been programmed after it's programmed. When programming the Ember there are different versions and they have part filenames of _xon_off, or _dtr_dsr (or something similar).

Anyway, hopefully this makes it work with either, although as it doesn't actually perform any flow control with XON/XOFF, it's still probably best to use HW flow control if possible.

cdjackson, 2017-10-12:
Closed by #77

closed 2017-10-12 by cdjackson