zsmartsystems/com.zsmartsystems.zigbee #95: Add network layer incoming packet interface

cdjackson, 2017-11-05:
It would be good to implement a standard network layer interface to allow notifications from dongles into the framework.

Specific notifications I want to see implemented -:
* Ack/Nak responses - these should cut short the internal timeout on transactions if the stick has already given up on a transaction (I've started to implement this, but want a standard interface if possible).
* Incoming route notifications.
* Trust centre notifications.
* ...

Comments/thoughts appreciated...

cschwer, 2017-11-14:
I would love to see the incoming route notifications being pushed up to the ZigBeeTransportReceive level

cdjackson, 2017-11-14:
Yep - this is also one of my use cases. Any thoughts on an interface?

I've not thought this through fully yet, but I  was thinking of a single method in the receive interface, and we define a base class for network level responses that each individual network command can implement (similar to the hierarchy for commands in general).

I think this is neater, and more future proof than adding a separate method for each event. Given that these network level responses tend to be dongle specific I don't think we can use a binary interface such as is used in the cluster library, but if you have other thoughts, please feel free to suggest.

I will likely look at this in a couple of weeks so any thoughts are welcome :)

PR #479 (2018-12-24): Initial addition of transaction manager classes with dongle feedback
This refactors the transaction handling in the framework. It introduces a new ```ZigBeeTransactionManager``` class which manages the transactions. It also introduces a new method ```ZigBeeTransportReceive. receiveCommandStatus()``` which provides a mechanism for the transport layer to provide feedback on the state of its transaction management.

```ZigBeeTransportReceive. receiveCommandStatus()``` should be called by the dongle when there is an update to the transaction - eg when the dongle actually sends the data over the ai [... cut]
closed 2018-12-24 by cdjackson