zsmartsystems/com.zsmartsystems.zigbee #109: Support EZSP version 5 and later.

haojiesong, 2017-11-13:


cdjackson, 2017-11-13:
I believe that EZSP5 is simply an extension of EZSP4 with a number of new commands, so this should already be supported in general. If there are new commands that you require implementing, then please list them and I'm sure they can be added.

If there's something more general that has changed then I'll take a look, but I think it's just an extension of 4?

haojiesong, 2017-11-18:
From EmberZNet 5.9.x, the EZSP version has upgraded from 4 to 5. It is incompatible and need firstly do special handling.

cdjackson, 2017-11-18:
Ok, from what I could see there was minimal difference - UG100 doesn't seem to indicate anything special - can you provide a reference to the differences as I can't seem to find anything from a quick search.
Thanks.

cdjackson, 2017-11-18:
Ok, I see in the protocol format (section 3) of UG100 that there is now an extra byte in the protocol. Strangely this "small" change doesn't feature in the documents "what's changed" list!!!

I'll take a look at adding support for this in the near future as I might flash one of my devices with the latest NCP firmware.

PR #207 (2018-02-04): Migrate EZSP5+ code from previous codebase with existing ASH2 Handler
Refactor #122 to only use ASH2 and add tests.

Closes #109 
Closes #122 
Closes #200
closed 2018-02-04 by cdjackson