zsmartsystems/com.zsmartsystems.zigbee #113: add a public method to send EzspFrameRequest?

cschwer, 2017-11-14:
Thoughts on adding a new public method to the ZigBeeDongleEzsp class to allow sending of raw EzspFrameRequest messages?  I'm currently re implementing the entire class in order to send some basic commands to the stick.. ie, getting the local ieee and network address of the stick.   If this method were added I would be able to use the built in class.

public EzspFrameResponse sendFrameRequest(EzspFrameRequest request, Class responseType) {
    EzspSingleResponseTransaction transaction = new EzspSingleResponseTransaction(request, responseType);
    ashHandler.sendEzspTransaction(transaction);
    return transaction.getResponse();
}

cdjackson, 2017-11-14:
I'm not sure that this is needed to be honest. It is possible to do this right now (and the library already does it) by using the ZDO commands. The network address (I think) must always be 0 as it's configured with the coordinator, and you can get the Ieee address directly form the node as the discovery should populate this.

Something along the following lines should work -:

```
node = networkManager.getNode(0);
address = node.getIeeeAddress();
```

In general, I prefer to use ZDO/ZCL messages where at all possible so that we have commonality between different dongles. I'm sure that there will be a need in some cases to provide dongle specific interfaces, but I would like to minimise this and I think in general it is probably only required for configuration (there is a case in Telegesis dongle where it requires a password to be configured so this is simply done as a public method on the dongle class).

What do you think? If you have other use cases then we can discuss further, but I would really prefer to avoid putting too much into the dongle if it can be avoided as this should be a common interface.

cschwer, 2017-11-14:
A couple other commands I'm using are EzspSetConcentratorRequest and EzspAddEndpointRequest.   I noticed that you removed the createEndpoints() method recently so I need a way to still configure the endpoints as well as configure the stick as a concentrator.

Although now that I think of it, the createEndpoints needs to be invoked before the stick is brought online, so I'm not sure if there is a way to add a hook into the initialize function for that, so it may be just a matter of fact that I need to continue implementing my own ZigBeeDongleEzsp.

cdjackson, 2017-11-14:
> EzspSetConcentratorRequest 

We can create a public method for this - either in the EzspDongle or possibly the transport interface since it's probably a global need (ie across multiple dongles)

> EzspAddEndpointRequest

Why do you need this? I've not found a need for this in any dongle yet so I'm wondering why you have found this is something you need to add. Normally on all dongles I've implemented (CC2531, Telegesis, and also Ember) the dongle will pass all data through to the serial interface if no interface is defined. 

Can you describe the issue that implementing this actually solves?

cschwer, 2017-11-14:
for AddEndpoint, It's not strictly needed, since the code I'm writing could respond to the Match Descriptor Requests on it's own, but I find it easier to just let the ember stack take care of it and less code for me to maintain.  I suppose that is a trade off of writing a library, trying to make a standard interface for all the dongles.

cdjackson, 2017-11-14:
Agreed. I would probably prefer to add functionality to the main library and avoid adding specifics to the dongle if possible.

I'm happy to add functions for handling the match descriptor if needed - this is actually now partly included in the OTA code - we could expand on this if needed.

cdjackson, 2017-11-15:
Another suggestion. To avoid adding lots of methods to configure the system, how about we add a single method that takes a dictionary of configuration strings. I think the dictionary should be an enumeration so that it's handled consistently across dongles - we can then add the concentrator type etc into this dictionary.

If you're ok with this, I'll make a proposal.

cschwer, 2017-11-16:
Yep, that sounds like a good idea.   as to your earlier comment about match descriptor, the important one is  the IAS Zone cluster.  Some devices won't join if the dongle doesn't support that cluster.

cdjackson, 2017-11-16:
Ok, I'll look at adding this tonight as I have a PR ready to add configuration for trust centre join/rejoin settings which I'd implemented as a separate config method - I'll change this to use the new system.

Re the match descriptor - is this SmartThings sensor? I've not looked in detail but I noticed the ST motion sensor is looking for a descriptor which I guess is a Samsung custom descriptor (something like FFC0) but I think I've also seen the AIS cluster requested by a device I have as well.

cschwer, 2017-11-16:
Yes, it was the SmartThings motion sensor.  It would drop off the network and refuse to be configured if I didn't respond to match descriptor requests for 0x0500 (IAS Zone).  I also noticed that is was requesting the custom cluster, but never responded to it and didn't have a problem.

cdjackson, 2017-11-18:
#116 is merged - any further comments are welcome (and more configuration settings of course).

cdjackson, 2018-01-31:
Closing this as I think we've covered most of the issues discussed. If there is anything missing then please open another issue so we can specifically discuss it.

closed 2018-01-31 by cdjackson