zsmartsystems/com.zsmartsystems.zigbee #128: send reset when ember znet sends error

cschwer, 2017-11-24:
According to the docs:

When the NCP enters the FAILED state, the NCP sends an ERROR frame containing a reset or error code and will reply to all subsequent frames received, except RST, with an ERROR frame. To reinitialize the ASH protocol, the Host must reset the NCP by either asserting the nRESET pin or sending the RST frame.

cschwer, 2017-11-24:
I will create a PR for this shortly.

cdjackson, 2017-11-25:
It would be really good if you could take a look at the ASH/EZSP WIP PR (#122) as I'd like to move over to this some time soon. It needs a little more testing, but I've now tested it against ASH V2 with both EZSP versions 4 and 5 having updated the firmware to 5.10 and then back to 5.8. It still needs a bit more work on ASH V3, but as it stands the ASH V2 implementation now has more features as the current code so it would be good to get it migrated before doing too many more changes to the existing ASH handler.

Let me know what you think.

cdjackson, 2018-11-03:
Resetting the NCP and continuing at the ASH level does not seem like it will resolve the issue. Once RESET, the NCP must be reconfigured - the VERSION command must be sent, endpoints configured, and the NCP brought back into operation again.

Instead of resetting the ASH layer, and leaving the NCP unusable at application layer, I propose to set the dongle link state offline and leave the application to resolve this by restarting the link.

There may be a case for performing this reinitialisation within the Dongle layer, and if someone thinks this is a better approach, we can look at this.