zsmartsystems/com.zsmartsystems.zigbee #159: Example code

anton-johansson, 2017-12-30:
Is there any example code available associated with this repository? I'm looking at getting into the ZigBee market (mainly for Philips HUE at the moment) and I'm looking for a decent Java library that I can utilize.

This looks promising (commits in the last few days) but I would like to see some example code before actually buying any products.

cdjackson, 2017-12-30:
Yes - there is the console application which is a fully working example. It's hacky, and simple (it's a CLI!) but it's what I use to test so it does for the most part work and for sure provides a good example.

anton-johansson, 2017-12-30:
Ahh, [this one](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/tree/master/com.zsmartsystems.zigbee.console/src/main/java/com/zsmartsystems/zigbee/console). I didn't realize. Thanks!

closed 2017-12-30 by anton-johansson

cdjackson, 2017-12-30:
Yes - if you open up the project in Eclipse at least, then you should see it as a separate package/app that can be run…

anton-johansson, 2017-12-30:
Yep, I'll try it out! I guess I'll order a ConBee and a basic Philips HUE device to mess around with.

By the way, is this project some sort of replacement for [zigbee4java](https://github.com/tlaukkan/zigbee4java)? I saw that you were a major contributor there, but the project seems dead.

cdjackson, 2017-12-30:
Note that ConBee still has a couple of issues - probably nothing major, but it might need a little bit of work still to tidy a few things up.

I guess it could be called a replacement for zb4j - at the time I wanted to do a lot of additions in zb4j there was no response from the other maintainer and I was uncomfortable with merging major changes without discussion. This project is being used in a couple of places, and I expect development to continue :)

anton-johansson, 2017-12-30:
> This project is being used in a couple of places, and I expect development to continue :)

Good to hear! :)

> Note that ConBee still has a couple of issues

Can you suggest any other dongle that works better? I might end up buying the ConBee anyway, because it's the only one I can find on my regular webshop. If so, I'll try to help sorting out those minor issues. But don't get your hopes up, I'm all new to this. :)

cdjackson, 2017-12-30:
Currently most of my testing is using the Telegesis dongle which are reasonably readily available in most countries on element14/farnell. Where are you located?

anton-johansson, 2017-12-30:
[This one](https://www.m.nu/zigbee-usb-gpio/telegesis-usb-zigbee-transciever-module)? :) Swedish site, but still.

cdjackson, 2017-12-30:
Yes - that’s the one…

anton-johansson, 2018-02-20:
@cdjackson: Is there any list of known supported devices? I'm planning on buying some lights and maybe a motion sensor. When it comes to the motion sensor, I'm not sure if I should go for a ZWave one or a ZigBee one.

I didn't want to open up a new issue for this, which leads me to another question: Is there any other place for questions and discussions like this? Like a Slack or Gitter channel?

cdjackson, 2018-02-20:
Please see [this link](https://github.com/openhab/org.openhab.binding.zigbee). It's for the openHAB binding which uses the library here. It lists a lot of devices that people are using although I'm sure there are more.

I don't currently have other channels for questions - I guess I could create a Slack channel, but at the moment I've not done this.

anton-johansson, 2018-02-20:
Ah, great! Thanks for the link. :)

Allright, good to know!