zsmartsystems/com.zsmartsystems.zigbee #166: Karaf NPE: ZigBeeCommandNotifier.java

openhab-5iver, 2018-01-05:
Zigbee binding 2.3.0.201801042145 

In case it is related, I had a bulb that refused to come online, so I reset it and rediscovered. I also had recently updated the binding before I noticed this. I have some debug logs, if you want them.

```
Exception in thread "pool-3555-thread-2" java.lang.NullPointerException
        at com.zsmartsystems.zigbee.zcl.ZclAttributeNormalizer.normalizeZclData(ZclAttributeNormalizer.java:24)
        at com.zsmartsystems.zigbee.zcl.ZclCluster.handleAttributeStatus(ZclCluster.java:843)
        at com.zsmartsystems.zigbee.ZigBeeEndpoint.commandReceived(ZigBeeEndpoint.java:410)
        at com.zsmartsystems.zigbee.ZigBeeNode.commandReceived(ZigBeeNode.java:680)
        at com.zsmartsystems.zigbee.ZigBeeCommandNotifier$1.run(ZigBeeCommandNotifier.java:65)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        at java.lang.Thread.run(Thread.java:748)
```

cdjackson, 2018-01-05:
> I have some debug logs, if you want them.

Yes please - I just need to see the bit immediately before this that shows the packet received. It's a strange place for an NPE...

openhab-5iver, 2018-01-05:
To clarify, I have the log but do not know the exact time the NPE occurred, but it was soon after the start of the binding. I have two more that have popped up though, so it looks like this is may not be a one off NPE. If the whole file isn't useful, I can try tee trick to get the log close to where these occur.

cdjackson, 2018-01-05:
Ok, send the log and I'll take a look as I know broadly what to look for so it shouldn't be too much hassle.

Thanks.

openhab-5iver, 2018-01-05:
Here you go...[trimmed.zip](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/1606702/trimmed.zip)

cdjackson, 2018-01-05:
Probably caused by this -:

```
2018-01-04 21:48:16.153 [DEBUG] [com.zsmartsystems.zigbee.ZigBeeNetworkManager     ] - RX APS: ZigBeeApsFrame [sourceAddress=24779/1, destinationAddress=0/1, profile=0104, cluster=0, addressMode=null, radius=0, sequence=50, payload=18 32 01 01 00 86]
2018-01-04 21:48:16.153 [DEBUG] [com.zsmartsystems.zigbee.ZigBeeNetworkManager     ] - RX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=SERVER_TO_CLIENT, disableDefaultResponse=true, manufacturerCode=0, sequenceNumber=50, commandId=1]
2018-01-04 21:48:16.153 [DEBUG] [com.zsmartsystems.zigbee.ZigBeeNetworkManager     ] - RX CMD: ReadAttributesResponse [Basic: 24779/1 -> 0/1, cluster=0000, TID=32, records=[Read Attribute Status Record: attributeDataType=null, attributeIdentifier=1, status=134, attributeValue=null]]
```

cdjackson, 2018-01-05:
Please advise if this can be closed...

openhab-5iver, 2018-01-06:
I have not seen it occur with the new version (2.3.0.201801052345).

cdjackson, 2018-01-06:
Ok, thanks. I'll close this - I'm sure you'll let me know if it does happen again ;)

closed 2018-01-06 by cdjackson