zsmartsystems/com.zsmartsystems.zigbee #169: Mesh update exception and Readsync exception at shutdown

openhab-5iver, 2018-01-06:
2.3.0.201801052345 

When stopping the zigbee binding, the "Mesh update exception' occurs every time, but always only five times. Five of my devices are currently powered off (one battery and 4 GE Link bulbs). The 'Readsync exception' occurs right before these, but not every time. After starting again, everything looks good.

[010618 readsync exception.zip](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/1608722/010618.readsync.exception.zip)

cdjackson, 2018-01-06:
I commented on this in our private discussion - this is the [interrupted exception](https://docs.oracle.com/javase/7/docs/api/java/lang/InterruptedException.html).  The reason is that the binding is stopping, and the threads are closing, so the thread is interrupted. I could silently ignore these exceptions I guess - it just means changing the way exceptions are caught.

openhab-5iver, 2018-01-06:
OK. It was the Readsync that looked different. It's on shutdown, so don't affect anything... just jump out at you when watching for exceptions. :)

cdjackson, 2018-01-06:
Yep - understood. Leave this open and I think I’ll change to silently ignore these exceptions - I do this in some places, but it depends on the location and what other exceptions are also being trapped.

PR #171 (2018-01-06): Don't log stack trace for InterruptedException
Doesn't log the stack trace if there's an InterruptedException during a transaction in readSync.

Closes #169
closed 2018-01-06 by cdjackson