zsmartsystems/com.zsmartsystems.zigbee #199: Ember cannot be initialized

TomTravaglino, 2018-02-01:
For our project we're using com.zsmartsystems.zigbee in combination with org.openhab.binding.zigbee and trying to create a bridge for the Ember-Controller. The bridge itself is created but it's never set in status ONLINE. We stepped with a debugger through the code and the summary of our investigation is that the ZigBee initialization apparently gets stuck because the AshFrameHandler never receives a com.zsmartsystems.zigbee.dongle.ember.ash.AshFrame.FrameType.RSTACK an therefore can't be completed. So the bridge stays in status Status=INITIALIZING.

Below you see the key points which describe how the ZigBee initialization gets stuck and does therefore not complete. The description starts in com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp.initialize(). Since the process is thread-driven some things can happen in a slightly different order or simultaneously. But finally it ends up in the same situation.

In com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp.initialize()
- The statement if (!serialPort.open()) returns true.
- The statement ashHandler.start(serialPort) starts the thread to read packets from a stream. Initially there is no data available and the thread goes into wait state at org.openhab.binding.zigbee.internal.ZigBeeSerialPort.read(int)
- The statement ashHandler.connect() sends a reset frame. See below what happens in com.zsmartsystems.zigbee.dongle.ember.ash.AshFrameHandler.outputFrame(AshFrame).

In com.zsmartsystems.zigbee.dongle.ember.ash.AshFrameHandler.outputFrame(AshFrame)
- the for-loop
  - com.zsmartsystems.zigbee.dongle.ember.ash.AshFrameRst.getOutputBuffer() returns [26, 192, 56, 188, 126]. So these values represent the reset frame, right? Is 126 <=> ASH_FLAG_BYTE (end of a frame)?
  - port.write writes the content of the buffer byte by byte into the output stream gnu.io.RXTXPort$SerialOutputStream
  - the invocation of port.write leads to at least one invocation of org.openhab.binding.zigbee.internal.ZigBeeSerialPort.serialEvent(SerialPortEvent) depending on whether it is already in execution. It can happen up to five times since port.write is invoked five times.
 
In org.openhab.binding.zigbee.internal.ZigBeeSerialPort.serialEvent(SerialPortEvent)
- When serialEvent(SerialPortEvent) is called gnu.io.RXTXPort$SerialInputStream.read() delivers the values 3, 0, 167 five times in total.
- All the occurrences of 3, 0, 167 are stored in a buffer of type int[]. This results in buffer = [3, 0, 167, 3, 0, 167, 3, 0, 167, 3, 0, 167, 3, 0, 167, 0...]
- When all the available data is processed this.notify() is invoked which wakes up the thread that is waiting in org.openhab.binding.zigbee.internal.ZigBeeSerialPort.read(int). Each a byte will be read out there and returned to com.zsmartsystems.zigbee.dongle.ember.ash.AshFrameHandler.getPacket()

In com.zsmartsystems.zigbee.dongle.ember.ash.AshFrameHandler.getPacket()
- port.read() returns a byte from the buffer populated in org.openhab.binding.zigbee.internal.ZigBeeSerialPort.serialEvent(SerialPortEvent) and waits for ASH_FLAG_BYTE in order to return the collected data.
- Since buffer doesn't contain ASH_FLAG_BYTE the thread gets stuck in the loop and will never return to the caller.
- So the caller (thread "AshFrameHandler") will never receive com.zsmartsystems.zigbee.dongle.ember.ash.AshFrame.FrameType.RSTACK and consequently not proceed with the initialization process and set "stateConnected" to true. So no further communication is possible since it depends on this variable being set to true.

cdjackson, 2018-02-01:
Please can you confirm what version of the Ember stack you are using?

TomTravaglino, 2018-02-01:
I assume you're question is about com.zsmartsystems.zigbee, right? It's 1.0.3
If it's about the FW then it's 5.8.5

cdjackson, 2018-02-01:
It was the firmware version I was interested in. I think this version should be compatible but I will need to check as around this version silabs changed th protocol. 

Is this running on an embedded system?

Can you provide a trace level log - I think this ought to then log all data sent and received.

tomhoefer, 2018-02-01:
@TomTravaglino I thought that 5.8.1 is the corresponding FW version

cdjackson, 2018-02-01:
What is being received is invalid - both ASH V2 and V3 use the flag value of 0x7E (126 dec) to mark the start/stop of a frame, and as you're not seeing this byte, something must be fundamentally wrong with communications.

Are you sure that the serial baud rate being used is compatible with the hardware? What rate are you using?

TomTravaglino, 2018-02-02:
@cdjackson Here the logs you asked for.
We're using a baud rate of 115200 and we're sure it's compatible with our hardware.
[Logs-Ember.log](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/1689219/Logs-Ember.log)

cdjackson, 2018-02-02:
It looks like you are using the SPI connection and not a UART - is that correct?

The SPI connection uses a different protocol than the ASH protocol used by the UART connection.

tomhoefer, 2018-02-02:
@cdjackson we are using a SPI connection ... is there something special that is to be considered?

cdjackson, 2018-02-02:
Yes - unfortunately SPI protocol is different than the UART protocol used in the dongles. Instead of the ASH layer that is used for the dongles, there is another low level layer employed. I’ve not looked at it in detail, but there are some differences in the way it works other than the framing I think.

I can’t fully correlate the 3 bytes that are seen in the logs since frames are meant to be longer than this, but frames in the SPI protocol are terminated with the byte A7 as seen in the logs so I’m confident this is the issue. There are a number of other hardware lines involved with the SP protocol as well - I’m not sure how they are mapped to the TTY driver in Linux.

I suggest we might want to discuss this issue further.

TomTravaglino, 2018-02-02:
@cdjackson Here are the logs of a test with a dongle. Don't know if it's helpful. 
[Logs-Ember-2.log](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/1689672/Logs-Ember-2.log)

tomhoefer, 2018-02-02:
@cdjackson yes - happy to discuss this issue further ... do you need any additional information?

cdjackson, 2018-02-04:
I confirm that the sequence ```03 00 A7``` is a valid SPI frame meaning the command was invalid. My guess is that the SPI protocol in this case has been implemented to look like a serial port and (hopefully!) it hides all the hardware requirements in the driver.

TomTravaglino, 2018-02-07:
@cdjackson Could you look into issue with the dongle? (See my last post with the attached file.)

cdjackson, 2018-02-07:
There is no response received from the dongle at all following the reset request.

TomTravaglino, 2018-02-07:
Is there anything we can do to provide you more information?

cdjackson, 2018-02-07:
At the moment, there's simply nothing received, so I can't think of anything to ask. I assume this is repeatable - the only other thing I can suggest is to get the trace log, but that will probably just show the same thing.

Also double check the port - I guess you have plugged an ember stick into your box, but I note on my (similar?) box this port already exists with nothing plugged in to the USB.

I'm currently implementing the SPI protocol, but am having a few issues getting a test environment configured...

hsudbrock, 2018-02-12:
@cdjackson @TomTravaglino I also looked into the issue with the dongle that @TomTravaglino raised; for me the dongle works fine when setting the baud rate to 57600 (but does not work when using the baud rate of 115200). So my best guess is that the issue with the dongle is caused by a wrong baud rate.

For completeness sake: I use the USB dongle from bitron video, which is identified on a Linux system by vendor ID 10c4 and product id 8b34 - i.e., the dongle which is called 'Qivicon ZigBee USB Radio Stick' in the Linux source code.

@TomTravaglino Does it work for you when using 57600 as baud rate?

TomTravaglino, 2018-02-12:
I can confirm, that it works with 57600.

PR #274 (2018-05-17): Implement the Ember SPI protocol
This implements the Ember SPI protocol. It currently polls the callbacks every 250ms and will continue to poll immediately if further callbacks are outstanding.

Fixes #199 - @TomTravaglino you might like to take a look. I've currently not tested it extensively other than to make sure the network comes up and communications seems to work ok. I did get an ST motion sensor to join, so I think it's good.
closed 2018-05-17 by cdjackson