zsmartsystems/com.zsmartsystems.zigbee #210: Adding direct access to ezsp dongle

sebastienmarion, 2018-02-09:
**Goal** :
- be able to use mfglib frame :
-- to create a 802.15.4 sniffer
-- to pass radio certification process of a final product

**Need** :
- split initialization function in individual function (mfglib start shall be done just after a protocl version request)
- add a function to call ezsp frame
- add ezspFrameHandler listener
- managed a null zigbeeTransportReceive handler

**Request** :
create a new branch to develop this new functionnalities

cdjackson, 2018-08-18:
This has now been implemented. Not all mfglib methods have been added, however the start and end methods, and ability to receive packets.

A sniffer is implemented in this project 
 - https://github.com/zsmartsystems/com.zsmartsystems.zigbee.sniffer.

cdjackson, 2018-09-28:
#407 implements the remaining ```mfglib``` methods. I think this issue can now be closed @sebastienmarion

closed 2018-09-28 by cdjackson