zsmartsystems/com.zsmartsystems.zigbee #211: Implement Green Power Commissioning Tool capability

sebastienmarion, 2018-02-09:
Goal
Implement green power device  : GP Commissioning Tool (0x0064)

Need
- Implement missing ezsp frame
- Open a discussion of the best way to do this and support with other dongle

Request
- create a new branch to develop this functionalities
- add on this branch ezsp frame (information from UG100: EZSP Reference Guide Rev. 3.2) :
-- gpProxyTableProcessGpPairing (0xC9)
-- dGpSend (0xC6)
-- dGpSentHandler (0xC7)
-- gpepIncomingMessageHandler (0xC5)
-- gpProxyTableGetEntry (0xC8)
-- gpProxyTableLookup (0xC0)

sebastienmarion, 2018-02-19:
create branch "green_power_commissioning_tool_#211"

goal to implement GPCT and so green power cluster on endpoint 242:

[image]

sebastienmarion, 2018-02-19:
@cdjackson, I see cluster is auto-generated. Can you generate green power cluster (0x0021) describe in document "ZigBee PRO Green Power feature specification - Basic functionality set - Version 1.0" ?
I propose to developp GPCT as an new app like iasclient.

cdjackson, 2018-03-10:
> Can you generate green power cluster (0x0021) describe in document "ZigBee PRO Green Power feature specification - Basic functionality set - Version 1.0" ?

Sorry - I've not had the time to do this, but note that this is something that you can do as the ZCL autocoder is included. 

You can find the definition files [here](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/tree/master/com.zsmartsystems.zigbee.autocode/src/main/resources). The format is hopefully reasonably obvious and hopefully the GP cluster fits without additional constructs being needed - if there are things that don't fit, then feel free to discuss or extend.

sebastienmarion, 2018-03-12:
ok, great.
I have a time issue too and hope to come back on the subject in two months.

cdjackson, 2019-07-03:
I have rebased the branch from @louisgourinchas onto the current master with a view to reviewing this further in the near future.

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/tree/legrandgroup

It might be worth using this branch for any future updates if possible.