zsmartsystems/com.zsmartsystems.zigbee #223: Frequent logging when command is not supported

openhab-5iver, 2018-02-20:
[openHAB forum](https://community.openhab.org/t/zigbee-binding/15763/864?u=5iver)

cdjackson, 2018-02-27:
```
PowerDescriptorResponse [0/0 -> 0/0, cluster=8003, TID=NULL, status=NO_DESCRIPTOR, nwkAddrOfInterest=null, powerDescriptor=null]
```

cdjackson, 2018-03-03:
From the original OH forum discussion -:

```
2018-02-19 19:15:50.060 [DEBUG] [ystems.zigbee.internal.ZigBeeNodeServiceDiscoverer] - 7CE524000014AF33: Node SVC Discovery ManagementRoutingRequest returned CommandResult [ERROR (UNKNOWN,0x02), ManagementRoutingResponse [7491/0 -> 0/0, cluster=8032, TID=NULL, status=NOT_SUPPORTED, routingTableEntries=null, startIndex=null, routingTableList=[]]]
```

openhab-5iver, 2018-03-03:
Are you asking/stating something or making notes? 😄

cdjackson, 2018-03-03:
Just making notes so I don’t have to find the relevant bits in the forum each time ;)

openhab-5iver, 2018-03-03:
And I'll make a note of that for when reporting issues in the future!

cdjackson, 2018-03-03:
Thanks - the reference to the forum is good as it gives the background, but it’s nice to have the nitty gritty in the issue :)

cdjackson, 2018-12-08:
This should be closed. The mesh update is handling the NOT_SUPPORTED response and will not request these messages again after the first unsupported response.

closed 2018-12-08 by cdjackson