zsmartsystems/com.zsmartsystems.zigbee #225: Philips Hue Dimmer Off command not working

tavalin, 2018-02-21:
I've been testing the On/Off switch functionality with a Philips Hue Dimmer.  The device pairs and is discovered in OH.  

Pressing the 'on' button on the Dimmer works and the OH channel updates to ON correctly but when pressing the 'off' button the library does not translate the zigbee packet into an off command,

From what I've gathered purely from reading the src it looks like there is a mismatch between what the dimmer sends for the commandId (64) and what is expected for an off command (0).

Here's the 'on' button press log:

```
2018-02-21 22:41:09.981 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TELEGESIS RX: Data 52 58 3A 33 44 33 45 2C 30 31 30 34 2C 30 31 2C 30 31 2C 30 30 30 36 2C 30 33 3A 01 14 01 2C 2D 35 31 2C 46 46
2018-02-21 22:41:09.986 [DEBUG] [gesis.internal.TelegesisFrameHandler] - Telegesis event received: TelegesisReceiveMessageEvent [ieeeAddress=null, networkAddress=15678, profileId=260, destinationEp=1, sourceEp=1, clusterId=6, messageData=01 14 01, rssi=-81, lqi=255]
2018-02-21 22:41:09.991 [DEBUG] [ngle.telegesis.ZigBeeDongleTelegesis] - Telegesis RX: TelegesisReceiveMessageEvent [ieeeAddress=null, networkAddress=15678, profileId=260, destinationEp=1, sourceEp=1, clusterId=6, messageData=01 14 01, rssi=-81, lqi=255]
2018-02-21 22:41:09.995 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=15678/1, destinationAddress=0/1, profile=0104, cluster=6, addressMode=null, radius=0, sequence=0, payload=01 14 01]
2018-02-21 22:41:10.000 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=CLUSTER_SPECIFIC_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=20, commandId=1]
2018-02-21 22:41:10.005 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: OnCommand [On/Off: 15678/1 -> 0/1, cluster=0006, TID=14]
2018-02-21 22:41:10.009 [DEBUG] [converter.ZigBeeConverterSwitchOnoff] - 0017880103E47DAF: ZigBee command receiveds OnCommand [On/Off: 15678/1 -> 0/1, cluster=0006, TID=14]
2018-02-21 22:41:10.014 [DEBUG] [converter.ZigBeeBaseChannelConverter] - 0017880103E47DAF: Channel zigbee:device:zigbee:0017880103e47daf:0017880103E47DAF_1_switch_onoff updated to ON
```

Here's the 'off' button press log:

```
2018-02-21 22:39:42.435 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TELEGESIS RX: Data 52 58 3A 33 44 33 45 2C 30 31 30 34 2C 30 31 2C 30 31 2C 30 30 30 36 2C 30 35 3A 01 12 40 00 00 2C 2D 35 31 2C 46 46
2018-02-21 22:39:42.441 [DEBUG] [gesis.internal.TelegesisFrameHandler] - Telegesis event received: TelegesisReceiveMessageEvent [ieeeAddress=null, networkAddress=15678, profileId=260, destinationEp=1, sourceEp=1, clusterId=6, messageData=01 12 40 00 00, rssi=-81, lqi=255]
2018-02-21 22:39:42.448 [DEBUG] [ngle.telegesis.ZigBeeDongleTelegesis] - Telegesis RX: TelegesisReceiveMessageEvent [ieeeAddress=null, networkAddress=15678, profileId=260, destinationEp=1, sourceEp=1, clusterId=6, messageData=01 12 40 00 00, rssi=-81, lqi=255]
2018-02-21 22:39:42.469 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=15678/1, destinationAddress=0/1, profile=0104, cluster=6, addressMode=null, radius=0, sequence=0, payload=01 12 40 00 00]
2018-02-21 22:39:42.478 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=CLUSTER_SPECIFIC_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=18, commandId=64]
2018-02-21 22:39:42.488 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - No command type found for CLUSTER_SPECIFIC_COMMAND, cluster=6, command=64, direction=CLIENT_TO_SERVER
2018-02-21 22:39:42.491 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - Incoming message did not translate to command.
```

cdjackson, 2018-02-22:
The remote is using an optional command that is not currently supported by the framework - I'll add this in the coming days.

cdjackson, 2018-02-22:
Oh - please open an issue on the openhab binding as well as there will need to be changes there once the library supports this command.

tavalin, 2018-02-22:
Thanks @cdjackson, will do.

OT - do you happen to have a windows 10 coding environment? I'd like to contribute code rather than Issues but I'm struggling to make my Telegesis stick work with Windows 10.

cdjackson, 2018-02-22:
Thanks. Sorry but I use a Mac. I wouldn't have experience acted problems getting the environment running on windows though.

PR #226 (2018-02-22): Add new commands to OnOff Cluster
Closes #225
closed 2018-02-22 by cdjackson