zsmartsystems/com.zsmartsystems.zigbee #294: ConBee/RaspBee Support

saurabh984, 2018-04-26:
Hi Chris,

The README says more work is required for ConBee and RaspBee. Would you mind creating a list/issue here with details?

I'd like to help!

Thanks
S

cdjackson, 2018-04-26:
Hi @saurabh984 thanks for offering to help. If I remember correctly, it is working ok, but when used for a longer period, or when integrated into other applications, there was an issue where communications stopped. I suspect it is a thread synchronisation issue or something like this. It probably just needs a bit of time, and for someone to pick it up - since I normally use the Ember and do most of my testing with that take time like this I don't see.

I'd suggest if you want to help, then just give it a go - get the console up and running with the ConBee and see what you find. I'm happy to help if you have any questions.

Cheers
Chris

curlyel, 2019-06-11:
Hi Chris,
since I'm still hoping to make use of my Dresden "RaspBEE"-modules (not the ConBEE), I've started to configure a Pi for trying to get your Zigbee console running on it.

First, I've installed Eclipse IDE (with m2e) and Maven on the Pi. Unfortunately I wasn't unable to successfully build the Zigbee packages within the IDE and therefore was'nt able run the Zigbee console from it.

So I've let Maven directly build the project and started the console from the bash command line and gave it a logger config to redirect debug output to the terminal too (see below)
Doing so, I currently get the output of the console as well as the log4j-output. Btw.: passing commands into the Zigbee console is not working yet ...

_Please bear with me since I'm neither a Java developer nor familiar with Eclipse IDE - you might remember this ;-), Therefore, before I pursue a complete wrong road, please advise:_

Is it a valid approach to gather some debug information about the ConBEE/RaspBEE communication issue or is it complete nonsense trying it outside the IDE?

```
pi@RaspBEE-TEST:~/com.zsmartsystems.zigbee/com.zsmartsystems.zigbee.console.main/target $ java -jar com.zsmartsystems.zigbee.console.main-1.2.0-SNAPSHOT.jar -dongle CONBEE -port /dev/ttyS0 -baud 38400 -flow software -reset
Initialising ZigBee console...
13:29:17.275  DEBUG  Default: Set profile to ZigBeeTransactionProfile [maxOutstandingTransactions=2, interTransactionDelay=50, maxRetries=2]
13:29:17.279  DEBUG  Broadcast: Set profile to ZigBeeTransactionProfile [maxOutstandingTransactions=3, interTransactionDelay=1200, maxRetries=0]
13:29:17.280  DEBUG  Multicast: Set profile to ZigBeeTransactionProfile [maxOutstandingTransactions=3, interTransactionDelay=1200, maxRetries=0]
13:29:17.297  ERROR  Error creating network database folder database/CONBEE
13:29:17.330  DEBUG  Adding supported cluster 1280
13:29:17.346  DEBUG  ClusterMatcher starting
13:29:17.348  DEBUG  ClusterMatcher adding cluster 1280
13:29:17.357  DEBUG  Adding supported cluster 25
13:29:17.358  DEBUG  ClusterMatcher adding cluster 25
13:29:17.680  DEBUG  Network state is updated to INITIALISING
13:29:17.680  DEBUG  ConBee transport initialize
ZigBee network state updated to INITIALISING
13:29:17.683  DEBUG  Ignoring invalid transport state transition in ZigBeeNetworkState.INITIALISING by ZigBeeTransportState.UNINITIALISED
13:29:17.685  DEBUG  Opening port /dev/ttyS0 at 38400 baud with FLOWCONTROL_OUT_XONOFF.
13:29:18.031  DEBUG  ConBeeReceiveHandler thread started
13:29:18.055  DEBUG  TX CONBEE queue: 1
13:29:18.058  DEBUG  TX CONBEE queue: 1 sync
13:29:18.059  DEBUG  TX CONBEE queue: 1 send
13:29:18.060  DEBUG  CONBEE TX: DeviceStateRequest [sequence=1]
13:29:18.097  DEBUG  CONBEE TX: 07 01 00 08 00 00 00 00 F0 FF
13:29:18.099  DEBUG  TX CONBEE queue: 1 exit
13:29:18.113  DEBUG  CONBEE RX: Data 07 01 00 08 00 A2 00 1C 32 FF
13:29:18.124  DEBUG  CONBEE RX Frame: DeviceStateResponse [sequence=1, status=SUCCESS, state=DeviceState [networkState=NET_CONNECTED, dataConfirm=false, dataIndication=false, dataRequest=true, configChanged=false]]
13:29:18.126  DEBUG  ConBeeDeviceState=DeviceState [networkState=NET_CONNECTED, dataConfirm=false, dataIndication=false, dataRequest=true, configChanged=false]
13:29:18.126  DEBUG  ConBeeDeviceState-sync
13:29:18.127  DEBUG  ConBeeDeviceState out before
13:29:18.128  DEBUG  ConBeeDeviceState out QueryVersionRequest [sequence=0]
13:29:18.128  DEBUG  CONBEE TX: QueryVersionRequest [sequence=2]
13:29:18.135  DEBUG  CONBEE TX: 0D 02 00 09 00 00 00 00 00 E8 FF
13:29:18.135  DEBUG  ConBeeDeviceState out after
13:29:18.136  DEBUG  ConBeeDeviceState-done
13:29:18.147  DEBUG  CONBEE RX: Data 0D 02 00 09 00 00 05 33 26 8A FF
13:29:18.151  DEBUG  CONBEE RX Frame: QueryVersionResponse [sequence=2, version=640877824]
13:29:18.153  DEBUG  ConBeeDeviceState=DeviceState [networkState=NET_OFFLINE, dataConfirm=false, dataIndication=false, dataRequest=false, configChanged=false]
13:29:18.154  DEBUG  ConBeeDeviceState-sync
13:29:18.155  DEBUG  ConBeeDeviceState-done
13:29:18.161  DEBUG  TX CONBEE queue: 1
13:29:18.162  DEBUG  TX CONBEE queue: 1 sync
13:29:18.164  DEBUG  TX CONBEE queue: 1 send
13:29:18.174  DEBUG  CONBEE TX: DeviceStateRequest [sequence=3]
13:29:18.178  DEBUG  CONBEE TX: 07 03 00 08 00 00 00 00 EE FF
13:29:18.180  DEBUG  TX CONBEE queue: 1 exit
13:29:18.187  DEBUG  CONBEE RX: Data 07 03 00 08 00 A2 00 33 19 FF
13:29:18.189  DEBUG  CONBEE RX Frame: DeviceStateResponse [sequence=3, status=SUCCESS, state=DeviceState [networkState=NET_CONNECTED, dataConfirm=false, dataIndication=false, dataRequest=true, configChanged=false]]
13:29:18.190  DEBUG  ConBeeDeviceState=DeviceState [networkState=NET_CONNECTED, dataConfirm=false, dataIndication=false, dataRequest=true, configChanged=false]
13:29:18.191  DEBUG  ConBeeDeviceState-sync
13:29:18.192  DEBUG  ConBeeDeviceState out before
13:29:18.193  DEBUG  ConBeeDeviceState out ReadParameterRequest [sequence=0, parameter=DEVICE_TYPE]
13:29:18.195  DEBUG  CONBEE TX: ReadParameterRequest [sequence=4, parameter=DEVICE_TYPE]
13:29:18.204  DEBUG  CONBEE TX: 0A 04 00 08 00 01 00 09 E0 FF
13:29:18.205  DEBUG  ConBeeDeviceState out after
13:29:18.206  DEBUG  ConBeeDeviceState-done
13:29:18.212  DEBUG  CONBEE RX: Data 0A 04 00 09 00 02 00 09 01 DD FF
13:29:18.217  DEBUG  CONBEE RX Frame: ReadParameterResponse [sequence=4, status=SUCCESS, parameter=DEVICE_TYPE, value=01]
[... cut ...]
13:29:18.665  DEBUG  TX CONBEE queue: 1 send
13:29:18.667  DEBUG  CONBEE TX: DeviceStateRequest [sequence=17]
13:29:18.677  DEBUG  CONBEE TX: 07 11 00 08 00 00 00 00 E0 FF
13:29:18.679  DEBUG  TX CONBEE queue: 1 exit
13:29:18.680  DEBUG  ConBeeDeviceState=DeviceState [networkState=NET_OFFLINE, dataConfirm=false, dataIndication=false, dataRequest=false, configChanged=false]
13:29:18.682  DEBUG  ConBeeDeviceState-sync
13:29:18.684  DEBUG  ConBeeDeviceState sendFrame!=null
13:29:18.704  DEBUG  CONBEE RX: Data 07 00 08 00 A2 00 1C 22 FF
```

cdjackson, 2019-06-11:
It should be perfectly valid to try an debug using the console and then analysing the logs to understand what is happening, but it is often much more beneficial to step through the program to see what is happening. I think the issues here are not super simple to debug if you are not a Java developer as the likely issue is related to thread synchronisation (so it is likely reasonably simple fix, but can take a little time to understand what is happening). Getting the IDE working is likely to be beneficial.

I’m not sure why passing commands to the console doesn’t work - this should work normally otherwise the console would not be of any use. This is not dongle specific.

If you have any questions about debugging, just ask and I’ll try and help.

Cheers
Chris

curlyel, 2019-06-12:
Hi Chris,
Thanks for the guidance, it encouraged me still trying to get the IDE up ;-)

But: I gave up on doing it on the RaspberryPi directly, since the Eclipse version available for Raspbian-arm32 is probably to old (version 3.8). If I got it right, it does not support Java 1.8. So I guess, I had no chance...

2nd try:
I decided to install the IDE on an Windows system and connected the RaspBEE sitting on the Pi's TTL UART via ser2net to a Windows virtual COM-port. This setup works - So I can run the console from within the IDE now :-)  _And of course: Commands are now working too_ ;-)

e.g.
```
11:19:15.938  DEBUG  CONBEE TX: QuerySendDataRequest [sequence=185]
11:19:15.939  DEBUG  CONBEE TX: 04 B9 00 07 00 00 00 3C FF
11:19:15.939  DEBUG  ConBeeDeviceState-done
11:19:15.998  DEBUG  CONBEE RX: Data 04 B9 00 13 00 0C 00 22 74 02 00 00 00 00 00 00 00 00 00 8C FE
11:19:15.998  DEBUG  CONBEE RX Frame: QuerySendDataResponse [sequence=185, networkState=DeviceState [networkState=NET_CONNECTED, dataConfirm=false, dataIndication=false, dataRequest=true, configChanged=false], destinationAddress=(NWK=0/0), requestId=116]
11:19:15.998  DEBUG  ConBeeDeviceState=DeviceState [networkState=NET_CONNECTED, dataConfirm=false, dataIndication=false, dataRequest=true, configChanged=false]
11:19:15.998  DEBUG  ConBeeDeviceState-sync
11:19:15.998  DEBUG  ConBeeDeviceState-done
nodes
Network  Addr  IEEE Address       Logical Type   EP   Profile                    Device Type      Manufacturer     Model          
      0  0000  00212EFFFF008959   COORDINATOR   
  36044  8CCC  8418260000D98CCF   ROUTER       

node 36044
IEEE Address     : 8418260000D98CCF
Network Address  : 36044
Node Descriptor  : NodeDescriptor [apsFlags=0, bufferSize=64, complexDescriptorAvailable=false, manufacturerCode=48042, logicalType=ROUTER, serverCapabilities=[], incomingTransferSize=0, outgoingTransferSize=0, userDescriptorAvailable=false, frequencyBands=[FREQ_2400_MHZ], macCapabilities=[MAINS_POWER, RECEIVER_ON_WHEN_IDLE, FULL_FUNCTION_DEVICE], extendedEndpointListAvailable=true, extendedSimpleDescriptorListAvailable=true, stackCompliance=0]
Power Descriptor : RECEIVER_ON_IDLE, [MAINS, RECHARGABLE_BATTERY, DISPOSABLE_BATTERY], MAINS, FULL
Associations     : []
Endpoints        : 
Neighbors:
Routes:
 
```
I have to admit, that I was not able to form a new zigbee network through the console so far and bring it to "join enable" to discover an Osram bulb ("join enable" command worked, but did not brought the bulb to join _after resetting it of course_ ;-)) .
Therefore I used DeCONZ to do the join and copied over the PAN and Nwkey to the ZigbeeConsole for the first steps. Is this likely the reason, why no endpoint for the bulb have been discovered?

_For sure, I need more time to understand the right sequence when playing with the console ;-)_

So I have two questions for now:
Can I split/separate/redirect the (extensive ;-)) logging output in the console window from the console input/output?
What is the intended sequence to bring up an Zigbee-PAN, bring it to accept joins and discover clusters supported by the endpoints?

cdjackson, 2019-06-12:
> Is this likely the reason, why no endpoint for the bulb have been discovered?

No - the framework will try to discover endpoints if they are unknown at startup. You will have to tell the system that the device exists, or maybe it will occur automatically simply by power cycling the bulb if the device sends an ANNOUNCE message (which it should).

> Can I split/separate/redirect the (extensive ;-)) logging output in the console window from the console input/output?

Yes, you can adjust the logging in the log4j.xml file and set up different files etc if you prefer.

> What is the intended sequence to bring up an Zigbee-PAN,

From the command line, set the PAN / EPAN etc, and use the reset parameter to initialise the network. There are other ways, but this is the easiest.

- bring it to accept joins

Use the join command

- discover clusters supported by the endpoints?

This is automatic.

curlyel, 2019-06-12:
Chris,
after some minutes, the ZigbeeConsole stalles with a couple of threads waiting for `Thread [pool-2-thread-3]`

I've suspended this thread and took the screenshot below. I started digging into the variables/objects - but don't know where to look after ;-)

[image]

curlyel, 2019-06-12:
> No - the framework will try to discover endpoints if they are unknown at startup. You will have to tell the system that the device exists, or maybe it will occur automatically simply by power cycling the bulb if the device sends an ANNOUNCE message (which it should).
 
Ok. did this (a couple of times). Thought it was more complicated ;-) I'm going to try another device...
And, thanks for the logging hint too...

cdjackson, 2019-06-12:
> after some minutes, the ZigbeeConsole stalles

This is the problem that we need to resolve. You need to spend some time to work out why this is happening...

curlyel, 2019-06-12:
Sorry Chris,
I'm afraid, I'm not really in the position yet to work out something... :-(

Yes, the console is talking (RX _and_ TX) from/to the ConBEE but I think we **do not get any packets** back from the Zigbee network. The zigbee library tries to discover the endpoints etc. as you said, but all the discoveries end up with timeout:
`-dongle CONBEE -port COM9 -baud 38400 -flow software -channel 25 -pan 0xd824  -epan 1 -nwkkey 0xcccccccccccccccccccccccccccccccc -linkkey 0x<deleted> -reset`
```
15:21:53.859  DEBUG  8418260000D98CCF: Node SVC Discovery: ActiveEndpointsResponse returned CommandResult [TIMEOUT]
15:21:53.859  DEBUG  8418260000D98CCF: Node SVC Discovery: request ACTIVE_ENDPOINTS failed. Retry 2, wait 2329ms before retry.
15:21:54.784  DEBUG  00212EFFFF008959: Node SVC Discovery: ManagementRoutingRequest returned CommandResult [TIMEOUT]
15:21:54.784  DEBUG  00212EFFFF008959: Node SVC Discovery: request ROUTES failed. Retry 2, wait 4450ms before retry.
15:21:54.877  DEBUG  001788011051E209: Node SVC Discovery: ActiveEndpointsResponse returned CommandResult [TIMEOUT]
15:21:54.877  DEBUG  001788011051E209: Node SVC Discovery: request ACTIVE_ENDPOINTS failed. Retry 11, wait 9280ms before retry.
....
15:21:56.190  DEBUG  8418260000D98CCF: Node SVC Discovery: running ACTIVE_ENDPOINTS
15:21:56.190  DEBUG  8418260000D98CCF: Added transaction to queue, len=2, transaction=ZigBeeTransaction [queueTime=0, state=WAITING, se
15:22:12.853  DEBUG  8418260000D98CCF: Node SVC Discovery: running ACTIVE_ENDPOINTS
15:22:26.448  DEBUG  DISCOVERY Extension: Starting mesh update
15:22:26.448  DEBUG  DISCOVERY Extension: Starting mesh update for 001788011051E209
15:22:26.448  DEBUG  001788011051E209: Node SVC Discovery: Update mesh
15:22:26.448  DEBUG  001788011051E209: Node SVC Discovery: has no tasks to perform
15:22:26.448  DEBUG  DISCOVERY Extension: Starting mesh update for 8418260000D98CCF
15:22:26.448  DEBUG  8418260000D98CCF: Node SVC Discovery: Update mesh
15:22:26.448  DEBUG  8418260000D98CCF: Node SVC Discovery: already scheduled or running
15:22:26.448  DEBUG  8418260000D98CCF: Node SVC Discovery: scheduled [ACTIVE_ENDPOINTS, NWK_ADDRESS, NEIGHBORS, ROUTES]
15:22:26.448  DEBUG  DISCOVERY Extension: Starting mesh update for 00212EFFFF008959
15:22:26.448  DEBUG  00212EFFFF008959: Node SVC Discovery: Update mesh
15:22:26.448  DEBUG  00212EFFFF008959: Node SVC Discovery: already scheduled or running
15:22:26.448  DEBUG  00212EFFFF008959: Node SVC Discovery: scheduled [ROUTES, NEIGHBORS]
15:22:26.764  DEBUG  8418260000D98CCF: Node SVC Discovery: running ACTIVE_ENDPOINTS
15:22:28.595  DEBUG  00212EFFFF008959: Node SVC Discovery: running ROUTES
```
To check if it is dongle dependend, I digged out some CC2531 and started the console against it _(directly plugged into the Windows PC not the Pi)_:

`-dongle CC2531 -port COM3 -baud 9600 -flow software -pan 0x1234 -epan 987654321 -reset`

```
nodes
Network  Addr  IEEE Address       Logical Type   EP   Profile                    Device Type      Manufacturer     Model          
      0  0000  00124B0009EB11FC   UNKNOWN       

> 15:43:50.933  DEBUG  Transaction complete: ZigBeeTransaction [queueTime=10052, state=FAILED, sendCnt=1, command=PowerDescriptorRequest [0/0 -> 0/0, cluster=0003, TID=1F, nwkAddrOfInterest=0]]
15:43:50.933  DEBUG  transactionListenerRemoved: 0 outstanding
15:43:50.933  DEBUG  00124B0009EB11FC: transactionComplete FAILED 0
15:43:50.933  DEBUG  00124B0009EB11FC: Added transaction to queue, len=1, transaction=ZigBeeTransaction [queueTime=10052, state=WAITING, sendCnt=1, command=PowerDescriptorRequest [0/0 -> 0/0, cluster=0003, TID=1F, nwkAddrOfInterest=0]]
15:43:52.892  DEBUG  00124B0009EB11FC: Node SVC Discovery: PowerDescriptorResponse returned null
15:43:52.892  DEBUG  00124B0009EB11FC: Node SVC Discovery: request POWER_DESCRIPTOR failed after 13 attempts.
15:43:52.893  DEBUG  00124B0009EB11FC: Node SVC Discovery: complete
15:43:52.893  DEBUG  00124B0009EB11FC: Node 0 update
15:43:52.895  DEBUG  00124B0009EB11FC: Set profile to ZigBeeTransactionProfile [maxOutstandingTransactions=1, interTransactionDelay=50, maxRetries=2]
... <restart>...
15:50:50.972  DEBUG  CC2531 transport initialize
15:50:50.972  DEBUG  Opening port COM3 at 9600 baud with FLOWCONTROL_OUT_XONOFF.
15:50:50.996  DEBUG  ->  SYS_RESET (Packet: subsystem=null, length=1, apiId=41 00, data=FE 01 41 00 01 41, checksum=41, error=false) 
15:50:53.340  DEBUG  <-- SYS_RESET_RESPONSE (FE 06 41 80 00 02 00 02 06 03 C2)
15:50:53.340  DEBUG  Received Async Cmd: Packet: subsystem=null, length=6, apiId=41 80, data=FE 06 41 80 00 02 00 02 06 03 C2, checksum=C2, error=false
15:50:53.342  DEBUG  ->  SYS_VERSION (Packet: subsystem=null, length=0, apiId=21 02, data=FE 00 21 02 23, checksum=23, error=false) 
15:50:53.347  DEBUG  <-  SYS_VERSION_RESPONSE (FE 05 61 02 02 00 02 06 03 63)
15:50:53.347  DEBUG  CC2531 version is Software=2.6 Product=0 Hardware=3 Transport=2
15:50:53.348  DEBUG  ->  ZB_GET_DEVICE_INFO (Packet: subsystem=null, length=1, apiId=26 06, data=FE 01 26 06 01 20, checksum=20, error=false) 
15:50:53.351  DEBUG  <-  ZB_GET_DEVICE_INFO_RSP (FE 09 66 06 01 FC 11 EB 09 00 4B 12 00 3E)
networkManager.initialize returned SUCCESS
15:50:53.352  DEBUG  ->  ZB_GET_DEVICE_INFO (Packet: subsystem=null, length=1, apiId=26 06, data=FE 01 26 06 06 27, checksum=27, error=false) 
15:50:53.356  DEBUG  <-  ZB_GET_DEVICE_INFO_RSP (FE 09 66 06 06 00 20 EB 09 00 4B 12 00 F4)
[... cut ...]
15:50:53.413  DEBUG  <-  ZToolPacket (FE 01 61 14 03 77)
15:50:53.414  DEBUG  Unsupported configuration option "TRUST_CENTRE_LINK_KEY" in CC2531 dongle
15:50:53.414  DEBUG  CC2531 transport startup
```
Even with the CC2531 I'm not able to join resetted bulbs from Osram and Philips.
Honestly, I feel a bit lost :-(

Meanwhile I tried the master branch as well as the release-1.10.
Firmware on the RaspBEE is version 26330500.

Since both dongles are not working as expected, I'd assume some flaw on my side. Any hints how to proceed?

cdjackson, 2019-06-12:
> Even with the CC2531 I'm not able to join resetted bulbs from Osram and Philips.

Both the ConBee and the CC2531 drivers work. The ConBee is not reliable due to the issue you identify above, and that is what needs to be resolved. The CC2531 is perfectly reliable and used extensively - both by users of openHAB, and some customers running under Android and also Java.

> Since both dongles are not working as expected, I'd assume some flaw on my side

Yes, this is likely the case. Unfortunately I'm not really sure what to suggest. I test with the 2531 quite often and it should work fine. I don't really see anything wrong with the log above either.

curlyel, 2019-06-12:
> The CC2531 is perfectly reliable and used extensively - ... by users of openHAB

Yes, I know. I am one of them. I do have CC2531 running in two openHAB instances too.
For that I'm very grateful to you, Chris!

> don't really see anything wrong with the log above either.
Shouldn't we see some reply packets as response to the discoveries (endpoint, routes etc.)?

Once I start the Dresden software, I can immediately let join bulbs and control them. The CC2531 was in use on openHAB too. Therefore I assume the hardware is working fine in both cases. 

The Zigbee console is "talking" to the dongles. But somehow, my Zigbee PAN created with it, seems odd. Could you please have a last look on my command line parameters?
Did I made some wrong initialization?
```
-dongle CONBEE -port COM9 -baud 38400 -flow software -channel 25 -pan 0xd824 -epan 1 -nwkkey 0xcccccccccccccccccccccccccccccccc -linkkey 0x<deleted> -reset
```
and for the CC2531
```
-dongle CC2531 -port COM3 -baud 9600 -flow software -pan 0x1234 -epan 987654321 -reset
```

Thanks for your patience and sorry for not being a big help here ;-)

cdjackson, 2019-06-12:
> I know. I am one of them. 

Why are you looking at the ConBee then as a matter of interest? Is it just that you are looking for a solution to work "neatly" with the RPi? If so, I may soon be able to supply an Ember solution using the latest Ember Chipset (the EFR32) if that's of interest?

> Shouldn't we see some reply packets as response to the discoveries (endpoint, routes etc.)?

Well, there are no packets being sent. The log doesn't provide that information. The first log doesn't have that, and the second log looks like it's part of the startup, and is only doing the initialisation.

Even if frames are sent, the fact that there are no responses doesn't make it an error with the software. Of course we expect replies in this case, but not receiving a reply is also valid and it's hard to know what happens without a lot more information (eg sniffer logs).

> But somehow, my Zigbee PAN created with it, seems odd.

What looks odd? I don't immediately see anything wrong...

> Did I made some wrong initialization?

No - it looks ok. Normally the 2351 uses 38400 baud rate I think, but as it is communicating, it probably doesn't matter as it's a direct USB link.

curlyel, 2019-06-13:
> Is it just that you are looking for a solution to work "neatly" with the RPi?

_Yes and no. ;-)_
I do have the RaspBEE modules here and _would_ give them a try due to the expectation, that they have better reception sensitivity and more transmit power - and yes - will fit into Rpi.

BUT: I'm not in need to get it work, since I already have a reliable solution with the CC2531's already (as you know). Again: I'm a happy user of your Zigbee as well as of the ZWave binding ;-)

The main reason for digging out this issue was my thought it would be a chance giving something back to the community. Since I'm not a Java developer, I cannot contribute with coding etc ...
But I have plenty of zigbee hardware and a rough understanding how it should work together - So I was hoping I could drive some long-run tests to help isolating the cause ;-)

> What looks odd?

I meant: I wasn't able to join any device to my Zigbee networks created with the Zigbee console so I assumed that I have created some oddity in my test Zigbee networks ;-)

cdjackson, 2019-06-13:
> I do have the RaspBEE modules here and would give them a try due to the expectation, that they have better reception sensitivity and more transmit power - and yes - will fit into Rpi.

I would treat this with some level of doubt. The problem with sensitivity specifications is it needs to be against a specific BER or PER. I'm not sure what this is for Atmel - I know what it is for the Silabs chips as they quote with a 1% PER which is very good (this means only 1 packet in 100 is lost). Others may quote at 20% PER - I'm not sure about Atmel. I also note that the numbers quoted in the ConBee spec are considerably better than quoted by Atmel, which seems strange given that the ConBee/RaspBee is simply this single Atmel chip.

So, please be a little careful about just looking at the numbers.  It's also worth noting that RF power is generally limited by law (eg CE requirements).

>  So I was hoping I could drive some long-run tests to help isolating the cause ;-)

That's great that you want to help. Unfortunately it also needs someone with coding skills to look into this I think as it's a low level problem that probably won't be solved just by looking at logs. I do have these devices and wrote the initial driver for it, but nearly all commercial customers use the Silabs chipset, and some are using the TI (although TI told me recently they are not planning any further updates to their stack so I would not recommend that in future).

> I meant: I wasn't able to join any device to my Zigbee networks created with the Zigbee console so I assumed that I have created some oddity in my test Zigbee networks ;-)

Ok, agreed - it doesn't work, but my point is there is nothing abnormal in the logs you provided. The logs don't show any joining attempts, or even the general operation of the stack as it is still in the initialisation phase. So, while it's not working, there's nothing actually wrong anywhere - based on the logs I see at least.