zsmartsystems/com.zsmartsystems.zigbee #296: License goal not working

cdjackson, 2018-04-27:
@hsudbrock the change you made to the pom seems to complete stop the license formatter... 

Any thoughts on how to resolve this so it works in both scenarios?

hsudbrock, 2018-04-27:
I just had a deeper look into the mycila license-maven-plugin; the problem here appears to be the interplay between its `aggregate` setting (which I introduced with my change in the pom), and its `includes` setting.

Context: According to the plugin doc for the `aggregate` flag: "You can set this flag to true if you want to check the headers for all modules of your project. Only used for multi-modules projects, to check for example the header licenses from the parent module for all sub modules."

However, the `includes` setting appears to require path patterns relative to the parent pom, not relative to the submodule poms (which differs from the setting with `aggregate` being set to false, where the inherited configuration will interpret the path patterns in the `include` property relative to the submodule poms).

So what will solve this bug is to specify the `include` pattern relative to the parent pom, I'll add a PR for that.

With this change, `mvn com.mycila:license-maven-plugin:format` and `com.mycila:license-maven-plugin:check` should both work as expected (tested on my machine).

cdjackson, 2018-04-28:
Closed by #297

closed 2018-04-28 by cdjackson