zsmartsystems/com.zsmartsystems.zigbee #307: Raspbee + deConz + OpenHAB-Binding

TwE3d, 2018-05-17:
Hello,

right now we are trying to get the Raspbee-Gateway working. We set up DeConz on the Raspbee and the Openhab-Server but we ran into an issue using the Zigbee-Binding. 
If we go to the OpenHab-Site and go to "Choose Binding" there are only three options and Raspbee/Conbee is not included. 
Now we are trying to use the console to get the Zigbee Module working.
To us it just isnt clear how to use the console. 
As stated in the "console application" paragraph there is a command to start the console so the framework is ready to go. Do we have to copy it to the shell? To us its also not rly clear how the syntax of the command has to look like. Can someone give an example? 

Your help would be greatly appreciated.
Tim

saurabh984, 2018-05-17:
Hi Tim,

As I understand the ZigBee binding is work in progress and not yet ready. 

If you’re using deCONZ then you’ll need to use Hue Binding. 

deCONZ REST API is compliant or based on the HUE API.

Regards
S

cdjackson, 2018-05-17:
I'm a little confused - what does deCONZ have to do with this repository?

For that matter, openHAB isn't really related to this repository. This repo is solely about the ZigBee libraries.

Anyway, to address some points.

> As I understand the ZigBee binding is work in progress and not yet ready.

That's not correct - well - as with any binding, there is work to do, but it is available for use.

> If you’re using deCONZ then you’ll need to use Hue Binding.

Unsure what this has to do with this repo?

> To us it just isnt clear how to use the console.

The console is a java application, so it needs to be run as a java application. I normally do this from within the Eclipse IDE as it's normally used for debugging. You can also run Java applications from the command line if you prefer though, but as I understand you want to debug, I would suggest to run it from the IDE. The command line arguments are shown in the readme.

Sorry if I've not answered the questions - there's a lot of confusing and conflicting comments here so if this is really related to deCONZ, then we should just close this issue as this repo is in no way linked to the Dresden software.

saurabh984, 2018-05-18:
@cdjackson Apologies Chris, I meant to say support for ConBee/RaspBee Binding is work in progress ;-)

Working on it but no progress yet!