zsmartsystems/com.zsmartsystems.zigbee #319: Provide timeout handling for Ember dongle

TomTravaglino, 2018-06-08:
The current implementation for the Ember dongle doesn't recognize when the controller has stopped communicating. The implementation of such a feature is under progress for the Telegesis dongle. Please, provided it also for the Ember dongle.

triller-telekom, 2018-10-30:
Since https://github.com/openhab/org.openhab.binding.zigbee/issues/285 was closed with a link to this issue I would like to understand where exactly the problem is.

My questions are:
- Is it at all possible to figure out that the communication is broken?
- Where in this zsmartsystems library is the code for telegesis and why does nit not work for ember chips?

cdjackson, 2018-10-30:
The different drivers (Telegesis/Ember) each need to monitor their communications. I added this for Telegesis, but it has not been done for Ember yet. Of course it's possible to do - it is just a case of monitoring communications regularly (same as I implemented for Telegesis) and updating the state (which would then set the bridge offline in openHAB).

triller-telekom, 2018-10-30:
Thank you for the clarification.

From a users perspective if I switch from a broken Ember stick to a newly bought one I would like to see my old bridge as offline if it cannot communicate anymore. Otherwise I would have 2 bridges in state ONLINE.

I a backup/restore scenario this can become tricky if the broken bridge is still ONLINE, so the system has a hard time figuring out which device is the working one for restoring the backup to.

cdjackson, 2018-10-30:
> I would like to see my old bridge as offline if it cannot communicate anymore.

That is the purpose of this issue.

> I a backup/restore scenario this can become tricky if the broken bridge is still ONLINE, so the system has a hard time figuring out which device is the working one for restoring the backup to.

I don't understand this point? I don't think it's related, but I'm not sure I understand what you mean anyway?

triller-telekom, 2018-10-30:
I was referring to a possible solution based on openHAB/ESH that backups the state of all bridges/devices and restores it at some point. But that is off topic for this issue, its just to highlight that it is important to know which bridge is working (ONLINE) and which is not (OFFLINE).

Are you planning to implement the telegesis fix for ember anytime soon?

cdjackson, 2018-10-30:
Restore is quite complex as you need to migrate the TC. This has been implemented and tested recently, but you would need to take the other dongle out of the system first anyway.

At the moment this is not high on my todo list and it was not considered important during recent discussions.

PR #464 (2018-12-14): EmberStick using EZSP protocol goes offline/online if stick is re-plugged
Implementation is based on the one for the Telegesis, see

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/322

It is implementes in the EZSP layer, so it should work for dongles using SPI or ASH

Fixes #319
commit 3b01f525 (2018-12-14): EmberStick using EZSP protocol goes offline/online if stick is re-plugged (#464)

* EmberStick using EZSP protocol goes offline/online if stick is re-plugged

Implementation is based on the one for the Telegesis, see

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/322

Fixes #319

* Revert bugfix for retry in ASH Layer

* Revert changes on NetworkManager
closed 2018-12-14 by cdjackson