zsmartsystems/com.zsmartsystems.zigbee #328: Ember console uses classes from internal Ember dongle packages

hsudbrock, 2018-06-20:
Classes from the module `com.zsmartsystems.zigbee.console.ember` use classes from internal packages from the module `com.zsmartsystems.zigbee.dongle.ember` (viz. the package `com.zsmartsystems.zigbee.dongle.ember.internal.ezsp` and sub-packages).

This leads to problems (in particular in the context of this [issue](https://github.com/openhab/org.openhab.binding.zigbee/issues/219)) when using these modules as OSGi bundles, because the internal packages are not declared as exported in the manifest file that is generated by the maven build (as the maven build adds export statements for all bundles, unless bundles that contain `internal` or `impl` - cf. the [maven plugin docs](http://felix.apache.org/documentation/subprojects/apache-felix-maven-bundle-plugin-bnd.html)).

The simplest approach for solving this would be to simply configure the maven build to make the generated manifest export all bundles for the ember dongle module. This could by done by adding the following to the pom file of that module:
```
<build>
   <plugins>
    <plugin>
        <groupId>org.apache.felix</groupId>
         <artifactId>maven-bundle-plugin</artifactId>
         <configuration>
           <instructions>
             <Export-Package>com.zsmartsystems.zigbee.dongle.ember.*</Export-Package>
           </instructions>
         </configuration>
    </plugin>
   </plugins>
 </build>
```

Exporting only packages without `internal`/`impl` except the affected packages appears to be more difficult to me.

Another option would be to rewrite the ember console / the ember dongle modules such that imports of internal classes are no longer needed - this option also seems not so easy.

hsudbrock, 2018-06-20:
I, personally, could live with the simple option by exporting all packages (with a heavy heart though). What do you think, @cdjackson ?

cdjackson, 2018-06-20:
Yes, I came across this issue when creating the ember console. I don't think there's any easy way around this since the console needs access to the low level commands in Ember - by virtue of the fact that it's working at the lowest level with the Ember NCP. 

I think that the best way around this is to move these packaged out of the internal package and they should then be exported in the manifest.

What do you think? I think it should work ok for you?

hsudbrock, 2018-06-21:
Yes, I think that this should work ok for me. And I think that it is a better solution than simply exporting all packages. Let me try to make that change on my local machine and see if it really works out, and if yes, I would create a PR here.

hsudbrock, 2018-06-21:
I have seen in the comments in several of the classes in the affected packages that the code is autogenerated, and manual changes would be overwritten. However, I could not find the code generator (the folder `com.zsmartsystems.zigbee.dongle.ember.autocode` is empty). Does the code generator need to be involved in the move of the packages?

cdjackson, 2018-06-21:
Yes - I will need to update this in the generator. I'll move these all down a level (ie out of ```.internal```)

hsudbrock, 2018-06-21:
So there's nothing more to do than updating the generator - correct?

cdjackson, 2018-06-21:
And the console bundles I guess as that will import the classes...

Leave it with me and I'll create a PR and you can take a look...

hsudbrock, 2018-06-21:
Thanks @cdjackson !

PR #329 (2018-06-26): Refactor ember packages to export EZSP commands and structures
@hsudbrock please take a look at this - fundamentally it moves all the Ember structures and commands out of the internal package so I think should fix your issue.

Fixes #328
closed 2018-06-26 by cdjackson