zsmartsystems/com.zsmartsystems.zigbee #333: Test in SpiFrameHandlerTest failing

TomTravaglino, 2018-06-28:
Version 1.0.13

The test mentioned above fails very often during the build.

`12:44:33 testSendEzspFrame(com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandlerTest)  Time elapsed: 0.024 sec  <<< FAILURE!
12:44:33 java.lang.AssertionError: expected:<7> but was:<9>
12:44:33 	at org.junit.Assert.fail(Assert.java:88)
12:44:33 	at org.junit.Assert.failNotEquals(Assert.java:743)
12:44:33 	at org.junit.Assert.assertEquals(Assert.java:118)
12:44:33 	at org.junit.Assert.assertEquals(Assert.java:555)
12:44:33 	at org.junit.Assert.assertEquals(Assert.java:542)
12:44:33 	at com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandlerTest.testSendEzspFrame(SpiFrameHandlerTest.java:279)
12:44:33 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
12:44:33 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
12:44:33 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
12:44:33 	at java.lang.reflect.Method.invoke(Method.java:498)
`

hsudbrock, 2018-06-29:
Hm, I ran this test about 250 times on my machine, and it never failed. What do you mean with "fails very often", @TomTravaglino ?

cdjackson, 2018-06-29:
I can see how this might fail on a slow computer as there will be other frames sent after 250ms. This has not changed recently so maybe the environment has changed?

Possibly the driver needs refactoring slightly to remove this ambiguity to improve the testability...

TomTravaglino, 2018-06-29:
@hsudbrock @cdjackson 
it means that since a couple of days our CI job permanently fails. On my local machine it runs fine.

cdjackson, 2018-06-29:
I suspect that the CI has changed then? Maybe other things are running on it or something and this is slowing down the test.

I will take a look at refactoring to remove this issue.

cdjackson, 2019-02-16:
@TomTravaglino where do we take this? Are you able to provide a PR on the tests, or shall we close this?

TomTravaglino, 2019-02-22:
Meanwhile, the test is running. We can close this.

closed 2019-02-22 by TomTravaglino