zsmartsystems/com.zsmartsystems.zigbee #338: Provide default values for cluster if Values are not properly formated

zalke, 2018-07-02:
I'm getting the following exception every few seconds, 

```java.lang.ClassCastException: java.lang.String cannot be cast to java.lang.Integer
        at com.zsmartsystems.zigbee.zcl.clusters.ZclBasicCluster.getHwVersion(ZclBasicCluster.java:327) [193:com.zsmartsystems.zigbee:1.0.11]
        at org.openhab.binding.zigbee.discovery.ZigBeeNodePropertyDiscoverer.getProperties(ZigBeeNodePropertyDiscoverer.java:138) [247:org.openhab.binding.zigbee:2.4.0.201807010828]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.doNodeInitialisation(ZigBeeThingHandler.java:254) [247:org.openhab.binding.zigbee:2.4.0.201807010828]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.access$0(ZigBeeThingHandler.java:158) [247:org.openhab.binding.zigbee:2.4.0.201807010828]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler$1.call(ZigBeeThingHandler.java:152) [247:org.openhab.binding.zigbee:2.4.0.201807010828]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler$1.call(ZigBeeThingHandler.java:1) [247:org.openhab.binding.zigbee:2.4.0.201807010828]
        at java.util.concurrent.FutureTask.run(FutureTask.java:266) [?:?]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180) [?:?]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293) [?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149) [?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624) [?:?]
        at java.lang.Thread.run(Thread.java:748) [?:?]
```

looks like its expecting an Integer for the hardware version, but the device returns a string, since the hardware version is not critical for device operation how about we surround querying those parameters in a try/catch and provide default values in case there is an exception?

cdjackson, 2018-07-03:
What is the device? It seems it doesn't conform to the ZigBee standard!

The framework has the ability to transform data from non-compliant devices, but I'm not sure it will work in all instances. I will take a look at the log and try and provide some handling, but it may not report correctly.

zalke, 2018-07-03:
Have quite a few devices in this setup, but I think its the one we worked on it [here](https://community.openhab.org/t/zigbee-binding/15763/436?u=solomon_m).

Thanks so much Chris for your great work.

cdjackson, 2018-07-03:
Unfortunately the log doesn't have the actual ZigBee requests. This data is cached in the network XML file so it doesn't get requested again.

Can you provide the network XML - in OH it's stored in the {userdata}/zigbee folder.

zalke, 2018-07-03:
file attached
[network.zip](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/2160809/network.zip)

zalke, 2018-07-03:
see attached hope its enough
[log.zip](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/2161267/log.zip)

cdjackson, 2018-08-01:
Unfortunately the log did not contain the information I needed, but I did make a change a week or so back to try and resolve this. It should now (or very soon) be included in OH, so please advise if the issue is resolved.

cdjackson, 2018-10-14:
An update was made to resolve this - if it's still not working, then please reopen and provide an updated log.

zalke, 2018-12-14:
I just checked this issue and its still not resolved,  see the log
```
2018-12-14 17:02:40.146 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=0/0, profile=0000, cluster=54, addressMode=null, radius=0, apsSecurity=false, apsCounter=206, payload=00 3C 01]
2018-12-14 17:02:40.147 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ManagementPermitJoiningRequest [0/0 -> 0/0, cluster=0036, TID=NULL, permitDuration=60, tcSignificance=true]
2018-12-14 17:02:40.151 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=1, notRdy=false]
2018-12-14 17:02:40.306 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=1, ackNum=3, reTx=false, data=78 80 34 00 CF]
2018-12-14 17:02:40.307 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - ASH: Frame acked and removed AshFrameData [frmNum=2, ackNum=0, reTx=false, data=78 00 34 00 B3 DA 04 01 00 00 01 01 40 11 00 00 78 78 05 00 78 00 03 00]
2018-12-14 17:02:40.308 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - RX EZSP: EzspSendUnicastResponse [status=EMBER_SUCCESS, sequence=207]
2018-12-14 17:02:40.309 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX: EzspSendUnicastResponse [status=EMBER_SUCCESS, sequence=207]
2018-12-14 17:02:40.309 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - Unhandled EZSP Frame: EzspSendUnicastResponse [status=EMBER_SUCCESS, sequence=207]
2018-12-14 17:02:40.310 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=2, notRdy=false]
2018-12-14 17:02:40.311 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - TX EZSP: EzspSendUnicastRequest [type=EMBER_OUTGOING_DIRECT, indexOrDestination=6830, apsFrame=EmberApsFrame [profileId=260, clusterId=0, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY, EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY], groupId=0, sequence=122], messageTag=122, messageContents=00 7A 00 03 00]
2018-12-14 17:02:40.312 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameData [frmNum=3, ackNum=2, reTx=false, data=7A 00 34 00 AE 1A 04 01 00 00 01 01 40 11 00 00 7A 7A 05 00 7A 00 03 00]
2018-12-14 17:02:40.314 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=2, ackNum=3, reTx=false, data=78 90 3F 00 B3 DA 04 01 00 00 01 01 40 11 00 00 CF 78 00 00]
2018-12-14 17:02:40.315 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - RX EZSP: EzspMessageSentHandler [type=EMBER_OUTGOING_DIRECT, indexOrDestination=55987, apsFrame=EmberApsFrame [profileId=260, clusterId=0, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY, EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY], groupId=0, sequence=207], messageTag=120, status=EMBER_SUCCESS, messageContents=]
2018-12-14 17:02:40.316 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX: EzspMessageSentHandler [type=EMBER_OUTGOING_DIRECT, indexOrDestination=55987, apsFrame=EmberApsFrame [profileId=260, clusterId=0, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY, EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY], groupId=0, sequence=207], messageTag=120, status=EMBER_SUCCESS, messageContents=]
2018-12-14 17:02:40.316 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - Unhandled EZSP Frame: EzspMessageSentHandler [type=EMBER_OUTGOING_DIRECT, indexOrDestination=55987, apsFrame=EmberApsFrame [profileId=260, clusterId=0, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY, EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY], groupId=0, sequence=207], messageTag=120, status=EMBER_SUCCESS, messageContents=]
2018-12-14 17:02:40.317 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=3, notRdy=false]
2018-12-14 17:02:40.319 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=3, ackNum=3, reTx=false, data=78 90 45 00 04 01 00 00 01 01 00 01 00 00 6A FE A9 B3 DA FF FF 0A 18 78 01 03 00 00 42 02 01 04]
2018-12-14 17:02:40.320 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - RX EZSP: EzspIncomingMessageHandler [type=EMBER_INCOMING_UNICAST, apsFrame=EmberApsFrame [profileId=260, clusterId=0, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY], groupId=0, sequence=106], lastHopLqi=254, lastHopRssi=-87, sender=55987, bindingIndex=255, addressIndex=255, messageContents=18 78 01 03 00 00 42 02 01 04]
2018-12-14 17:02:40.321 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX: EzspIncomingMessageHandler [type=EMBER_INCOMING_UNICAST, apsFrame=EmberApsFrame [profileId=260, clusterId=0, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY], groupId=0, sequence=106], lastHopLqi=254, lastHopRssi=-87, sender=55987, bindingIndex=255, addressIndex=255, messageContents=18 78 01 03 00 00 42 02 01 04]
2018-12-14 17:02:40.322 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=55987/1, destinationAddress=0/1, profile=0104, cluster=0, addressMode=null, radius=0, apsSecurity=false, apsCounter=106, payload=18 78 01 03 00 00 42 02 01 04]
[... cut ...]
2018-12-14 17:02:41.049 [DEBUG] [scovery.ZigBeeNodePropertyDiscoverer] - 00124B000767EF91: ZigBee node property discovery complete: {zigbee_logicaltype=ROUTER, zigbee_powerlevel=FULL, modelId=abb71ca5fe1846f185cfbda554046cce, vendor=???, zigbee_networkaddress=45879, zigbee_powermode=RECEIVER_ON_IDLE, zigbee_powersources=[MAINS], zigbee_powersource=MAINS, hardwareVersion=1, zigbee_datecode=2017-02-28      , zigbee_zclversion=1}
2018-12-14 17:02:41.050 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=6, ackNum=6, reTx=false, data=7B 90 3F 04 FC FF 00 00 36 00 00 00 08 01 00 00 CE 77 00 00]
```

cdjackson, 2018-12-14:
So the issue here is a bigger problem... The device is reporting that it is sending STRING data, when the data should be INTEGER type (from the cluster library), and the data IS an INTEGER. So probably the data is right, but the data-type that the device is saying it is sending is wrong!

The device is sending the following -:

```
42 02 01 04
```

The ```42``` is the data type (STRING), the ```02```` is the length, and the ```01 04``` is the data (clearly NOT a STRING).

I will have a think about how to handle this situation. The current system was designed to fix the case where the device was sending data that was the wrong type compared to the ZCL, but the type of data it said it was sending, and the data it is sending, are assumed to be consistent!

cdjackson, 2018-12-14:
Just pulling out the received data in question for future reference -:

```
2018-12-14 17:02:40.322 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=55987/1, destinationAddress=0/1, profile=0104, cluster=0, addressMode=null, radius=0, apsSecurity=false, apsCounter=106, payload=18 78 01 03 00 00 42 02 01 04]
2018-12-14 17:02:40.323 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=SERVER_TO_CLIENT, disableDefaultResponse=true, manufacturerCode=0, sequenceNumber=120, commandId=1]
2018-12-14 17:02:40.324 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ReadAttributesResponse [Basic: 55987/1 -> 0/1, cluster=0000, TID=78, records=[ReadAttributeStatusRecord [attributeDataType=CHARACTER_STRING, attributeIdentifier=3, status=SUCCESS, attributeValue=��]]]
```

zalke, 2018-12-14:
Just my two cents.

As you said the data is correct but just the type description is wrong i guess we should try to parse the data to the data type it supposed to be, and if it fails provide default values.

Or since its only related to this specific device maybe we should use static configuration via txt file “if its available to the end user” as in 
https://github.com/openhab/org.openhab.binding.zigbee/pull/169

Either way since the property is there for information only why not surround it with a try/catch so it doesnt fail?

cdjackson, 2018-12-14:
Please remember that the library here has to cope with ALL devices, and provide a reasonable reaction for ALL clusters/attributes. So, the fact the this property is only for information, and the fact that this issue is only related to a single device, doesn't really help.

However, I will look at providing a default value and logging an error - I've partly coded this already so hope to have a fix in the coming day or two.

closed 2018-12-15 by cdjackson

zalke, 2018-12-16:
Thanks much chris

Looks like its not in the lates snapshot yet, any idea when it will be pulled in?

cdjackson, 2018-12-16:
It will likely be a few months now before it goes into the openhab binding, but you can use it without the binding update by using the script from @5iver which is posted on the forum (somewhere).

Before that script will work I need to do another release, which will likely be tonight or tomorrow.

zalke, 2018-12-17:
Thanks so much 
looks like its working good now

for the record here is the script
https://github.com/openhab-5iver/openHAB-utils

cdjackson, 2018-12-17:
Thanks for confirming @zalke

PR #467 (2018-12-15): Add default value if the normaliser fails to normalise the data type
This adds a facility to provide a default value if the attribute normalizer throws an exception.
Closes #338