zsmartsystems/com.zsmartsystems.zigbee #340: AshFrameHandlerTest#testAshFrameData fails continuously

Xrygramming, 2018-07-09:
With the latest version (1.0.14), the _AshFrameHandlerTest#testAshFrameData_ continuously fails as follows:

`14:34:24 testAshFrameData(com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandlerTest)  Time elapsed: 0.008 sec  <<< FAILURE!
14:34:24 java.lang.AssertionError
14:34:24 	at org.junit.Assert.fail(Assert.java:86)
14:34:24 	at org.junit.Assert.assertTrue(Assert.java:41)
14:34:24 	at org.junit.Assert.assertTrue(Assert.java:52)
14:34:24 	at com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandlerTest.testAshFrameData(AshFrameHandlerTest.java:225)
14:34:24 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)`

Similar to #333, the test case works locally stable but continuously fails on slow CI machines.

cdjackson, 2018-07-09:
I assume it is the same root cause I explained in the SPI tests. I’ll make the same refactoring.

cdjackson, 2019-02-16:
@hanneshofmann where do we take this? Are you able to provide a PR on the tests, or shall we close this?

hanneshofmann, 2019-02-16:
Sorry, I am not able to provide a PR to improve this test case. I therefore believe we shall better close this.

cdjackson, 2019-02-16:
Ok, thanks. Since this can't be reproduced by me, and I've never seen it fail on CI, I'll close.

closed 2019-02-16 by cdjackson