zsmartsystems/com.zsmartsystems.zigbee #343: Dedicated loggers for transmit/receive logs

hsudbrock, 2018-07-26:
Currently, the framework logs information about what commands are transmitted to / received from the ZigBee chips using the logger of the class that handles the transmission / reception. The log level is `TRACE` for the single RX/TX bytes in the `SpiFrameHandler` as well as for the RX bytes in the `AshFrameHandler`; it is `DEBUG` for the other logs (e.g., the complete RX/TX frames in the `SpiFrameHandler` and the `AshFrameHandler`, and for the logging of more high-level frame information like in the `ZigBeeDongleEZSP`, `ZigBeeDongleXBee`, and the like).

I propose to use dedicated loggers for the transmit/receive log statements, instead of using the logger of the class that handles transmission/reception, for the following reason: While these log messages are sometimes really useful, it is sometimes helpful to disable those log messages, so that they do not flood the log. Of course, one can disable them by setting the log level to `INFO` - but this will also suppress the other `DEBUG`-level log messages of the classes (which are a lot rarer, and do not flood the log easily). With dedicates loggers for transmit/receive log statements, one can disable those without disabling the logging for the complete class.

I would like to introduce such loggers for different protocol layers; at the example of Ember chips, there could be one logger for SPI frames (used in the `SpiFrameHandler`), one for ASH frames (use in the `AshFrameHandler`), and one for EZSP commands (used in the `ZigBeeDongleEzsp`). (One could also think about using two loggers for each, one for RX and one for TX, but I think that the simpler solution with one logger for RX and TX is sufficient.)

Regarding the naming of these loggers, my suggestion would be to use the namespace of the module producing the log statements, plus protocol name, plus `wire`, as these loggers log the data that actually goes over the wire to the ZigBee chip. Example: `com.zsmartsystems.zigbee.dongle.ember.ash.wire`.

I'd be happy to add a PR for such loggers, but would first like to discuss here about if and how this could be added to the library, and whether this would entail other changes (e.g., in the [ZigBee Log Viewer](https://www.cd-jackson.com/index.php/openhab/zigbee-log-viewer)).

cdjackson, 2018-07-26:
I don't think I have any objection to this, and it won't impact the log viewer so long as the log entries themselves do not change.

That said, is if trace is disabled (ie debug level is used) then it's only a single entry per packet transmitted and received isn't it? Or is this driven by the SPI handler which has the polling, and clearly this is causing some issue?

hsudbrock, 2018-07-26:
Yes, if trace is disabled then it is only a single entry per packet transmitted / received.

My first motivation for this suggestion came from flooded logs due to SPI's polling. However, I think that this is useful also for non-SPI, as there will also be many packets sent on a busy system. From my past experience it was often quite useful to have the ability to enable / disable wire logging when analyzing bugs (for TCP or HTTP traffic then - but I think it is somewhat comparable).

cdjackson, 2018-07-26:
I wonder if instead of changing the logging (or maybe as well as), we might be better to address the logging of the poll messages in the SPI handler and find a way to not log the polls and their responses. Otherwise, there will always exist the problem that IF you enable logging, it will still swamp logs?

hsudbrock, 2018-07-27:
I agree that, in the case of SPI, the polling will swamp the logs if enabled. Let me try to find a way to not log the polls and their responses.

hsudbrock, 2018-07-27:
Let me - for the sake of completeness - document the SPI-polling-related log messages here, and what could be done to suppress them.

These `DEBUG`-level log messages are created due to one polling cycle:

```
(1) [com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - TX EZSP: EzspCallbackRequest []
(2) [com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - --> TX SPI frame: FE 03 64 00 06 
(3) [com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - SPI Timer task stop
(4) [com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - SPI Timer task start
(5) [scom.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - <-- RX SPI frame: FE 03 64 88 07 
(6) [com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - SPI Timer task stop
(7) [com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - RX EZSP: EzspNoCallbacksResponse []
(8) [com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX: EzspNoCallbacksResponse []
(9) [com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - Unhandled EZSP Frame: EzspNoCallbacksResponse []
```

* (1) EzspFrameRequest TX logging in `SpiFrameHandler#sendNextFrame`: Could be suppressed easily, as this method checks explicitly whether a callback request must be sent for polling.
* (2) Frame bytes TX logging in `SpiFrameHandler#outputFrame`: Not so easy to suppress - one could  pass in an additional parameter to indicate the log level, or inspect frame bytes to check whether they represent an `EzspCallbackRequest` - both variants appear not really right to me, as they break the level of abstraction of the method.
* (3), (4), (6) SPI Timer Task logging in `SpiFrameHandler#startRetryTimer` and `...#stopRetryTimer`: After outputting the frame at (2), the old retry timer is stopped, then a new one is started, which is then later stopped - all resulting in a log entry. Like in the previous bullet point, we do not know at these points whether the timers are started/stopped due to SPI polling, so it is not straightforward to disable the logging in these cases.
* (5) Frame bytes RX logging in `SpiFrameHandler#processSpiCommand`: As for the TX bytes logging, we do not know here whether this is an `EzspNoCallbackResponse` - so not possible to suppress logging here without some kind of packet inspection.
* (7) EzspFrameResponse RX logging in `SpiFrameHandler#processSpiEzsp`: Could be suppressed by checking whether the response is an `EzspNoCallbackResponse`.
* (8), (9) Logging in `ZigBeeDongleEzsp#handlePacket`: Here, one could suppress the log messages for the `EzspNoCallbacksResponse` as well.

From these observations, I think that we should not try to not log the messages related to SPI polling, because some of the log messages cannot be suppressed easily (in particular, RX bytes logging at (5), where one would have to look into the packet to see whether the logging should be suppressed).

One other thing I realized is that, regarding retry timer logging, this should probably be done on `TRACE` - as these will occur for every sent frame.

Using the original suggestion of dedicated "wire loggers" one could then stop log flooding, and would only have to live with it when looking at SPI traffic; in this case, one would have to suppress unwanted log lines at log analysis time...

What do you think?

cdjackson, 2018-07-27:
> (1) EzspFrameRequest TX logging in SpiFrameHandler#sendNextFrame: Could be suppressed easily, as this method checks explicitly whether a callback request must be sent for polling.

Fine - let's do this.

> (2) Frame bytes TX logging in SpiFrameHandler#outputFrame: Not so easy to suppress - one could pass in an additional parameter to indicate the log level, or inspect frame bytes to check whether they [...]

Maybe we can implement a new method to send without logging that is only used for the polling?

> (3), (4), (6) SPI Timer Task logging in SpiFrameHandler#startRetryTimer and ...#stopRetryTimer: After outputting the frame at (2), the old retry timer is stopped, then a new one is started, which is [...]

I would actually argue that these can be reduced to TRACE, or removed even. This was mainly added when I was debugging and I'm not really sure it's of a lot of interest now.

> (5) Frame bytes RX logging in SpiFrameHandler#processSpiCommand: As for the TX bytes logging, we do not know here whether this is an EzspNoCallbackResponse - so not possible to suppress logging here without some kind of packet inspection.

This one is a problem, but could be addressed 'easily' at the expense of some small duplication. We could remove the general log at the top of the method. Add the packet to the end of the logged parameters on the ```lastFrameSent is null``` message, and then log individually in the case statements (I think there are maybe 3 log entries required there). In the ```SPI_CMD_EZSP``` case, we could inspect and conditionally log.

> (7) EzspFrameResponse RX logging in SpiFrameHandler#processSpiEzsp: Could be suppressed by checking whether the response is an EzspNoCallbackResponse.

As with point 1, I'd suggest to do this?

> (8), (9) Logging in ZigBeeDongleEzsp#handlePacket: Here, one could suppress the log messages for the EzspNoCallbacksResponse as well.

Agreed.

---

I would suggest that doing this is a good idea in general - even if we then split the loggers. The reason being that splitting the loggers would still end up with flooded SPI logs, and the above addresses this.

Personally, I think that if we address the above, we arguably don't need to split the loggers as the impact is quite low, and if you are interested enough in the EZSP protocol to want to see debug logging, then you will also want to have the packets being sent/received. I could be convinced otherwise though, but I kind of think that having the other protocol level debug without knowing what is received and transmitted seems a bit strange as you don't know what is going on without this.