zsmartsystems/com.zsmartsystems.zigbee #365: Failed to update firmware for EMBER EFR32MG

Shyamranny, 2018-08-13:
I am receiving 'Received CAN' message while trying to update the firmware of the dongle. 

Opening port /dev/ttyS2 at 115200 baud with FLOWCONTROL_OUT_NONE.
Ember bootloader: Serial port opened.
Ember bootloader: Got bootloader prompt.
Dongle firmware status: FIRMWARE_TRANSFER_STARTED.
Ember bootloader: Clearing input stream...
Ember bootloader: Starting transfer.
Ember bootloader: Transfer frame 1, attempt 0.
Ember bootloader: Received CAN.
Ember bootloader: Transfer failed.

cdjackson, 2018-08-13:
I have also seen this and I believe it is caused by trying to load an incompatible firmware.  Some files will load fine, and others will respond with this CAN after 1 or 2 frames - my guess is that the bootloader is doing some sort of check on the file header.

This is something I was discussing with Silabs recently and need to pick this up with them again.

cdjackson, 2018-08-13:
This is what I found on Friday when this was raised -:

I can successfully bootload the rts-cts versions of the code to the EFR32 no problem - it’s the HW version (which have the gbl format in the precompiled versions at least) that I can’t load. These files return the CAN after the first frame.

Shyamranny, 2018-08-13:
Working now, it was because I have used incorrect file type. For SiLab's EFR32MG, '.gbl' files worked. Thanks Chris.

closed 2018-08-13 by Shyamranny