zsmartsystems/com.zsmartsystems.zigbee #366: EMBER - reinitialize fails

Shyamranny, 2018-08-14:
networkManager.startup(true) is failing because of supplying invalid network key. Following is the log:-

08-13 18:01:03.511 5963-6033/ D/c.z.z.d.e.i.a.AshFrameHandler: [pool-6-thread-1     ] TX EZSP: EzspSetInitialSecurityStateRequest [state=EmberInitialSecurityState [bitmask=[EMBER_HAVE_NETWORK_KEY, EMBER_TRUST_CENTER_GLOBAL_LINK_KEY, EMBER_REQUIRE_ENCRYPTED_KEY, EMBER_NO_FRAME_COUNTER_RESET, EMBER_HAVE_PRECONFIGURED_KEY], preconfiguredKey=EmberKeyData [contents={00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00}], networkKey=EmberKeyData [contents={3E 11 25 BC 43 FC 14 31 48 D1 3C 00 CA 3D 00 FF}], networkKeySequenceNumber=0, preconfiguredTrustCenterEui64=0000000000000000]]
08-13 18:01:03.516 5963-6033/ D/c.z.z.d.e.i.a.AshFrameHandler: [pool-6-thread-1     ] --> TX ASH frame: AshFrameData [frmNum=2, ackNum=3, reTx=false, data=4B 00 FF 00 68 04 1B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 3E 11 25 BC 43 FC 14 31 48 D1 3C 00 CA 3D 00 FF 00 00 00 00 00 00 00 00 00]
08-13 18:01:03.532 5963-6032/ D/c.z.z.d.e.i.a.AshFrameHandler: [AshFrameHandler     ] <-- RX ASH frame: AshFrameData [frmNum=3, ackNum=3, reTx=false, data=4B 80 FF 00 68 B2]
08-13 18:01:03.536 5963-6032/ D/c.z.z.d.e.i.a.AshFrameHandler: [AshFrameHandler     ] ASH: Frame acked and removed AshFrameData [frmNum=2, ackNum=3, reTx=false, data=4B 00 FF 00 68 04 1B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 3E 11 25 BC 43 FC 14 31 48 D1 3C 00 CA 3D 00 FF 00 00 00 00 00 00 00 00 00]
08-13 18:01:03.538 5963-6032/ D/c.z.z.d.e.i.a.AshFrameHandler: [AshFrameHandler     ] RX EZSP: EzspSetInitialSecurityStateResponse [status=EMBER_KEY_INVALID]
08-13 18:01:03.539 5963-6032/ D/c.z.z.d.e.i.a.AshFrameHandler: [AshFrameHandler     ] --> TX ASH frame: AshFrameAck [ackNum=4, notRdy=false]
08-13 18:01:03.540 5963-6015/ D/c.z.z.d.e.i.EmberNetworkInitialisation: [pool-1-thread-1     ] EzspSetInitialSecurityStateResponse [status=EMBER_KEY_INVALID]

We have tried:-
1) networkKey = ZigBeeKey.createRandom();
2) networkKey = zigBeeDongleEzsp.getZigBeeNetworkKey();
In both the above cases, we are getting EMBER_KEY_INVALID.

It worked in 1.0.12. This error is in 1.0.15-SNAPSHOT.

cdjackson, 2018-08-14:
The issue here is that the link key is all 00 which is not valid. Previously, the link key was set within the Ember driver to the standard ZigBeeAlliance09 HA 1.2 key, but I removed this recently to allow other link keys to be used. So, you need to now set the link key with ```ZigBeeNetworkManager.setTcLinkKey()```.

Take a look in the ```ConsoleMain``` file where you will also see this being set.

Shyamranny, 2018-08-14:
That worked. Thanks @cdjackson

closed 2018-08-14 by Shyamranny