zsmartsystems/com.zsmartsystems.zigbee #398: SPI Bridge doesn't come up in v1.1.2

TomTravaglino, 2018-09-19:
Since we switched to version 1.1.2 on our machine with SPI the bridge doesn't come up. The status of the bridge is `Status=OFFLINE: Bad response received from dongle`. We stepped back to our previous version (1.0.15) and then it worked again. So we assume that something has changed and  makes the bridge not going ONLINE.
We also checked v1.1.2 on a machine with a Ember-dongle (UART) and it works there. It seems to be a SPI specific issue.

I attached a log file which contains the entries of the network start procedure.

[logs.txt](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/2397305/logs.txt)

cdjackson, 2018-09-21:
It looks like your firmware has an issue with the ```EZSP_CONFIG_BROADCAST_TABLE_SIZE``` setting. I would suggest trying a lower value.

You can adjust this with the ```updateDefaultConfiguration()``` method from within your implementation before calling ```initialize()```.

I don't think this is related to SPI, but to the version of firmware, or possibly to the available memory in the processor.

tomhoefer, 2018-09-21:
do you think that we can really use a number lower than 15 for this parameter? the spec says that the min value for this param should be 15

cdjackson, 2018-09-21:
I don’t know - you log shows the device complains about this value, so either try another one, or remove this configuration by setting it to null and it will be the same as the previous version.

cdjackson, 2018-10-14:
@TomTravaglino can we close this or is there still an issue?

TomTravaglino, 2018-10-15:
This issue is solved.

closed 2018-10-15 by TomTravaglino

hsudbrock, 2018-10-15:
The issue was - as far as I see - not related to the broadcast table size (although it looks weird to me that our dongles complain that the set value (which looks totally valid to me) would be invalid). The issue was caused by introducing the possibility to set an arbitrary link key, and to changing the default value handling for the radio channel, and both is resolved by using the most recent version of the binding.