zsmartsystems/com.zsmartsystems.zigbee #399: NPE when reading the ZigBeeNetworkKey for Telegesis

TomTravaglino, 2018-09-20:
In v1.1.2 the method com.zsmartsystems.zigbee.dongle.telegesis.ZigBeeDongleTelegesis.getZigBeeNetworkKey() throws a NPE. This happens because after sending a TelegesisGetFrameCntCommand the frameCnt is null and the method still tries to get its int value.

We could change this statement
```
networkKey.setOutgoingFrameCounter(frameCntCommand.getFrameCnt().intValue());
```

by something like this
```
Long frameCnt = frameCntCommand.getFrameCnt();
if (frameCnt != null) {
    networkKey.setOutgoingFrameCounter(frameCnt.intValue());
}
```

But how to correctly handle the non-existent frameCnt? Will it just be ignored and the network key stays as it is?

cdjackson, 2018-09-20:
There are methods starting with has... to check if the field exists, so this should be used here. It doesn’t matter if the counter is not provided - it just won’t be used. It’s only needed when transferring trust centre.

cdjackson, 2018-09-20:
Can you provide the trace of the exception please Tommaso?

> On 20 Sep 2018, at 07:44, Tommaso Travaglino <notifications@github.com> wrote:
> 
> networkKey.setOutgoingFrameCounter

TomTravaglino, 2018-09-20:
Here is the Exception. I took out the parts that contained our internal classes.

```
java.lang.NullPointerException
	at com.zsmartsystems.zigbee.dongle.telegesis.ZigBeeDongleTelegesis.getZigBeeNetworkKey(ZigBeeDongleTelegesis.java:737)
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.getZigBeeNetworkKey(ZigBeeNetworkManager.java:418)
	at xxx
	at org.openhab.binding.zigbee.console.internal.ZigBeeConsoleCommandExtension.handleZigbeeCommand(ZigBeeConsoleCommandExtension.java:145)
	at org.openhab.binding.zigbee.console.internal.ZigBeeConsoleCommandExtension.handleCommand(ZigBeeConsoleCommandExtension.java:113)
	at org.openhab.binding.zigbee.console.internal.ZigBeeConsoleCommandExtension.execute(ZigBeeConsoleCommandExtension.java:85)
	at org.eclipse.smarthome.io.console.ConsoleInterpreter.execute(ConsoleInterpreter.java:56)
```

cdjackson, 2018-09-20:
Ok, I misunderstood what you were doing here which is why I was after the stack dump.

You should not that this feature is not available in most Telegesis sticks - it is only available in sticks sold by DT. Do you have this problem with a DT stick running the modified software or a regular stick? If so, can you email me a debug log showing this failure?

cdjackson, 2018-11-16:
@TomTravaglino any update on this or answers to the above questions?

cdjackson, 2018-12-08:
@TomTravaglino any update on this or answers to the above questions? If not, I will close as it's been nearly 3 months.

cdjackson, 2018-12-15:
@TomTravaglino closing this - if there is still an issue, then please provide the information requested above and re-open.

closed 2018-12-15 by cdjackson