zsmartsystems/com.zsmartsystems.zigbee #400: Xbee Drive and Light Link Devices (Philips Hue)

otaviojr, 2018-09-21:
Hi,

I've been trying OH2 with XBee Xstick, and, I have a lot of hue bulbs.

OH2 is not been able to control hue bulbs. They join the network but after that the bulbs became unresponsive.

Looking deeper I endup at the XBee driver and notice that it forces aps encryption. 

I disabled aps encryption and hue start to work seamless.

And I think it make sense. Light Link Devices do not share the same link key as HA profile devices. So, APS encryption could not work with them.

Now, we could have at least an option to disable APS. I don't think about all security concerns, but, at least it will work with light Link Devices.

I will digg a little bit more on that. But that is what I could found so far.

regards

cdjackson, 2018-09-21:
Thanks for reporting. So I assume you mean the EE command is enabling APS security on all frames? Can you provide a document reference that shows this - the standard documentation just talks about enabling security and it's unclear what security it refers to.

The framework normally doesn't use APS security except for specific clusters or profiles where it is enabled specifically, so if this command is enabling APS, security it can be removed.

otaviojr, 2018-09-21:
Hi Chris,

No, EE must be enabled in order to hue works.

I'm attaching what I changed to make it works.

`  
...gbee.dongle.xbee/src/main/java/com/zsmartsystems/zigbee/dongle/xbee/ZigBeeDongleXBee.java
@@ -340,11 +340,11 @@ public void sendCommand(final ZigBeeApsFrame apsFrame) {
            command.setIeeeAddress(apsFrame.getDestinationIeeeAddress());
        }
         if (apsFrame.getDestinationAddress() < 0xFFF8 && apsFrame.getDestinationAddress() != 0) {
        //if (apsFrame.getDestinationAddress() < 0xFFF8 && apsFrame.getDestinationAddress() != 0) {
            // There seems to be a bug in the XBee that causes it to hang if APS Encryption is
            // enabled when sending a command to the local coordinator. Don't do it!
            command.addOptions(TransmitOptions.ENABLE_APS_ENCRYPTION);
        }
            //command.addOptions(TransmitOptions.ENABLE_APS_ENCRYPTION);
        //}
         command.setData(apsFrame.getPayload());
`

regards

cdjackson, 2018-09-21:
Thanks - I’d forgotten about this… I will change the way this works as since I wrote the XBee driver I have implemented an APS security flag from the upper layers.

cdjackson, 2018-09-21:
See #401 - can you give this a test please?

Also, what version of Hue bulbs do you have? Are they especially old (or new)? I have tested the XBee with Hue bulbs in the past, so I guess your bulbs are a bit different and it would just be interesting to know if they are old ones (or new ones!).

otaviojr, 2018-09-22:
Hi Chris, I tested the patch and it is working. I have many of them. Some, the first models, and the leds, the newest ones. I tried with both models, and both works. I bought Hue leds and they arrived yesterday. Will test them too.

I have ge bulbs, working Ok. Ge switchs and dimmers, working Ok. And Nyce Motion Sensors  and open/close sensors. Not working. They join, but, gets unresponsible. I will start looking for it now. Try to understand what is going on.

regards