zsmartsystems/com.zsmartsystems.zigbee #419: Frame handler use ZigBeePort after closing

hsudbrock, 2018-10-15:
This issue relates to https://github.com/openhab/org.openhab.binding.zigbee/issues/192, the issue that the serial port could not be correctly closed by the openHAB binding on some IoT gateways with an old implementation of `RXTX` (a `gnu.io` implementation). Summarized briefly, sometimes (but not always) the call to `RXTXPort.removeEventListener()` never returned, and, in consequence, the call to `RXTXPort.close()` was never executed.

I have looked a little deeper into this issue, and found out that this issue happens when the frame handler (`SpiFrameHandler`, `AshFrameHandler`) accesses the `ZigBeePort` instance *after* the shutdown of the handler was already initiated by calling the `setClosing()` method of the handler. (With SPI, this happens much more often than with ASH, probably simply because SPI uses polling, which increases the likelihood that the frame handler accesses the port after the closing has started.)

I do not know whether this behaviour of the frame handlers leads to an error for all `ZigBeePort` implementations (from looking at the nrjavaserial-based implementation used in openHAB, the error would probably be less visible, as it would result in a thread leak by not correctly terminating the event listener thread, instead of resulting in a non-closed port as in the issue linked above).

I open this issue as you might have some more knowledge, Chris, whether this is something that should be improved in this library; currently, the variable `closeHandler` is checked before accessing the serial port in the frame handlers - but sometimes, this flag is set to `true` after the check, but before accessing the serial port. Should this be changed here? Or is this an issue in the RXTX implementation, that should be able to deal with such situations?

cdjackson, 2018-10-15:
Hi Henning,
I thought we discussed this already in the past - is it not also related to https://github.com/openhab/org.openhab.binding.zigbee/issues/192?  There may also have been another issue, or possibly it was discussed elsewhere...

I recall spending some time looking at how RXTX closes when writing the original SPI driver for QHB and found a number of issues reported on StackOverflow (from memory) reporting that there are issues with closing the port.

>  currently, the variable closeHandler is checked before accessing the serial port in the frame handlers - but sometimes, this flag is set to true after the check, but before accessing the serial port.

I'm not sure I fully understand your point here? Do you mean that it is still possible to write to the port after setting the ```closeHandler``` flag? If so, then I agree that this might be worth changing to see if it helps - we could add a ```if(closeHandler) return;``` statement in ```outputFrame()``` to stop further access, although I do think that the RXTX should handle this!

I guess it can't hurt to add this and if it resolves the issue, then great.  Have you tried this yourself to see if it resolves the problem in the SPI Handler?

hsudbrock, 2018-10-15:
Hi Chris;

> I'm not sure I fully understand your point here? Do you mean that it is still possible to write to the port after setting the closeHandler flag?

Yes, exactly. The frame handlers still attempt to write to the port after the `closeHandler` flag is set (in repeated test runs I saw that every time the error closing the port occurred, it was accompanied by log messages indicating the the SPI polling caused a TX to the serial port at that time).

I do not know whether this is something that should be handled by RXTX or by this library, and trust your judgment here, as you have so much more experience with Java to serial communication.

> I guess it can't hurt to add this and if it resolves the issue, then great. Have you tried this yourself to see if it resolves the problem in the SPI Handler?

I have written a small patch that makes the `SpiFrameHandler#setClosing()` method wait until all threads that are executing a code block guarded by the `closeHandler` flag have exited that code block. My patch is a little more elaborate than adding `if(closeHandler) return;`, using synchronization to get rid of the tiny probability that the `closeHandler` flag is set just between this check and the following TX operation - but due to the synchronization my patch will of course have more runtime overhead.

Without this patch, I could reproduce the error when disposing the bridge 1 out of 5 test runs. With the patch, I can only reproduce it 1 out of 170 test runs - and for those fewer errors, I could not see any log messages indicating a write to the serial port after starting to close it.

I can re-do my experiment with a simple `if (closeHandler) return;` in the `outputFrame()` method, to check if this has the same positive effect as my solution using synchronization.

cdjackson, 2018-10-15:
> I do not know whether this is something that should be handled by RXTX or by this library, and trust your judgment here, as you have so much more experience with Java to serial communication.

I think it *should* be handled by RXTX, however, if it doesn't, then no problem to add it here, and in any case, the framework probably *shouldn't* write to the port once closed.

>  My patch is a little more elaborate than adding if(closeHandler) return;, using synchronization

Yes - I over simplified this I know ;)

> With the patch, I can only reproduce it 1 out of 170 test runs

Interesting that it's not 100% successful?!?

> I can re-do my experiment with a simple if (closeHandler) return; in the outputFrame() method, to check if this has the same positive effect as my solution using synchronization.

I don't mind. I would have assumed this should be synchronised, but if it's easy to rerun, it might be an interesting data point.

hsudbrock, 2018-10-15:
> Interesting that it's not 100% successful?!?

Yes, and I am currently investigating if I can find out more here (maybe my patch just has a bug, maybe there's some deeper issue...)

hsudbrock, 2018-10-19:
I investigated deeper here; it appears that using the serial port in the frame handler after is is set to close is not the root cause. 

The problem is rather how the event listener on the `RXTXPort` is removed in the ZB binding (see https://github.com/openhab/org.openhab.binding.zigbee/pull/196/files), which causes a deadlock: The removal of the event listener is initiated in a `synchronized(this)` block, which makes the event listener callback method block, as it also contains a `synchronized(this)` block - and the blocking event listener callback in turn blocks finalizing the removal of the event listener from the port.

I will hence close this issue, and open one in the ZB binding.

hsudbrock, 2018-10-19:
Side note: The fix I described above simply reduced the chance that the event listener was called after starting to close down the serial port, thereby reducing the likelihood of the deadlock. That's also why it was not 100% successful.

closed 2018-10-19 by hsudbrock