zsmartsystems/com.zsmartsystems.zigbee #429: Permit manufacturer-specific clusters on an endpoint

hsudbrock, 2018-11-02:
I want to work with a device that has a manufacturer-specific cluster (i.e., with a cluster ID in the range reserved for manufacturer-specific clusters, `0xfc00 – 0xffff`).

With the current state of this library, I could work with this cluster by 
* subclassing from `ZclCluster`, and sending commands / accessing attributes via my subclass, and
* registering listeners with the `ZigBeeNetworkManager` to get commands and reports from the cluster.

This is already nice :)

However, I want to use the manufacturer-specific cluster within the openHAB ZigBee binding; the code in this binding relies at some points on accessing clusters via the `ZigBeeEndpoint` of a `ZigBeeNode` - and with the approach described above the cluster will not be available via the `ZigBeeEndpoint`. A concrete example for this:
* The  `ZigBeeDeviceConfigHandler` of the binding accesses attributes via `ZigBeeNode#getEndpoint#get{In|Out}putCluster`. This class, hence, cannot write attributes of the manufacturer-specific cluster, because this cluster is unknown to the `ZigBeeEndpoint`.

One solution to improve here would be to provide the possibility to add clusters to a `ZigBeeEndpoint` (currently, all clusters of a `ZigBeeEndpoint` are derived from the endpoint's simple descriptor, ignoring manufacturer-specific clusters). This method could look as follows:
```java
  /**
     * Add a cluster to this endpoint. This method is intended to be used for clusters that are (a) not reported
     * within the simple descriptor of the endpoint, or (b) not supported directly by the library, and, hence, are not
     * contained in {@link ZclClusterType}.
     */
    public void addCluster(ZclCluster cluster) {
        if (cluster.isClient()) {
            inputClusters.put(cluster.getClusterId(), cluster);
        } else {
            outputClusters.put(cluster.getClusterId(), cluster);
        }
    }
```

Then, in the openHAB binding, one could add manufacturer-specific clusters to an endpoint of a node, e.g., using a custom thing handler.

What do you think, @cdjackson?

hsudbrock, 2018-11-02:
I got some more insights after playing around with the idea described above... I realised that it is not possible to write implementations for clusters that are not contained in the `ZclClusterType` enum. When subclassing `ZclCluster`, one has to initialize the attribute map with instances of the `ZclAttribute`, and objects of this class must hold a reference to the corresponding entry in `ZclClusterType`.

I don't know what the best solution here would be though. Rough ideas are the following:
1. not reference a `ZclClusterType` instance in the `ZclAttribute`, but only the cluster ID; return `null` on `ZclAttribute#getCluster()` if the enum does not contain an entry for the cluster ID;
2. make the `ZclClusterType` enum extensible (this can be done without giving up enum constants for clusters from the ZCL, using, e.g., the approach for extensible enums from Bloch's Java in Practice book);
3. add manufacturer-specific clusters to the core framework (I don't prefer this, and consider this a non-option, just listing it here for completeness).

hsudbrock, 2018-11-05:
I have played around some more with a device with a manufacturer-specific cluster, and realized that  adding manufacturer-specific clusters to an endpoint is not sufficient. 

The reason is that the `ZigBeeNetworkManager` never sets the manufacturer-specific flag and a manufacturer code when assembling the ZCL header. In consequence, the commands sent by the `ZigBeeNetworkManager`to manufacturer-specific clusters will be ignored on the device (in accordance with the ZCL spec).

I.e., as far as I see, currently manufacturer-specific clusters cannot be used via the `ZigBeeNetworkManager`, even without having them available in a `ZigBeeEndpoint`.

cdjackson, 2018-11-05:
The "manufacturer specific”`field, I believe, should be used to indicate a manufacturer specific extension to a standard cluster. Are you saying that in your device, even the manufacturer specific clusters have this set?

In any case, this point can easily be resolved, so I don’t have a big issue with it. The issue does however need to be considered within a wider refactoring that I’m looking at to solve some other issues in the cluster library that appear on some “exotic” clusters (as we have discussed some time back) so I don’t propose to rush in to this too quickly.

cdjackson, 2018-11-06:
Sure thing.

I will likely look at the manufacturer specific data in the header in the near term as this should be simple. The larger issue we can discuss on the phone.

cdjackson, 2019-01-08:
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/500

cdjackson, 2019-05-07:
This is now merged into the 1.2 branch so closing this.

PR #606 (2019-05-07): Add support for manufacturer-specific clusters, commands, and attributes
Replacement for PR #498

Closes #429

Also-by: Henning Sudbrock <henning.sudbrock@telekom.de>
closed 2019-05-07 by cdjackson