zsmartsystems/com.zsmartsystems.zigbee #435: Telegesis can not be initialized

TomTravaglino, 2018-11-14:
The Telegesis dongle can't be intilialized anymore. In the logs there is one noticeable entry which says: 
`Error setting Telegesis port echo`

Probably this issue is the result of the PR #360 

[telegesis-logs.txt](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/2580942/telegesis-logs.txt)

cdjackson, 2018-11-14:
I can not replicate the issue here.

> Probably this issue is the result of the PR #360

This PR was not merged as you said it did not resolve the issue.

TomTravaglino, 2018-11-19:
We shouldn't close it since the issue definitely exists on our machine. Did you check the attached logs? There you can see that the message "Error setting Telegesis port echo" is produced and then the bridge status is set to offline. What does it mean when this error occurs?

cdjackson, 2018-11-19:
Yes - I looked at the logs - they don't provide any information other than there is no communications.  Error setting the port echo is just an artefact of no communications as it's the first thing that is done (IIRC).

I thought you said that this was related to #360 - this was not merged - can you confirm that you are running the master and not some modified version?

TomTravaglino, 2018-11-19:
I can confirm that we're running the master. The assumption that it was related to #360 was a mistake. In tat PR we stumbled upon this diff [image] which let us think the instruction to disable echo had been added there. We didn't realize that the PR hadn't bee merged and we also didn't see that the instruction already existed in the previous version.
Anyway, something goes wrong when executing this instruction. Don't know if I can provide you more detailed info.

cdjackson, 2018-11-19:
I can't replicate the problem so I don't really know how I can help at the moment.  The problem appears to be a fundamental communication problem so you might want to look outside of the framework (eg serial drivers or configuration) as from a quick look at the repo, the Telegesis driver hasn't changed for 2 or 3 months other than to change return values and online/offline notifications to the stack.

hsudbrock, 2018-11-19:
@cdjackson First, let me say that I don't know whether #360 is related to the issue that @TomTravaglino mentions...

But: according to GitHub, #360 was merged - or do I overlook something here? (GitHub states on the PR page: "cdjackson merged 1 commit into master from telegesis_improve_startup on Aug 13 ")

cdjackson, 2018-11-19:
Sorry - I got muddled - I thought this was the refactoring to fix the testing. 360 is actually ok - it doesn't (I think) change any basic code - it just changes the online/offline notifications to the stack (from memory).

In any case, it doesn't change the fact that I can't replicate the issue with the master branch so unless you can provide more information I'm not sure what I can do. As I've said, the problem appears to be a basic communications issue in that no data is being received from the device and it is therefore failing on the first command sent.

hsudbrock, 2018-11-19:
I will take a look at the issue, @cdjackson .

cdjackson, 2018-11-19:
Thanks Henning.

hsudbrock, 2018-11-20:
I had a first look (just for info here - not expecting an answer to this comment):
* I can reproduce @TomTravaglino's issue on openHAB (i.e., with version 1.1.2 of this library) 
* I cannot reproduce with all Telegesis dongles (can reproduce with https://siliconlabs-h.assetsadobe.com/is/image//content/dam/siliconlabs/images/products/Bluetooth/zigbee-and-thread/telegesis_modules/etrx3usb-stick.jpg; cannot reproduce with https://www.qivicon.com/assets/Products/QIVICON/ZigBee-Funkstick/_resampled/SetSize500500-QIVICON-Zigbee-Funkstick-Pantone.jpg

I will continue investigating...

cdjackson, 2018-11-20:
I could be wrong, but I don't think that the configuration of the dongle (ie the S parameters) has changed between versions for a very long time (eg. never I think). 

It should be easy to check as there are 3 or 4 fields that set these parameters at the top of the TelegesisDongle class.

hsudbrock, 2018-11-21:
I see now the reason why some Telegesis dongles cannot be initialized: Those dongles simply do not terminate the responses to the AT commands with an `OK` (and also not with an `ERROR:`). In consequence, this library correctly never deems a transaction complete.

Now I need to investigate why these dongles do not send the `OK` terminator.

hsudbrock, 2018-11-21:
For my non-working Telegesis dongle, bit 1 of the S-register `0E` is set - disabling the `OK` prompt...

cdjackson, 2018-11-21:
I think there is a configuration to set this - I would have expected it was set to always return, but maybe there's the situation where this is not set early enough and it therefore is dependant on prior configuration (which is obviously not good and something I've tried to avoid).

cdjackson, 2018-11-21:
See here -:

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/master/com.zsmartsystems.zigbee.dongle.telegesis/src/main/java/com/zsmartsystems/zigbee/dongle/telegesis/ZigBeeDongleTelegesis.java#L232

hsudbrock, 2018-11-21:
Thanks Chris; I will take a look why this setting is not set correctly on my "bad" dongle.

hsudbrock, 2018-11-21:
I agree, Chris, that the initialization needed for transactions functionality should be done before the software reset. Does this comprise only setting the correct prompt, or also something else?

I don't yet see why this needs to be done in the `initialiseDongle` method as well, as the software reset won't change the S register for the prompt (at least according to the Telegesis AT-Command dictionary; it states that the `ATZ` command will not change the non-volatile S registers, and that the `S0E` register is non-volatile). Is there another reason for repeating this in the `initialiseDongle` other than the potential need to redo this after the software reset?

cdjackson, 2018-11-21:
I just checked - it's not the initialiseDongle method - it's the initialseNetwork method where it needs to be done (sorry - my bad). This resets the dongle to factory configuration so the registers may change.

hsudbrock, 2018-11-21:
Ah, I see, right. Then you were correct with the `initialiseDongle`, as this is called directly after factory resetting in the `initializeNetwork` :)

PR #440 (2018-11-27): Perform critical Telegesis initialisation prior to performing any transactions
This adds a method to send all commands required to complete transactions. This is then called before the stick is configured, and also again during the initialisation in case the stick was reset.

This also holds off the periodic status polling if a command is currently outstanding.

I tested this by setting all 3 of these configurations incorrectly, and the dongle correctly recovers.

@TomTravaglino I hope that this will resolve your issue - please can you test this ASAP as I'd like to flow this into the latest OH [... cut]
closed 2018-11-27 by cdjackson