zsmartsystems/com.zsmartsystems.zigbee #439: Issues when adding/subscribing an Osram plug

alexbrickwedde, 2018-11-21:
When adding an Osram plug to the network, i get a Packet cheksum failed.

And when subscribing to endpoint (OnOff, which i can read successfully) i get an NPE. The response does not contain records. But the status of the response itself is success.

[npe.txt](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/2603041/npe.txt)

cdjackson, 2018-11-21:
I would need to see a log that shows the data that causes the problem.

alexbrickwedde, 2018-11-22:
This one is ok?
[npe2.txt](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/2606842/npe2.txt)

cdjackson, 2018-11-22:
Yes - it's ok. Thanks. I will take a look - it might not be until late next week.

cdjackson, 2018-12-01:
I've created a fix for the NPE which is because some device is responding without part of the expected packet. This is just a printing issue.  I'm not sure what is causing the checksum exception - other than it seems there is an invalid frame received -:

```
DEBUG ZToolPacketParser - Received a BAD PACKET FE 00 00 00 00
```

Possibly the driver should simply ignore this frame as it's effectively just a bunch of 00s. Either way, it's not a valid frame and isn't causing any problem other than the printing of the error.

cdjackson, 2018-12-08:
Closed as above.

closed 2018-12-08 by cdjackson