zsmartsystems/com.zsmartsystems.zigbee #466: ASHFrameHandler does not handle retries correctly

triller-telekom, 2018-12-14:
As a follow-up to:
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/464#discussion_r241715960

the `ASHFrameHandler` starts a Thread for each request, but all seem to share the same `retries` field, which gets reseted whenever a new Thread is created.

Thus, the maximum retry limit is never reached and the `disconnect` won't be called to announce that the stack is OFFLINE.

cdjackson, 2018-12-14:
From the comment referenced above -:

> Because this is a bug. You are starting a new retrytimer in a thread and are setting the global retries to 0 here.

This is in fact correct - it is meant to be a global since we are looking at the number of retries of the specific frame, and not the total number of retries ever, since sooner or later, that would cause a reset (and it would be sooner, rather than later!).

The error I think is that the ```retries``` counter is set to 0 when the timer is started, and it should only be set to 0 when a new frame is sent. I suspect that this is the consequence of a change somewhere along the lines since the retry counter did work previously.

I will provide a PR for this in the next day or so.

closed 2018-12-15 by cdjackson

triller-telekom, 2019-01-04:
Running with the latest changes now and I just experienced this in out logs after I unplugged the stick:

```
2019-01-04_09:52:13:757 NA      DEBUG:  [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 1198
2019-01-04_09:52:14:767 NA      DEBUG:  [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 1199
2019-01-04_09:52:15:777 NA      DEBUG:  [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 1200
2019-01-04_09:52:16:787 NA      DEBUG:  [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 1201
2019-01-04_09:52:17:557 NA      DEBUG:  [symbolicName=org.eclipse.smarthome.core, className=org.eclipse.smarthome.core.items.events.ItemStateEvent] - zigbee_coordinator_ember_01381368_rx_dat updated to 2613
2019-01-04_09:52:17:577 NA      DEBUG:  [symbolicName=org.eclipse.smarthome.core, className=org.eclipse.smarthome.core.items.events.ItemStateEvent] - zigbee_coordinator_ember_01381368_tx_dat updated to 1805
2019-01-04_09:52:17:727 NA      DEBUG:  [symbolicName=org.eclipse.smarthome.core, className=org.eclipse.smarthome.core.items.events.ItemStateEvent] - zigbee_coordinator_ember_01381368_rx_ack updated to 144
2019-01-04_09:52:17:797 NA      DEBUG:  [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 1202
2019-01-04_09:52:17:817 NA      DEBUG:  [symbolicName=org.eclipse.smarthome.core, className=org.eclipse.smarthome.core.items.events.ItemStateEvent] - zigbee_coordinator_ember_01381368_tx_ack updated to 2613
2019-01-04_09:52:17:877 NA      DEBUG:  [symbolicName=org.eclipse.smarthome.core, className=org.eclipse.smarthome.core.items.events.ItemStateEvent] - zigbee_coordinator_ember_01381368_rx_nak updated to 114
2019-01-04_09:52:17:927 NA      DEBUG:  [symbolicName=org.eclipse.smarthome.core, className=org.eclipse.smarthome.core.items.events.ItemStateEvent] - zigbee_coordinator_ember_01381368_tx_nak updated to 0
2019-01-04_09:52:18:807 NA      DEBUG:  [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 1203
2019-01-04_09:52:19:817 NA      DEBUG:  [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 1204
2019-01-04_09:52:20:827 NA      DEBUG:  [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 1205
2019-01-04_09:52:21:837 NA      DEBUG:  [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 1206
2019-01-04_09:52:22:847 NA      DEBUG:  [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 1207
2019-01-04_09:52:23:857 NA      DEBUG:  [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 1208
2019-01-04_09:52:24:867 NA      DEBUG:  [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 1209
2019-01-04_09:52:25:877 NA      DEBUG:  [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 1210
2019-01-04_09:52:26:887 NA      DEBUG:  [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 1211
2019-01-04_09:52:27:897 NA      DEBUG:  [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 1212
2019-01-04_09:52:28:907 NA      DEBUG:  [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 1213
2019-01-04_09:52:29:917 NA      DEBUG:  [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 1214
2019-01-04_09:52:30:927 NA      DEBUG:  [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 1215
2019-01-04_09:52:31:937 NA      DEBUG:  [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 1216
2019-01-04_09:52:32:947 NA      DEBUG:  [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 1217
[... cut ...]
2019-01-04_09:52:52:137 NA      DEBUG:  [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 1236
```

And this continues...

So there is still something with the ASH retry counter, because the EZSP status requests are correctly queued and sent. Then I unplug the stick and the stick and the queuing continues indefinitely.

The PR #468 which is supposed to fix this is included in the code base where I am experiencing this issue.

cdjackson, 2019-01-04:
> So there is still something with the ASH retry counter, because the EZSP status requests are correctly queued and sent. Then I unplug the stick and the stick and the queuing continues indefinitely.

I don't see a problem here - this looks normal doesn't it? If you remove the stick, the device will go offline, and any subsequent requests that are sent from a higher layer will be queued - they will not be sent. I think that is what you are seeing here.

Maybe the messages should be rejected when the dongle is OFFLINE, but that won't change anything at a higher layer. I'm not really sure what you are expecting to happen?

Once the device is OFFLINE, then the higher layer should close the dongle and reinitialise. I use this in the sniffer application (https://github.com/zsmartsystems/com.zsmartsystems.zigbee.sniffer) and tested this a few days ago and it worked fine.

Can you explain what you think is wrong and what are you expecting to happen?  Do you have the log when this starts? Does it set the state to OFFLINE?

triller-telekom, 2019-01-04:
I agree that this should be processed by the higher layer.

And actually is gets processed and the device goes to OFFLINE, so far so good.

```
2019-01-04T10:23:55.137+0100   936  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: Retry Sent Queue Length 1
2019-01-04T10:23:55.137+0100   936  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameData [frmNum=3, ackNum=6, reTx=true, data=9C 00 18]
2019-01-04T10:23:55.657+0100   936  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 1
2019-01-04T10:23:56.077+0100   148  DEBUG          [symbolicName=org.eclipse.smarthome.core, className=org.eclipse.smarthome.core.items.events.ItemStateEvent] - zigbee_coordinator_ember_01381368_rx_dat updated to 1007
2019-01-04T10:23:56.147+0100   936  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: Retry Sent Queue Length 1
2019-01-04T10:23:56.147+0100   936  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameData [frmNum=3, ackNum=6, reTx=true, data=9C 00 18]
2019-01-04T10:23:56.167+0100   148  DEBUG          [symbolicName=org.eclipse.smarthome.core, className=org.eclipse.smarthome.core.items.events.ItemStateEvent] - zigbee_coordinator_ember_01381368_tx_dat updated to 636
2019-01-04T10:23:56.267+0100   148  DEBUG          [symbolicName=org.eclipse.smarthome.core, className=org.eclipse.smarthome.core.items.events.ItemStateEvent] - zigbee_coordinator_ember_01381368_rx_ack updated to 110
2019-01-04T10:23:56.267+0100   148  DEBUG          [symbolicName=org.eclipse.smarthome.core, className=org.eclipse.smarthome.core.items.events.ItemStateChangedEvent] - zigbee_coordinator_ember_01381368_rx_dat changed from 978 to 1007
2019-01-04T10:23:56.327+0100   148  DEBUG          [symbolicName=org.eclipse.smarthome.core, className=org.eclipse.smarthome.core.items.events.ItemStateEvent] - zigbee_coordinator_ember_01381368_tx_ack updated to 1007
2019-01-04T10:23:56.387+0100   148  DEBUG          [symbolicName=org.eclipse.smarthome.core, className=org.eclipse.smarthome.core.items.events.ItemStateEvent] - zigbee_coordinator_ember_01381368_rx_nak updated to 97
2019-01-04T10:23:56.457+0100   148  DEBUG          [symbolicName=org.eclipse.smarthome.core, className=org.eclipse.smarthome.core.items.events.ItemStateChangedEvent] - zigbee_coordinator_ember_01381368_tx_dat changed from 607 to 636
2019-01-04T10:23:56.517+0100   148  DEBUG          [symbolicName=org.eclipse.smarthome.core, className=org.eclipse.smarthome.core.items.events.ItemStateEvent] - zigbee_coordinator_ember_01381368_tx_nak updated to 0
2019-01-04T10:23:56.547+0100   148  DEBUG          [symbolicName=org.eclipse.smarthome.core, className=org.eclipse.smarthome.core.items.events.ItemStateChangedEvent] - zigbee_coordinator_ember_01381368_tx_ack changed from 978 to 1007
2019-01-04T10:23:56.667+0100   936  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 2
2019-01-04T10:23:57.677+0100   936  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 3
2019-01-04T10:23:57.787+0100   176  DEBUG          [symbolicName=org.eclipse.smarthome.config.discovery.usbserial, className=org.eclipse.smarthome.config.discovery.usbserial.internal.UsbSerialDiscoveryService] - Discovered removed USB-Serial device: UsbSerialDeviceInformation [vendorId=0x10C4, productId=0x8B34, serialNumber=01381368, manufacturer=Silicon Labs, product=BV 2010/10, interfaceNumber=0x00, interfaceDescription=BV 2010/10, serialPort=/dev/ttyUSB1]
2019-01-04T10:23:58.077+0100   936  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: Retry Sent Queue Length 1
2019-01-04T10:23:58.077+0100   936  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameData [frmNum=3, ackNum=6, reTx=true, data=9C 00 18]
2019-01-04T10:23:58.687+0100   936  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 4
2019-01-04T10:23:59.697+0100   936  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 5
2019-01-04T10:24:00.707+0100   936  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 6
2019-01-04T10:24:01.287+0100   936  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: Retry Sent Queue Length 1
[... cut ...]
the ASHFrameHandler was at count 12 and now it restarts at 1:

```
2019-01-04T10:24:07.777+0100   936  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue: 1
```

This is what I did not expect. When I tried to fix the retry counter in https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/455 this did not happen. I.e. the counter was increased until the network state was set to OFFLINE and afterwards all polling was stopped.

cdjackson, 2019-01-04:
> the ASHFrameHandler was at count 12 and now it restarts at 1:

The transaction queue gets reset to when OFFLINE is set. Why do you think this is a problem? If there's an issue with this, we can change it - I'm not sure it's necessary, but that is why the counter is reset.

> When I tried to fix the retry counter in #455 this did not happen. I.e. the counter was increased until the network state was set to OFFLINE and afterwards all polling was stopped.

But your implementation was flawed. You were counting the total number of retries since ASH went ONLINE - this is not the point of the counter. The counter is there to count the number of retries of the last message only (and this was stated in the comments in the code). It is in any case totally unrelated to the queue size. 

The fact that the queue is reset to 0 is simply that when the ASH disconnect is called, the queue gets cleared, but after that we continue to queue frames if the higher layers send them. As I suggested above, maybe we could look to reject any transactions passed down when OFFLINE, but ultimately this won't change anything.

triller-telekom, 2019-01-04:
> The transaction queue gets reset to when OFFLINE is set. Why do you think this is a problem? 

That's fine for me.

I was just wondering why we keep queuing up frames although the network state is offline. Maybe I missed that in https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/464

What do you think?

cdjackson, 2019-01-04:
I can interpret your question two ways, so I will answer twice :)

> I was just wondering why we keep queuing up frames although the network state is offline. 

The frames are simply queued because the upper layers are still sending frames. The ASH layer is OFFLINE, so it doesn't send it, and instead queues the frame. The queueing is done also during the initialisation to ensure that any messages sent during startup are queued until the ASH connection comes ONLINE - it will then send these messages.

If you think that this is a problem, then maybe we can change this so that if ASH goes from OFFLINE to ONLINE, all further transactions are rejected. I wouldn't just want to reject frames if ASH was OFFLINE, as that might then cause problems during the startup, and we would somehow need to synchronise upper layers to not start sending data until after ASH is online.

triller-telekom, 2019-01-04:
I forgot about the initialization phase where the queuing is indeed necessary. Thank you for the explanation.

I am not sure if you have mixed up OFFLINE and ONLINE in your second paragraph:

> then maybe we can change this so that if ASH goes from OFFLINE to ONLINE, all further transactions are rejected

Do you rather mean that if the ASH goes from **ONLINE** to **OFFLINE**, we do reject further transactions? because that would make sense IMHO.

cdjackson, 2019-01-04:
Yes - I meant going TO OFFLINE.

triller-telekom, 2019-01-04:
Alright, I have created https://github.com/zsmartsystems/com.zsmartsystems.zigbee/issues/494 to follow-up on this in order to not flood this old and closed issue.

PR #468 (2018-12-15): Fixed ASH retry counter and added tests
@triller-telekom FYI this ought to resolve your problem.
Closes #466