zsmartsystems/com.zsmartsystems.zigbee #493: Does this library support CC2531 with z-stack 3.0 firmware?

busyluo, 2019-01-04:


cdjackson, 2019-01-04:
I have not tested it with the latest TI firmware so I am not sure. As a minimum certain features such as setting the keys, install codes, etc are not going to work and would need to be added.

If you wanted to take a look at this, it would certainly be welcome and I'd be happy to support.

franklx, 2019-01-04:
I'm very sorry - I messed up with gradle configuration.

cdjackson, 2019-02-16:
Closing this as it is just a question, and this was answered.

closed 2019-02-16 by cdjackson

wsowa, 2020-02-13:
@franklx did you get any conclusions about Z-STACK 3.0.X?

@cdjackson has anything changed wrt support for ZB 3.0 security features since the last year ?

cdjackson, 2020-02-13:
I rewrote the driver mid last year to use the latest ZStack spec. For various reasons, this was not migrated into the library - it was tested with the latest TI2531 source though.

TI no longer support the TI2531 and at the time I found a number of bugs that they said they would not fix, but maybe they still support it on their other ZB chips.

I can make this available if someone wants to look at it?

wsowa, 2020-02-13:
I may be interested in the future but currently I don't really have time to push this forward. 

It is worrying thought that there are bugs they they won't fix. Could you list some most impactful or refer to discussion with IT if it was on some public forum? I would like to understand how much hopeless the situation is. 

Also when you say "latest source" do you mean 3.0 stack version? This one is still active so I'm surprised they refuse to fix bugs.

The other question still stands: has anything been added regarding support of ZigBee 3.0 feature to the library over the last year of would this need to be implemented if we want to support ZB 3.0 on TI's or other dongle?

cdjackson, 2020-02-13:
I don't really recall - I just did a search on my posts on the TI forum and came up with this -:

https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/t/802861#pi320995=2

This is where the sequence number returned by ZStack is always 0.

[image]

There were others, but I forget what they were.

> Also when you say "latest source" do you mean 3.0 stack version

Yes.

> This one is still active so I'm surprised they refuse to fix bugs.

I can only repeat what they told me on a couple of occasions - please see the image above as an example.

> The other question still stands: has anything been added regarding support of ZigBee 3.0

As I said, I completely rewrote the driver to support ZigBee 3.0 with the latest ZStack (ie 3.03 if I remember correctly).