zsmartsystems/com.zsmartsystems.zigbee #513: Refactoring of the network serialisation interface

cdjackson, 2019-01-19:
The current network serialisation has a couple of problems that need to be resolved.

1) On large networks or slow computers, serialisation of the network can be slow - ie it can take 5 seconds or more to serialise the whole network. We see this taking 5 seconds with an RPi, and 40 nodes on openHAB.
2) If there are a lot of updates, these get queued due to the thread synchronisation which further slows things down. There possibly should be some level of delay to ensure that only the last update is persisted.

I propose to change the network serialisation to deprecate the current interface, and either add a new interface, or to require implementors to implement their own solution using the ```ZigBeeNetworkNodeListener``` which is called when nodes are added and removed.

I think given the high dependency the framework has on this serialisation, that a specific interface is useful rather than using the ```ZigBeeNetworkNodeListener``` interface, but welcome any thoughts.

@hsudbrock @puzzle-star FYI if you have any comments.

puzzle-star, 2019-01-19:
> either add a new interface, or to require implementors to implement their own solution using the `ZigBeeNetworkNodeListener` which is called when nodes are added and removed.

I only have the OH vision, but from the OH binding it does not really seem necessary to keep a dedicated interface.

Everything seems to be possible through the NodeDao interfaces, the node add/update/remove events for saving, and the addNode for restoring.

> I think given the high dependency the framework has on this serialisation, that a specific interface is useful rather than using the `ZigBeeNetworkNodeListener` interface, but welcome any thoughts.

Well, as I said, I only have the view on OH binding and I do not know where else this is used, but if the implementation is managing evertything, I think the current existing events are enough.

I would though keep an eye on the possibility of giving the implementation the opportunity to restore the state (adding the nodes) before everything starts running (discoveries etc). I do not know if this is already currently like this

cdjackson, 2019-01-19:
Thanks for your comments @puzzle-star.

> Everything seems to be possible through the NodeDao interfaces, the node add/update/remove events for saving, and the addNode for restoring.

On further thought, I think there is actually a need to have a dedicated interface for this, and that is to restore information. This is not possible through the current node listener callbacks which only allow the framework to notify of changes to nodes, and not to restore information. We could just let the user restore all nodes into the system before starting, but that feels too uncontrolled.

I will therefore propose a new interface - something along the following lines -:

```
public interface ZigBeeNetworkDatabase {
    ZigBeeNodeDao getNode(IeeeAddress address);

    void setNode(ZigBeeNodeDao node);

    void removeNode(IeeeAddress address);
}
```

> I would though keep an eye on the possibility of giving the implementation the opportunity to restore the state (adding the nodes) before everything starts running (discoveries etc). I do not know if this is already currently like this

Yes, the state is restored (all nodes added) before the system is brought online.

puzzle-star, 2019-01-19:
> We could just let the user restore all nodes into the system before starting, but that feels too uncontrolled.

Is not it like this now? (as per _"Yes, the state is restored (all nodes added) before the system is brought online"_). In the current deserialization interface, the implementation goes through all saved nodes (even if today all in one file) and adds them to the network using "addNode".

```
public interface ZigBeeNetworkDatabase {
    ZigBeeNodeDao getNode(IeeeAddress address);
    void setNode(ZigBeeNodeDao node);
    void removeNode(IeeeAddress address);
}

```

With this, how does the NetworkManager know what it has to restore if the implementation is not starting the process or has an interface to say what to restore? Today deserialize() handles it as it adds all nodes, but I think your new approach is better, in which case I think we need in the interface something like:

>     Set<IeeeAddress> getNodes();

cdjackson, 2019-01-19:
> Is not it like this now?

No, it's not - it is controlled by the network manager. ie the network manager reads the nodes when it needs to. The option of allowing the user to addnodes when it wants to means that this could happen at any time.

> Yes, the state is restored (all nodes added) before the system is brought online"

Currently, the network manager reads the state before it brings the system online. This is controlled by the network manager - not the user.

> With this, how does the NetworkManager know what it has to restore 

Yes, I already came across this, so the interface is now a little different -:

```
/**
 * Interface to be implemented by users providing a node database for persisting device data outside of the framework.
 * <p>
 * ZigBee coordinators do not store any information about devices that have joined the network. In order to provide a
 * continuous service, information about nodes that have joined needs to be persisted between restarts of the framework.
 * <p>
 * The network database persists data in two stages - it keeps a list of node addresses that have joined the network,
 * and it writes detailed information about each node as required. These two functions are separated to improve
 * efficiency and speed, especially on low performing hardware.
 *
 * @author Chris Jackson
 *
 */
public interface ZigBeeNetworkDatabase {
    /**
     * Writes the list of nodes that are currently included in network.
     *
     * @param nodes the {@link Set} of {@link IeeeAddress}s of all nodes currently included in the network
     */
    void writeNetworkNodes(Set<IeeeAddress> nodes);

    /**
     * Reads the list of nodes that are currently included in network.
     *
     * return nodes the {@link Set} of {@link IeeeAddress}s of all nodes currently included in the network
     */
    Set<IeeeAddress> readNetworkNodes();

    /**
     * Called when information about a node has been updated, and the node must persist the node data.
     *
     * @param address the {@link IeeeAddress} of the node to retrieve
     * @return the {@link ZigBeeNodeDao} containing the node data
     */
    ZigBeeNodeDao readNode(IeeeAddress address);

    /**
     * Called when information about a node has been updated, and the node must persist the node data to non-volatile
     * storage.
     *
     * @param node the {@link ZigBeeNodeDao} to be persisted
     */
    void writeNode(ZigBeeNodeDao node);

    /**
     * Called when a node has been removed from the network. It is expected that the database implementation will remove
     * this data from the storage.
     *
     * @param address the {@link IeeeAddress} of the node to remove
     */
    void removeNode(IeeeAddress address);
}
```

This allows the network to write the list of nodes when the nodes in the network change (which is not often), but when an individual node changes, it can be written by itself (as this happens much more often). This allows the full list of nodes to be restored on startup, but individual nodes to be persisted.

WDYT?

puzzle-star, 2019-01-19:
Ok, let me see if I understand correctly the interface (as I think I may be confusing some of the api calls). I start with the simple ones :-)

`void writeNode(ZigBeeNodeDao node);`
`void removeNode(IeeeAddress address)`

When the NetworkManager adds / updates / removes a node from the network, these are called for the database to be updated

`ZigBeeNodeDao readNode(IeeeAddress address);`

When the NetworkManager wants to retrieve a node from the database.

But the description _"Called when information about a node has been updated, and the node must persist the node data."_ seems misleading (at least to me). Maybe something like _"Called to retrieve information for a node from the database"_ would be more clear?

`Set<IeeeAddress> readNetworkNodes();`

When the NetworkManager wants to know the list of devices in the database (normally at startup?).

If so, should not the description be _"Reads the list of nodes that are currently stored in the database"_ instead of _"currently included in network."_? (same for the param). At least it has confused me a bit when reading it the first time.

`void writeNetworkNodes(Set<IeeeAddress> nodes);`

I may be misunderstanding this one. If it is meant for the NetworkManager to "batch write" a set of nodes:
- The NetworkManager calls this method with a set of IeeeAddresses
- The implementation will call NetworkManager.getNode(address).getNodeDao() for each of these and write it to the database

Why is this needed, instead of the NetworkManager just iterating through the set of nodes itself and calling writeNode with the NodeDao directly? Also, this makes the implementation of the interface know about the NetworkManager (to call getNode()), while this is the only part of the interface requiring this and might be cleaner without this dependence?

Or am I missing something?

cdjackson, 2019-01-19:
> If so, should not the description be "Reads the list of nodes that are currently stored in the database" instead of "currently included in network."? (same for the param). At least it has confused me a bit when reading it the first time.

Yes, sorry - your understanding is correct, and this is a cut and paste error.

> I may be misunderstanding this one. If it is meant for the NetworkManager to "batch write" a set of nodes:

No - it's not intended to write the nodes. It only writes the node addresses. ie it is saving the list of nodes that the manager currently thinks are joined to the network.

> The implementation will call NetworkManager.getNode(address).getNodeDao() for each of these and write it to the database

No - this is not intended. The expectation is that the database should simply store the list of nodes. But I'll answer / discuss your next question as maybe it is a better one :)

> Why is this needed

Good question, and maybe it's not. My thinking was to maintain a symetrical interface - we must have a ```readNetworkNodes``` method, so I thought for completeness, it was useful to also write the network nodes list as an atomic entity - to avoid any confusion. It avoids the implementation then having to maintain a list of nodes that are actually included in the network - I know this could be derived from the list of files in a store (or something similar).

I agree that this could be removed, but I think it is safer to explicitly set the node list - that was my thinking anyway ;)

> Also, this makes the implementation of the interface know about the NetworkManager (to call getNode())

This is not intended - the method is only expected to store the node list. It is not intended to call back in to the network manager to serialise data itself.

When defining this, I also had another use case in mind, and that is that we _could_ maintain a database of devices (their clusters, descriptors, attributes etc) separately to the nodes that are currently joined to the network. By separating the saving of the nodes, and the saving of the network, it is possible to do this and not have to expect implementations to somehow maintain this separately somehow which could get out of sync. (I'm not sure I've explained that well - sorry).

puzzle-star, 2019-01-19:
> Always. There must be a list of list of nodes somehow - either the store infers this, or we explicitly set it. Either way, we must be able to get this list (with the `readNetworkNodes`).

This is what I am saying: the implementation will have to keep this info anyway when receiving writeNode/receiveNode calls.

> The data store should not call back in to the network manager.

This is what I understood, and this is why I initially thought the writeNetworkNodes was wrong, when I thought it was intended to do a "batch write". But if it is not the purpose, I still do not see the benefit of it, as the implementation should anyway probably keep track of the list when writeNode / receiveNode are called.

> I think that you have a different view on what I'm planning (inferred as you still think that the data store is allowed to call back into the network manager, and I guess this is why you are thinking about synchronisation issues?).

Not really. I do think the implementation should not be calling back the NetworkManager.

But I still see synchronization issues because, unless I am missing the point, both writeNode/removeNode and writeNetworkNodes target at the same persisted list.

So my understanding is:
- Network starting:
   - NetworkManager calls readNetworkNodes
   - NetworkManager calls readNode for each item in the list and adds it to the network

- Network working:
   - When a node is added, NetworkManager calls writeNode and the implementation persists it in the database (and adds it to its internal list, as it needs to keep a database)
   - When a node changes, NetworkManager calls writeNode and the implementation persists it in the database
   - When a node is removed, NetworkManager calls removeNode and the implementation removes it from the database (and from its internal list)
   - When is writeNetworkNodes called? If called:
        - Can it contain a different list from the already existing one?
        - What is the implementation expected to do with missing items in the list?
        - What is the implementation expected to do with extra items in the list, specially if it cannot call back the NetworkManager to see what these are?

- Network shutdown:
    - The NetworkManager could iterate through all nodes and call writeNode to make sure the database is up to date, but this is not really needed.

I hope this clarifies why I do not see the use of writeNetworkNodes :-)

cdjackson, 2019-01-19:
> This is what I am saying: the implementation will have to keep this info anyway when receiving writeNode/receiveNode calls.

Well, it can either build the list, or we can set it expliticly. I prefer to set it explicitly - if it has to build it, then there is the potential for errors, resulting in lost nodes.

> But I still see synchronization issues because, unless I am missing the point, both writeNode/removeNode and writeNetworkNodes target at the same persisted list.

I don't know what list you refer to?!? These calls do not operate on any list - they only operate on the node.

> Can it contain a different list from the already existing one?
> What is the implementation expected to do with missing items in the list?

I already answered these above - it is not expected to do anything!

> I hope this clarifies why I do not see the use of writeNetworkNodes :-)

Yes, I agree - it can be done like this, and is why I said that it arguably isn't needed. However, as I explained, IMHO it is better to explicitly set the list to avoid chance of error with having to rebuild the list.

cdjackson, 2019-01-20:
> Or the implementation should always answer to readNetworkNodes whatever it received from writeNetworkNodes, regardless the state of the DB?

My expectation is that it should return in readNetworkNodes whatever it received from writeNetworkNodes.

> This is where I see potential synchronization / corruption issues.

I can't see how this can cause synchronisation issues. Or when you refer to synchronisation, you don't refer to code synchronisation (as in synchronisation blocks) but database synchronisation? In this case maybe I misunderstood you (I had in mind block syncs due to other conversations ;) ).

Anyway... I have now implemented this, and having worked it through, I will agree with you that it adds unnecessary complexity for a use case that is not common, so I will remove these two API methods and just act on the nodes :). If it future the use case is necessary, then we can add this support and make it optional.

puzzle-star, 2019-01-20:
> Or when you refer to synchronisation, you don't refer to code synchronisation (as in synchronisation blocks) but database synchronisation?

I refer to DB synchronization: if we keep writeNetworkNodes, for each db update we will have to call. I.e., adding a node, we will call both writeNetworkNodes (to update "the list") and writeNode (to update the "node"). As these are two different separate calls, we may run into data consistency issues (let's not call it synchronization) as something can happen between the two calls that might leave the db inconsistent (ie. nodes stored but not in the list).

> I will remove these two API methods

I think readNetworkNodes is still necessary. How would otherwise the NetworkManager know which nodes to restore at startup?

cdjackson, 2019-01-20:
> we may run into data consistency issues

I really don't think this is an issue - if it were to happen, then we would simply loose a node which is the same thing that would happen if we didn't have the two calls. Having the extra call provides some additional flexibility, so it's something that could be looked at in future if we need it, but for now I won't implement this.

> I think readNetworkNodes is still necessary.

Yes, of course - I wrote to fast. The interface now looks like this -:

```
/**
 * Interface to be implemented by users providing a node database for persisting device data outside of the framework.
 * <p>
 * ZigBee coordinators do not store any information about devices that have joined the network. In order to provide a
 * continuous service, information about nodes that have joined needs to be persisted between restarts of the framework.
 * <p>
 * The underlying data store implementation must be able to store and retrieve node data with the
 * {@link #writeNode(ZigBeeNodeDao)} and {@link #readNode(IeeeAddress)} methods, and also provide a list of all nodes
 * currently in the store with the {@link #readNetworkNodes()} method.
 *
 * @author Chris Jackson
 *
 */
public interface ZigBeeNetworkDataStore {
    /**
     * Reads the list of nodes that are currently included in network. The underlying data store should return the list
     * of all nodes that have been stored with {@link #writeNode(ZigBeeNodeDao)}. If {@link #removeNode(IeeeAddress)}
     * has subsequently been called, the node shall not be returned in this Set.
     *
     * return nodes the {@link Set} of {@link IeeeAddress}s of all nodes currently included in the network
     */
    Set<IeeeAddress> readNetworkNodes();

    /**
     * Called when the library wants to restore the saved information about a node. This is normally only done on system
     * startup.
     *
     * @param address the {@link IeeeAddress} of the node to retrieve
     * @return the {@link ZigBeeNodeDao} containing the node data
     */
    ZigBeeNodeDao readNode(IeeeAddress address);

    /**
     * Called when information about a node has been updated, and the node must persist the node data to non-volatile
     * storage.
     *
     * @param node the {@link ZigBeeNodeDao} to be persisted
     */
    void writeNode(ZigBeeNodeDao node);

    /**
     * Called when a node has been removed from the network. It is expected that the database implementation will remove
     * this data from the storage.
     *
     * @param address the {@link IeeeAddress} of the node to remove
     */
    void removeNode(IeeeAddress address);
}
```

I will provide an initial PR shortly, although it's not fully tested yet.

puzzle-star, 2019-01-20:
One added, I can have a look and implement the replacement of the current serializer in the OH binding (if you plan to keep backwards compatiility while testing it, otherwise this is too coupled and has to be done and tested at the same time)

cdjackson, 2019-01-20:
Thanks.

I was thinking that we can easily create a converter - ie the first time the binding runs after this is implemented, it converts to the new format. Remember that the new format is fundamentally the same as the old format, so it can read the old file, and write individual files for each node, then delete the old file so this doesn’t run again.

I’m just about to create the PR - I will not merge before I go on holiday, but feel free to test/play and if you have a PR for OH when I return we can get everything merged.

cdjackson, 2019-03-10:
Closing this as it's now merged into the ```dev-1.2.0``` branch

PR #515 (2019-03-03): Add new Data Store serialisation concept
This refactors the serialisation interface. It adds a new ```ZigBeeNetworkDataStore``` interface that is used to read/write/remove nodes. Users should implement this interface as they see appropriate for their system (eg SQL database, XML files, Json...).

This is different to the current serialisation interface which acts on the whole network. This causes problems with slower computers (eg RPi class) and larger networks where this can take in the order or 1 to 3 seconds to serialise. Additionally, multiple serialisations may occur at once (or close tog [... cut]
closed 2019-03-10 by cdjackson