zsmartsystems/com.zsmartsystems.zigbee #514: Decoupling of dongle notifications to the ZigBeeNetworkManager

cdjackson, 2019-01-19:
Currently the notifications from the dongle (eg received messages, state updates etc) are processed in the dongle receive handler thread. This leaves the dongle handler susceptible to any delays in the processing by the upper layers.

I propose to separate this by adding a thread queue in the ```ZigBeeNetworkManager``` to queue all incoming notifications, thus allowing the dongle notification thread to return immediately.

@puzzle-star FYI if you have any comments.

puzzle-star, 2019-01-19:
Hi @cdjackson,

Looking at the code where the contention happens, it is currently on the TelegesisFrameHandler when we notify the TelegesisListeners.

Of course. this ends up calling the ZigBeeDongleTelegesis listener implementation of the interface, which ultimately gets into the NetworkManager, where the blocking occurs.

The issue I find in solving this in the NetworkManager, or even in the ZigBeeDongleTelegesis, is that these are just listeners. The way the code is implemented (and I like it :-) ) seems to be targetting at allowing any kind of listener to do whatever it wants to do. If any of these listeners blocks (as the NetworkManager today), then the Dongle Handler will get blocked again, unless there is a hard compromise on the listeners implementing the interface that makes clear they MUST NOT block.

So it seems to me then that fixing it at the NetworkManager level is again a "workaround" rather than fixing the issue (well, there is something to fix there regarding synchronization on "this" before notifying, but probably another issue not related only to the Dongles) unless it is clear that the calls should never block.

A way to decouple (but this would need to be done on all the dongle handlers) would be i.e.:

```
    private boolean notifyTransactionComplete(final TelegesisCommand response) {
        boolean processed = false; // <-- Is this needed? The returned value is not currently being used

        logger.debug("RX Telegesis: {}", response);
        synchronized (transactionListeners) {
            for (TelegesisListener listener : transactionListeners) {
                NotificationService.execute(new Runnable() { // <--- Added here to decouple notifications
                    try {
                        if (listener.transactionEvent(response)) {
                            // processed = true; // <-- What to do with the response if on a different thread?
                        }
                    } catch (Exception e) {
                        logger.debug("Exception processing Telegesis frame: {}: ", response, e);
                    }
                })
            }
        }

        return processed; // <-- If we decople the notification, we cannot do this. The returned value does not seem to be used anyway in the code
    }

```

If done at the NetworkManager, is there really a need to implement a queue? Or it would just be necessary to make sure the notification handling functions will not block?

Does it make sense to you?

It seems to me that there is not much code shared between the different dongle implementations, as i.e. the TelegesisHandler and the ZigBeeDongleTelegesis are just subclasses of Object implementing some required interfaces.

To provide some common functionality, these would need to either inherit from some i.e. AbstractZigBeeDongle class implementing some common functionality, or use some helper classes providiong common functionality. Otherwise the only way I see is to do it dongle by dongle.

As the amount of supported dongles increases, maybe it would be worth doing some common stuff at some time?

cdjackson, 2019-01-19:
I think what I've suggested is the same thing as you are suggesting, only it is done in the common code rather than having to be implemented in every dongle implementation.

You are suggesting to have the thread executor in the frame handler before the call is made to the network manager. I am simply suggesting to do the same thing, but in the network manager. Doing it in the common code makes more sense to me as I don't need to do this in 5 different dongles.

Or am I not understanding what you are saying? I think it's basically the same though? 

Or do you think it is better to implement this in each dongle instead? There is a case for this - we could argue that each dongle should protect itself, but on the other hand, doing this in common code is IMHO better.

cdjackson, 2019-01-19:
To show what I mean - the following diagram shows the stack - you are proposing to provide the thread executor at the red circle, while I propose to put it at the green circle. I think it provides the same function, but by doing it where I suggest, it will work for all dongles.

[image]

Maybe I miss an advantage of doing it where you suggest?

puzzle-star, 2019-01-19:
Hi @cdjackson,

I think we are saying the same, and I agree that if it is clear that the Listeners should not block, it is better to do the decoupling in the NetworkManager.

Just I understood you wanted also the dongles to protect themselves, and I did not see where to do it in common code :-)

But is not it already like this? I mean, you said that you thought it was already decoupled, and it seems also to me that it is. 

Just then there is an issue in the current implementation in the NetworkManager where the decoupling happens inside a synchronized block, which I think is what should be fixed, as we seem to agree that it is better to do the decoupling in common code.

I.e., the notifications are decoupled here, but the issue is that it is done inside a synchronized branch that could get blocked for an undetermined amount of time before getting to it (as it is happening now with the serialization)

```
    @Override
    public void nodeStatusUpdate(final ZigBeeNodeStatus deviceStatus, final Integer networkAddress,
            final IeeeAddress ieeeAddress) {
        logger.debug("{}: nodeStatusUpdate - node status is {}, network address is {}.", ieeeAddress, deviceStatus,
                networkAddress);

        // This method should only be called when the transport layer has authoritative information about
        // a devices status. Therefore, we should update the network manager view of a device as appropriate.
        switch (deviceStatus) {
            // Device has gone - lets remove it
            case DEVICE_LEFT:
                // Find the node
                ZigBeeNode node = getNode(networkAddress);
                if (node == null) {
                    logger.debug("{}: Node has left, but wasn't found in the network.", networkAddress);
                } else {
                    // Remove the node from the network
                    removeNode(node);
                }
                break;

            // Leave the join/rejoin notifications for the discovery handler
            case UNSECURED_JOIN:
                break;
            case SECURED_REJOIN:
            case UNSECURED_REJOIN:
                break;
            default:
                break;
        }

        // Notify the listeners
        synchronized (this) {
            for (final ZigBeeAnnounceListener announceListener : announceListeners) {
                NotificationService.execute(new Runnable() {
                    @Override
                    public void run() {
                        announceListener.deviceStatusUpdate(deviceStatus, networkAddress, ieeeAddress);
                    }
                });
            }
        }
    }

```
But in this specific case. I do not see why we need the "synchronized (this)" at all before notifying the listeners, the the list can be accessed safely even in a concurrent scenario (this is an umnodifiable list being replaced by a new instance when adding / removing listeners), as it seems to be for the other event listeners.

The part that I think needs a bit more of thinking is why all these notification blocks are synchronized on (this), as I might be missing something or some reasonfe for which this needs to be done.

> I propose to separate this by adding a thread queue in the `ZigBeeNetworkManager` to queue all incoming notifications, thus allowing the dongle notification thread to return immediately.

So this is what I am asking whether it is necessary, as this is already (mostly) done in the NetworkManager already through the NotificationService (only if calling it was not inside synchronized blocks...)

cdjackson, 2019-01-19:
> Just I understood you wanted also the dongles to protect themselves, and I did not see where to do it in common code :-)

Yes, this is what I normally like - ie for code to protect itself. I think in this case, given the code is all internal to the framework, we don't need to protect ourselves at this point, and so long as we have protection before any calls to listeners are made outside of the framework, then this should be fine.

You are right that in this specific case, there is already decoupling already - this is exactly what I was talking about, but I was looking at it from the other entry points as well (eg command and command status updates).

However, in the case of your problem, you had such an issue here, right? Since this is already decoupled, I wonder if your problem might not have been in the earlier code with the ```removeNode``` call?

> Just then there is an issue in the current implementation in the NetworkManager where the decoupling happens inside a synchronized block, which I think is what should be fixed, as we seem to agree that it is better to do the decoupling in common code.

I don't think I see a problem with the synchronisation here (although I agree that given this class is a copy before write, it is not needed). This code should not block - it should add all the notifications into the thread pool and return - this should not take any significant time, so the synchronisation should not matter. Any other threads that are created here that required the same lock should not start until this block finishes.

As I said, I will remove this synchronisation as it's not needed, but I don't think it is causing the problem you raised is it? (I'm happy to be corrected here :) ).

> So this is what I am asking whether it is necessary, as this is already (mostly) done in the NetworkManager already through the NotificationService (only if calling it was not inside synchronized blocks...)

Correct - the point being "mostly". I think there may just be one or two places where this is not done.

puzzle-star, 2019-01-19:
> However, in the case of your problem, you had such an issue here, right? Since this is already decoupled, I wonder if your problem might not have been in the earlier code with the `removeNode` call?

The removeNode call also has a "synchronized (this)" for notifications. But this is not the case. The issue I posted as example was in addNode, but I have seen it happen when updating and not just adding.

> As I said, I will remove this synchronisation as it's not needed, but I don't think it is causing the problem you raised is it? (I'm happy to be corrected here :) ).

Yes and no :)

The thread dump in my [post in OH forum](https://community.openhab.org/t/zigbee-binding/15763/1890?u=puzzle-star)  refers to addNode, and the issue is not really because of the "synchronized (this)" but inside it:

```
        if (networkStateSerializer != null) {
            networkStateSerializer.serialize(this);
        }
```

This blocks for a while, but as this thread owns the lock for (this) at this point, all other notifications that are also inside "synchronized (this)" branches get also blocked :-)

puzzle-star, 2019-01-19:
And looking more into it, as far as the synchronizations on (this) cannot block for too long, it would be fine to keep them (if needed).

So the root cause is the long waits inside the synchronized areas (such as the call to serialize() or anything that could block) and these are the ones to decouple, which should be easy by using the executorService already present in the NetworkManager... unless I find some for which we need the result code

cdjackson, 2019-01-19:
> This will remove the issue for other threads not being blocked by this one, but that single thread will still be blocked on the serialize() call,

Sure - that's understood and is a separate issue (#513).

> So the root cause is the long waits inside the synchronized areas

Again, that's another issue (#513). Yes, I know its the root cause of your problem, but in this issue we're talking about decoupling the network manager from the dongle, and in that respect, the synchronise is a problem.

>  these are the ones to decouple, which should be easy by using the executorService already present in the NetworkManager...

Again, we already have that decoupling here - but it didn't work due to the synchronise block around it...

> I can, but you will have to review anyway, as even if I will have a deep look into it, I might miss the implications (i.e. race conditions) of desynchronizing some parts of the code.

Ok, I will do the update...

closed 2019-10-20 by stale[bot]