zsmartsystems/com.zsmartsystems.zigbee #518: No EzspScanCompleteHandler after EzspStartScanRequest

TomTravaglino, 2019-01-28:
I observed that on network re-initialization (in `EmberNetworkInitialisation.formNetwork(...)`) often the `EZSP_ENERGY_SCAN` and the `EZSP_ACTIVE_SCAN` don't complete. This two requests are performed in a `EzspMultiResponseTransaction` which in both cases requires a `EzspScanCompleteHandler` to complete. But very often it just receives the so called `relatedResponses`.  When then the scan is over, the statements
```
EzspScanCompleteHandler scanCompleteResponse = (EzspScanCompleteHandler) scanTransaction.getResponse();
```
and
```
EzspScanCompleteHandler activeScanCompleteResponse = (EzspScanCompleteHandler) transaction.getResponse();
```
cause a `ClassCastException` because the response is one of the `relatedResponses` instead of the (missing) `requiredResponse`.

Is there a timing or duration problem with the scan?

I attached a log file which shows the `EZSP_ACTIVE_SCAN` not receiving the `EzspScanCompleteHandler`. 

[logs-no-scan-complete.txt](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/2803236/logs-no-scan-complete.txt)

cdjackson, 2019-02-04:
Please can you advise what version of the library you are using? I'm guessing it might be quite old as I see ```EzspNetworkStateRequest``` messages being sent when I think that the network state is offline. I don't think that this is a main cause though.

The other issue I see is that the binding seems to reset. This might be caused by something you are doing as there is an exception with ```com.xxx.smarthome.config.system.backup.impl.internal.BackupServiceImpl```. I think this is causing some sort of timeout, and then the dongle is reset and this is likely why the scan doesn't complete. Please check your implementation here.

TomTravaglino, 2019-02-05:
> Please can you advise what version of the library you are using?

It is version 1.1.9

> The other issue I see is that the binding seems to reset. This might be caused by something you are doing as there is an exception with [...]

We checked the Exception. What we do there is to call `networkManager.startup(true)` after the trustcenter information have been restored. The Exception happens in `com.zsmartsystems.zigbee.dongle.ember.internal.EmberNetworkInitialisation.doActiveScan(int)` in the line `EzspScanCompleteHandler activeScanCompleteResponse = (EzspScanCompleteHandler) transaction.getResponse();`. This is exactly the code part I initially described in this issue. We aren't doing any additional or explicit reset of the dongle.

I invite @triller-telekom to join this discussion as he is deeply involved in this issue.

triller-telekom, 2019-02-05:
My question would be:

What is `networkManager.startup(true)` supposed to do? I read that `true` will **re**-initialize the dongle.

Are you saying that this times out and fails? If so, how can we make sure that the call succeeds?

cdjackson, 2019-02-05:
What is the cause of the other exception in the log?

Also, what are you setting the trust centre value to?

triller-telekom, 2019-02-05:
That is what is set in the `ZigbeeNetworkManager`:

```
networkManager.setZigBeePanId(Integer.parseInt(panID));
        networkManager.setZigBeeExtendedPanId(new ExtendedPanId(extendedPanID));
        networkManager.setZigBeeChannel(ZigBeeChannel.create(Integer.parseInt(channel)));
        networkManager.setZigBeeLinkKey(new ZigBeeKey(linkKey));

        ZigBeeKey key = new ZigBeeKey(networkKey);
        key.setOutgoingFrameCounter(frameCounter);
        networkManager.setZigBeeNetworkKey(key);

        networkManager.startup(true);
```

Regarding the Exception: @TomTravaglino Can you please provide details to the "other" Exception? I am not sure which two Exceptions we are talking about. I thought its only about the `ClassCastException` due to the fact that the reply is not received in time.

TomTravaglino, 2019-02-05:
Before the ClassCastException happens there is another one
```
2019-01-28_13:10:23:149	NA	WARN:	[symbolicName=org.openhab.binding.zigbee.ember, className=org.openhab.binding.zigbee.ember.handler.ZigBeeSerialPortSPI] - IO exception when writing bytes
	at org.openhab.binding.zigbee.ember.handler.ZigBeeSerialPortSPI.write(ZigBeeSerialPortSPI.java:145)
	at com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler.outputFrame(SpiFrameHandler.java:537)
	at com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler.outputFrame(SpiFrameHandler.java:517)
	at com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler.connect(SpiFrameHandler.java:558)
	at com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp.initialiseEzspProtocol(ZigBeeDongleEzsp.java:885)
	at com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp.initialize(ZigBeeDongleEzsp.java:309)
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.initialize(ZigBeeNetworkManager.java:334)
	at org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler.initialiseZigBee(ZigBeeCoordinatorHandler.java:420)
	at org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler.access$4(ZigBeeCoordinatorHandler.java:380)
	at org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler$1.run(ZigBeeCoordinatorHandler.java:526)
```

which happens in org.openhab.binding.zigbee.ember.handler.ZigBeeSerialPortSPI.write(int) on the line `outputStream.write(outBuffer);`
```
    @Override
    public void write(int value) {
        byte[] outBuffer = getPacket(value);
        if (outBuffer == null || outputStream == null) {
            return;
        }
        try {
            outputStream.write(outBuffer);
        } catch (IOException e) {
            logger.warn("IO exception when writing bytes", e);
        }
    }
```

TomTravaglino, 2019-02-15:
@cdjackson @triller-telekom 

I checked the Exception mentioned in my previous post. The log doesn't show the entire message of the exception. This is the message I saw debugging the class `java.io.IOException: Device or resource busy in writeArray`.

And here the Exception with the top line
```
$java.io.IOException: Device or resource busy in writeArray
	at gnu.io.RXTXPort.writeArray(Native Method)
	at gnu.io.RXTXPort$SerialOutputStream.write(RXTXPort.java:1167)
	at org.openhab.binding.zigbee.ember.handler.ZigBeeSerialPortSPI.write(ZigBeeSerialPortSPI.java:145)
	at com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler.outputFrame(SpiFrameHandler.java:537)
	at com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler.outputFrame(SpiFrameHandler.java:517)
	at com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler.connect(SpiFrameHandler.java:558)
	at com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp.initialiseEzspProtocol(ZigBeeDongleEzsp.java:885)
	at com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp.initialize(ZigBeeDongleEzsp.java:309)
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.initialize(ZigBeeNetworkManager.java:334)
	at org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler.initialiseZigBee(ZigBeeCoordinatorHandler.java:421)
	at org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler.access$6(ZigBeeCoordinatorHandler.java:381)
	at org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler$1.run(ZigBeeCoordinatorHandler.java:531)
```

cdjackson, 2019-02-16:
I'm not really sure what I can do about this if there is an exception being thrown in the SPI driver. And it's also not part of this library... If there's something you can suggest here to make it more tolerant though, then I'm happy to accept a PR.

TomTravaglino, 2019-02-19:
Not sure if there really is a need to change that. I looked deeper into the code and saw that in the SpiFrameHandler there is a retry logic for incomplete transactions and as far as I could see the retry logic seems to work. My assumption is that the Exception is not the cause for the failing re-initialization of the network.

TomTravaglino, 2019-02-20:
@cdjackson @triller-telekom fyi
In the meantime I switched to ZSS version 1.1.11. The ClassCastException mentioned at the start of this issue is still thrown.
```
java.lang.ClassCastException: com.zsmartsystems.zigbee.dongle.ember.ezsp.command.EzspNetworkFoundHandler cannot be cast to com.zsmartsystems.zigbee.dongle.ember.ezsp.command.EzspScanCompleteHandler
	at com.zsmartsystems.zigbee.dongle.ember.EmberNcp.doActiveScan(EmberNcp.java:741)
	at com.zsmartsystems.zigbee.dongle.ember.internal.EmberNetworkInitialisation.formNetwork(EmberNetworkInitialisation.java:119)
	at com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp.startup(ZigBeeDongleEzsp.java:385)
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.startup(ZigBeeNetworkManager.java:512)
```

cdjackson, 2019-02-21:
I come back to my original point that your connection to the dongle appears to reset. I don't really know what is happening here and I think you need to work this out?

Why is it that half way through the scan, and just before the exceptions start, that the SPI port is reinitialised -:

```
2019-01-28_13:10:23:064	NA	DEBUG:	[symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeSerialPort] - Connecting to serial port [/dev/ttyspi0] at 115200 baud, flow control FLOWCONTROL_OUT_NONE.
2019-01-28_13:10:23:147	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - SPI: connect

2019-01-28_13:10:23:149	NA	WARN:	[symbolicName=org.openhab.binding.zigbee.ember, className=org.openhab.binding.zigbee.ember.handler.ZigBeeSerialPortSPI] - IO exception when writing bytes

2019-01-28_13:10:23:154	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - SpiFrameHandler thread started

2019-01-28_13:10:23:275	NA	ERROR:	[symbolicName=com.xxx.smarthome.config.system.backup.impl, className=com.xxx.smarthome.config.system.backup.impl.internal.BackupServiceImpl] - [BackupRestore] Fail status: 1 reason: com.xxx.binding.zigbee.backup

2019-01-28_13:10:23:944	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - TX EZSP: EzspVersionRequest [desiredProtocolVersion=4]
```

This seems to indicate that the system restarts. I have tested the Active Scan function, and it seems to work and the NCP sends the required response.  I'm not sure what code you are running, so it's a bit difficult to comment much further.

TomTravaglino, 2019-02-22:
We're working on it and probably found the reason. We will come back here when we are sure that we are right.

TomTravaglino, 2019-02-25:
I will try to explain how it comes to this issue:

- As already mentioned in https://github.com/zsmartsystems/com.zsmartsystems.zigbee/issues/518#issuecomment-460567192, when we do a restore of a backup we call `networkManager.startup(restart)` where `restart` is `true`.
- The restart makes the network state (`ZigBeeTransportState`) intermediately change to `OFFLINE`
-  When the network state changes all `ZigBeeNetworkStateListener` are notified
- `org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler` is a `ZigBeeNetworkStateListener` and will consequently be notified.
- In `ZigBeeCoordinatorHandler.networkStateUpdated(...)` when the network state is OFFLINE the reconnect job will be triggered which then tries to initialize the zigbee network.
- But as we already invoked `networkManager.startup(restart)` the initialization is already in progress and so this apparently ends up with a mess.

So from my point of view this issue hasn't really to do with ZSS, but rather with the binding.
To work around this issue we tried to set the bridge status to 'UNINITIALIZED' before starting the restore and it seems to work. But in our eyes this is a terrible approach. Maybe there is a better way to prevent the reconnect job from starting?

cdjackson, 2019-02-25:
Thanks @TomTravaglino.

I think this needs to be solved in the binding - however - maybe in ZSS we should have a check in the ```ZigBeeNetworkManager.startup()``` and/or ```ZigBeeNetworkManager.initialize()``` methods to return an error if the state is already open (at least ```ZigBeeNetworkManager,initialize()``` this should be possible - ```startup()``` may not be possible).

I would suggest we close this, and open another issue in OH, and possibly another one here to implement the check I mention above.

WDYT?

TomTravaglino, 2019-02-25:
I agree, we can close this one and create follow-up issues.

closed 2019-02-25 by TomTravaglino