zsmartsystems/com.zsmartsystems.zigbee #520: Heavy network traffic results in NPE with Telegesis dongle

hanneshofmann, 2019-02-04:
When using a Telegesis dongle while heavy network traffic is generated (e.g. by turning several smart plugs on/off at the same time), the following NPE's do occur:
```
Exception in thread "Thread-302" java.lang.NullPointerException
at java.util.concurrent.ConcurrentHashMap.putVal(ConcurrentHashMap.java:1011)
at java.util.concurrent.ConcurrentHashMap.put(ConcurrentHashMap.java:1006)
at com.zsmartsystems.zigbee.dongle.telegesis.ZigBeeDongleTelegesis$1.run(ZigBeeDongleTelegesis.java:670)
Exception in thread "Thread-308" java.lang.NullPointerException
at java.util.concurrent.ConcurrentHashMap.putVal(ConcurrentHashMap.java:1011)
at java.util.concurrent.ConcurrentHashMap.put(ConcurrentHashMap.java:1006)
at com.zsmartsystems.zigbee.dongle.telegesis.ZigBeeDongleTelegesis$1.run(ZigBeeDongleTelegesis.java:670)
Exception in thread "Thread-311" java.lang.NullPointerException
at java.util.concurrent.ConcurrentHashMap.putVal(ConcurrentHashMap.java:1011)
at java.util.concurrent.ConcurrentHashMap.put(ConcurrentHashMap.java:1006)
at com.zsmartsystems.zigbee.dongle.telegesis.ZigBeeDongleTelegesis$1.run(ZigBeeDongleTelegesis.java:670)
Exception in thread "Thread-318" java.lang.NullPointerException
at java.util.concurrent.ConcurrentHashMap.putVal(ConcurrentHashMap.java:1011)
at java.util.concurrent.ConcurrentHashMap.put(ConcurrentHashMap.java:1006)
at com.zsmartsystems.zigbee.dongle.telegesis.ZigBeeDongleTelegesis$1.run(ZigBeeDongleTelegesis.java:670)
Exception in thread "Thread-329" java.lang.NullPointerException
at java.util.concurrent.ConcurrentHashMap.putVal(ConcurrentHashMap.java:1011)
at java.util.concurrent.ConcurrentHashMap.put(ConcurrentHashMap.java:1006)
at com.zsmartsystems.zigbee.dongle.telegesis.ZigBeeDongleTelegesis$1.run(ZigBeeDongleTelegesis.java:670)
```

@cdjackson Any thoughts on that? Could it be that these NPE's will get fixed with https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/497?

cdjackson, 2019-02-04:
@hanneshofmann thanks - can you provide a log showing this problem please?

hanneshofmann, 2019-03-01:
Sorry for being so unresponsive on this issue, @cdjackson.

I am not able to provide a log because the issue is no longer reproducible for me with the develop branch. Was this perhaps fixed by the recent update of the Telegesis transaction timers (https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/542)?

Where do we want to take this issue? Shall we close this?

cdjackson, 2019-03-02:
Ok, let's close. This might be related to issues with the transactions. The latest transaction code has been heavily tested by an OH user with over 50 devices using the Telegesis dongle.

Any further issues though, just yell.

closed 2019-03-02 by cdjackson