zsmartsystems/com.zsmartsystems.zigbee #525: BindRequest transactions do not seem to ever complete

puzzle-star, 2019-02-09:
Hi @cdjackson,

After upgrading the library and the OH binding, most of my devices do not seem to ever report attribute changes. Maybe this is happeneing from before, so I cannot tell exactly from when this is happening.

Looking at the log, I see that even if the BindRequest command succeeds and I get the response, this response does not match the transaction:

```
2019-02-09 09:45:38.802 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 000B57FFFEA46B77: Initializing channel zigbee:device:04000AB3:000b57fffea46b77:000B57FFFEA46B77_1_dimmer with org.openhab.binding.zigbee.internal.converter.ZigBeeConverterSwitchLevel@c27612
2019-02-09 09:45:38.804 [DEBUG] [zigbee.transaction.ZigBeeTransaction] - Sending transaction: BindRequest [null -> 15773/1, cluster=0021, TID=NULL, srcAddress=000B57FFFEA46B77, srcEndpoint=1, bindCluster=8, dstAddrMode=3, dstAddress=000D6F001113A331, dstEndpoint=1] ==== BindRequest [null -> null, cluster=0021, TID=NULL, srcAddress=null, srcEndpoint=null, bindCluster=null, dstAddrMode=null, dstAddress=null, dstEndpoint=null]
2019-02-09 09:45:38.806 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: BindRequest [0/0 -> 15773/1, cluster=0021, TID=D3, srcAddress=000B57FFFEA46B77, srcEndpoint=1, bindCluster=8, dstAddrMode=3, dstAddress=000D6F001113A331, dstEndpoint=1]
2019-02-09 09:45:38.808 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=15773/0, profile=0000, cluster=0021, addressMode=DEVICE, radius=31, apsSecurity=false, apsCounter=D3, payload=00 77 6B A4 FE FF 57 0B 00 01 08 00 03 31 A3 13 11 00 6F 0D 00 01]
2019-02-09 09:45:39.258 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: BindResponse [15773/0 -> 0/0, cluster=8021, TID=NULL, status=SUCCESS]
2019-02-09 09:45:47.240 [DEBUG] [zigbee.transaction.ZigBeeTransaction] - Transaction cancelled: BindRequest [0/0 -> 15773/1, cluster=0021, TID=D3, srcAddress=000B57FFFEA46B77, srcEndpoint=1, bindCluster=8, dstAddrMode=3, dstAddress=000D6F001113A331, dstEndpoint=1]
2019-02-09 09:45:47.242 [DEBUG] [converter.ZigBeeConverterSwitchLevel] - 000B57FFFEA46B77: Failed to bind level control cluster

```

The issue can be tracked up to "starting" here:

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/d7398dd5edcfb5f78a3e367a82146ea7d73d95a2/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/zcl/ZclCluster.java#L474

The ResponseMatcher is an empty BindRequest, and it seems not to be ever populated (not even in ZigBeeEndpoint.sendTransaction(). This can be seen in the above log.

So the later call to BindRequest.isTransactionMatch():

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/1a8f8a25a528a6bca540c973db4b785148ac49aa/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/zdo/command/BindRequest.java#L269

Will always return false (unless I am missing something).

But I have another concern: even if it this is fixed, in the example above the match will still fail, as the command goes to: "15773/1" but the response comes from "15773/0".

Not sure this second issue is on the response, but maybe on the command: should not the ZDO command go to endpoint 0, instead of endpoint 1?

If this is right, reporting must be broken for many people (?), depending on the implementation. i.e in OH (as I see it in my logs):
 - SwitchLevel channel converter sends a ConfigureReporting command only if the binding command works --> it never gets configured
 - ColorColor channel converter sends a ConfigureReporting command even if the binding command works --> it is working

puzzle-star, 2019-02-09:
Hi @cdjackson,

I have just sent a PR to fix this. Please let me know if it looks OK to you.

Pedro

cdjackson, 2019-02-09:
Thanks for this @puzzle-star. I hadn't noticed a problem with this, but have seen errors with the responses which I'd not had time to chase, so thanks for resolving this.

closed 2019-02-09 by cdjackson

puzzle-star, 2019-02-09:
Hi @cdjackson, 

My whole home was not reporting :-) So I decided to look into it.

I thought it was an issue with my network, so I have been several weeks with an OH rule that forces a refresh if after some seconds of sending a command there is no update. :-)

But once seen in the logs...