zsmartsystems/com.zsmartsystems.zigbee #532: Disable join not working properly

milansamardzic, 2019-02-13:
Disable join not working properly. 

Found and fix the issue. I will create a pull request. 
This ticket is only for tracking status.

cdjackson, 2019-02-14:
Please can you describe the problem that you don't really provide any explanation about what doesn't work?

cdjackson, 2019-03-02:
> Disable join not working properly.

Please can you advise your problem, and your system information as there is currently no known issue here - at least with Telegesis and Ember dongles.

cdjackson, 2019-03-30:
@milansamardzic do you have any further information or shall we close this?

cdjackson, 2019-04-18:
This has been open for 2 months with no feedback, so I'm closing this for now. If you have further information, please post it and we can discuss again!

closed 2019-04-18 by cdjackson