zsmartsystems/com.zsmartsystems.zigbee #534: Wrong datatypes configured in thermostat cluster

hsudbrock, 2019-02-14:
When testing the shiny new support for thermostats in the openHAB ZigBee binding, I discovered that some of the datatypes for the thermostat cluster's attributes are incorrectly defined (resulting in failing reporting configuration, as my thermostat replies with `INVALID_DATA_TYPE` when configuring, e.g., local temperature reporting).

The issue is in the current markdown files (https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/master/com.zsmartsystems.zigbee.autocode/src/main/resources/zcl_definition.md#thermostat-0x0201), and also in the new XML files from https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/527.

The following attributes should have a *signed* type (according to the ZCL document I looked at), but this library currently configures them as *unsigned*:
* LocalTemperature
* OutdoorTemperature
* AbsMinHeatSetpointLimit
* AbsMaxHeatSetpointLimit
* AbsMinCoolSetpointLimit
* AbsMaxCoolSetpointLimit
* LocalTemperatureCalibration
* OccupiedCoolingSetpoint
* OccupiedHeatingSetpoint
* UnoccupiedCoolingSetpoint
* UnoccupiedHeatingSetpoint
* MinHeatSetpointLimit
* MaxHeatSetpointLimit
* MinCoolSetpointLimit
* MaxCoolSetpointLimit
* MinSetpointDeadBand

Similarly, some types in the command signatures should be *signed* rather than *unsigned*:
* 'Heat Set' and 'Cool Set'  in the 'Set Weekly Schedule' command (and the corresponding response)
* 'Local temperature'  and 'Setpoint'  in the 'Get Relay Status Log' response

cdjackson, 2019-02-15:
Thanks @hsudbrock I will update these in https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/527

cdjackson, 2019-02-15:
I've updated the definition in #527 - hopefully I got all the ones you listed...

hsudbrock, 2019-04-03:
I think you missed the following:
* AbsMaxCoolSetpointLimit
* 'Heat Set' and 'Cool Set' in the 'Set Weekly Schedule' command
* 'Local temperature' in the 'Get Relay Status Log' response

mlriess, 2019-04-12:
I just had a look at https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/dev-1.2.0/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/zcl/clusters/ZclThermostatCluster.java and it seems like most of the attributes are still unsigned integers. Am I perhaps looking in the wrong place?

cdjackson, 2019-04-12:
I still need to check the additional attributes that Henning pointed out, but yes, you are looking in the wrong place - see the link two comments above to #527.

mlriess, 2019-04-12:
Thanks for the clarification! I need to get more familiar with the branches ...

cdjackson, 2019-04-13:
> I need to get more familiar with the branches

Normally there shouldn't be too many branches, but just at the moment I have this ```dev-1.2.0``` branch to avoid merging multiple breaking changes. You did reference this above, but the point is that the updated XML autogenerator is not currently merged into this branch.

cdjackson, 2019-04-18:
Closed by #527

closed 2019-04-18 by cdjackson