zsmartsystems/com.zsmartsystems.zigbee #537: Add possibility to remove a node that does not respond to Leave requests

hsudbrock, 2019-02-15:
This issue relates to https://github.com/openhab/org.openhab.binding.zigbee/issues/190, where the idea is to add some option to "force remove" an openHAB thing for which the ZigBee node does not respond to leave requests.

I think that it might not be straightforward to implement this in openHAB (as the ESH 'force leave' method will simply remove the thing from the thing registry, without giving the thing handler the chance to do some force-removal from its internal data structures).

However, I think that it would already be good to have that functionality in this library, including a console command for "force-removing" a node.

My suggestion would be to add a method `leave(final Integer destinationAddress, final IeeeAddress leaveAddress, boolean forceRemove)` to the `ZigBeeNetworkManager`; for `forceRemove==true` it would remove the node from the list of nodes, even if the node does not respond to the `Leave` command. The existing `leave(destinationAddress, leaveAddress)` would then simply call `leave(destinationAddress, leaveAddress, false)`.

What do you think, @cdjackson ? If you don't have immediate concerns, I could create a PR for this change.

cdjackson, 2019-02-15:
Can you describe what this actually is going to do? There are already methods to make a device leave the network, although there are (from memory) 3 different ways to do this.  I'm just not clear what you are actually wanting to do.

How does it "force the device to leave"? If it is a battery device, then it's not even listening. There is no central repository of devices that are part of the network (unlike some protocols).

cdjackson, 2019-02-15:
@hsudbrock you might want to read the following before answering my question -:

https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2016/07/14/ways_to_tell_zigbee-kfzE

hsudbrock, 2019-02-15:
It will not force the device to leave the network (as you say, that's not possible when the device is not listening). It will only remove the device from the list of nodes maintained by the library - if the device pops up again later, it will get back into that list.

Maybe "force remove" is not the best name.

Let me create a PR to illustrate what I mean.

hsudbrock, 2019-02-15:
Thanks for the pointer, Chris, I will take a look first.

cdjackson, 2019-02-15:
> It will only remove the device from the list of nodes maintained by the library

Hmmm - I thought it was already possible to remove a node from the node list if that's what you want to do? I'll have a look.

cdjackson, 2019-02-15:
Yep -:

```
    public void removeNode(final ZigBeeNode node) {
```

What is wrong with this? If all you want is to remove the device from the list of nodes, then that is what this does.

hsudbrock, 2019-02-15:
I must have overlooked this, sorry for the hassle, Chris. I think that I will simply add a console command for removing the node then (unless I find such a command already present as well when implementing :) ).

cdjackson, 2019-02-15:
Sure - that sounds fine...

cdjackson, 2019-02-15:
@hsudbrock what do you think about adding a "remove" option to the existing ```node``` command...

eg

```
node 0x1234 remove
```

rather than adding a new ```removenode 0x1234``` command?

hsudbrock, 2019-02-15:
@cdjackson I think that your suggestion is better than adding a `removeNode` command.

But: What do you think about extending the console `leave` command with an optional parameter, such that the command first attempts to make the node actually leave the network, and - if this is not successful - at least removes the node from the node list?

eg
```
leave -forceNodeRemoval 0x1234
```

I would prefer this, because usually one tries to remove the node from the network by sending a `Leave` command, and only if this fails one has to resort to remove the node from the node list.

cdjackson, 2019-02-15:
That’s fine with me Henning.

hsudbrock, 2019-02-15:
OK, I will implement that (and we can still change if necessary during PR review...)

PR #540 (2019-02-19): Console: Add an option to remove a node after unsuccessful leave attempt
Resolves #537 

As discussed in #537, I have added an option to the `Leave` CLI command that will first try to make a node leave the network, and, if the node does not respond to the leave request, removes the node from the list of nodes in the network manager.

I tried to implement this by first calling `networkManager.leave`, and , if this fails, calling `networkManager.removeNode`. But I did not see a way to find out whether the `leave` call succeeded, except setting a timeout and checking whether the node is still the [... cut]
commit 8f148024 (2019-02-19): Console: Add an option to remove a node after unsuccessful leave attempt (#540)

Console: Add an option to remove a node after unsuccessful leave attempt

Resolves #537
closed 2019-02-19 by cdjackson