zsmartsystems/com.zsmartsystems.zigbee #538: Reduce keep-alive polling in Ember dongle

cdjackson, 2019-02-16:
Currently there is polling in the Ember dongle that polls every second in order to check if the NCP is still alive. On large systems, this accounts for a considerable load and we should look at reducing this to improve efficiency.

@triller-telekom what do you think? Either we don't poll when there are frames in the queue, or we don't poll if frames have been passed through the dongle layer within x period?

triller-telekom, 2019-02-20:
> Either we don't poll when there are frames in the queue

Sounds plausible and we should do that.

> we don't poll if frames have been passed through the dongle layer within x period

Hmm, I am not entirely sure what is the best solution here. But at a first glance I would say that maybe we should just increase the polling period to 2 or 3 seconds. Int hat way it will take longer until the device goes offline, but should still be sufficient.

cdjackson, 2019-02-20:
In order to keep the noise down, in the Telegesis implementation it checks to see if there are messages queued, and if so, it doesn't send the poll as it is just overhead.

I know that the Telegesis is implemented a little differently in that the polling is done in the lower layer handler, but I think that doing something similar is best to remove any unnecessary noise.

When we added this to Telegesis, Thomas would not accept a value above 1 second....

PR #546 (2019-02-24): Reduce unnecessary polling of Ember dongle state when network is busy
This PR will prevent polling of the Ember dongle status if a command has been sent to the dongle within the ```pollRate``` period. This eliminates any unnecessary polling while still achieving detection of an offline dongle with the same ```pollRate``` period.
Closes #538
closed 2019-02-24 by cdjackson