zsmartsystems/com.zsmartsystems.zigbee #544: Waiting for Ember bootloader menu prompt fails on slow systems

hanneshofmann, 2019-02-22:
I just tried to update the firmware image of an Ember coordinator (EM3587) on a slow system and the update failed because waiting for the bootloader menu prompt was unsuccessful:
```
2019-02-21T14:16:57.714+0100   678  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.EmberFirmwareUpdateHandler] - Ember bootloader: Failed waiting for menu before transfer.
```
The `EmberFirmwareUpdateHandler` currently tries _five times_ to access the prompt and it waits for _1500ms_ on each try. These timeouts seem to be suitable when the dongle is connected to my local machine. However, the update fails on slow systems. By increasing the amount of retries and/or the sleep time, the update was also successful on these slow systems. 

@cdjackson Is there a chance to make these timeouts configurable or increasing them in general?

cdjackson, 2019-02-22:
It should be ok to increase the timer - how long is it taking on the slower machine?

hanneshofmann, 2019-02-22:
The bootloader menu prompt was found after increasing the timeout to 5 seconds. However, I believe that a smaller value shall work too. 

I will measure the time more precisely as soon as I get my Ember dongle working again. The dongle stopped working after transferring the image and loading it (accessing BL via CLI still works but `ebl info` claims that no application is installed anymore).

cdjackson, 2019-02-22:
Thanked Hannes. 5 seconds definitely sounds a little long. I would have expected maybe 2 or 3 so let’s see what is needed.

hanneshofmann, 2019-02-22:
Based on 10 executions, the average time was 16 seconds (between opening the serial port and waiting for the bootloader prompt) and the worst case took 22 seconds. 

Because the implementation retries 5 times with a sleep time of 1500ms (+250ms for reading data from the port), I would actually purpose to increase the sleep time to 5000ms as 3000ms would only fit for the average case. The consequence of increasing the sleep time to 5000ms is that the maximum total time would increase from 8750ms up to 27250ms.

I suppose that increasing the sleep time to 5000ms does not sound reasonable? If that is the case, would you prefer making the timeout configurable?

cdjackson, 2019-02-22:
Surprising…

Can you describe what this time is between? Maybe we need a longer delay for entry to the bootloader?

I’m also not really sure why this should be that system dependant. Even on a slow computer, we’re always sending at the same data rate (115k2 for the bootloader) and I don’t really understand why even a slow machine should take 20 seconds or more unless something is blocking?

I think it would be good to try and understand this a bit more before we implement such timers - can you provide some logs so I can see what is going on?

I’m also not against making it configurable, but then there needs to be a way for applications to change it… Let’s try and understand this a bit more first?

hanneshofmann, 2019-02-22:
>Can you describe what this time is between?

The time between is [T2](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/master/com.zsmartsystems.zigbee.dongle.ember/src/main/java/com/zsmartsystems/zigbee/dongle/ember/internal/EmberFirmwareUpdateHandler.java#L114) ("got BT prompt") - [T1](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/master/com.zsmartsystems.zigbee.dongle.ember/src/main/java/com/zsmartsystems/zigbee/dongle/ember/internal/EmberFirmwareUpdateHandler.java#L106)("serial port opened").

>Maybe we need a longer delay for entry to the bootloader?

If you like, I could try to perform some executions with an additional sleep time before the BT prompt gets entered. Additionally, I could also verify the behaviour using a Telegesis dongle (but I assume it behave similar slow).

>I think it would be good to try and understand this a bit more before we implement such timers - can you provide some logs so I can see what is going on?

Yes, I agree. A [log](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/2895193/ember-update.log) file is attached. Please let me know if further information are required.

cdjackson, 2019-02-23:
Thanks Hannes. The log is not super useful - can you enable trace logging please? It is surprising to me that it takes 15 seconds to respond, and that this is a function of a slow computer unless something is blocking the serial port, and in this case, I would have expected a lot more issues if data is taking 15 seconds to be received by the serial port (ie ZigBee won’t really work at all).

cdjackson, 2019-05-23:
Closing as no activity for 3 months.

closed 2019-05-23 by cdjackson