zsmartsystems/com.zsmartsystems.zigbee #548: Add possibility to recognize if a network initialization/creation is in progress

TomTravaglino, 2019-02-25:
As discussed in #518 we want to avoid multiple initialization of the network. Therefore we propose to introduce a possibility to check if an initialization is already in progress.

The idea is to introduce a flag `initializing` in the network manager.
- The flag must be set to `true` when `ZigBeeNetworkManager.initialize()` and/or `ZigBeeNetworkManager.startup(boolean)` are called and set back to `false` when these methods are done.
- There must be a public method to check the flag

We want to use this flag in the binding in order to skip any other initialization.

cdjackson, 2019-02-25:
As we discussed, this should be handled in the application in the first instance.

We can add a check to prevent calling ```initialize()``` multiple times, and return error if so.  However, it is valid to call ```startup()``` multiple times, so this can not be prevented.

triller-telekom, 2019-02-25:
Is it also valid to call `startup(true)` multiple times? Because this would lead to multiple initializations too if I am not mistaken...

cdjackson, 2019-02-25:
Sure - why not?  It is valid to reinitialise the network.

TomTravaglino, 2019-02-25:
But then multiple re-initialization could be triggered simultaneously. Would this not be a problem?

cdjackson, 2019-02-25:
It might be possible to stop simultaneous calling of these tasks, but that may or may not work for all dongles if it spawns other tasks.

I will take a look, but I think in the case that we discussed earlier, the application should not be doing this in the first place.

cdjackson, 2019-02-25:
I have created #550 to check that the state is correct when calling these methods. It doesn't check that the methods aren't already in progress - I think that this is something that has already gone wrong in your application code if you try to initialise the network twice at the same time from different threads.

TomTravaglino, 2019-02-26:
> It doesn't check that the methods aren't already in progress - I think that this is something that has already gone wrong in your application code if you try to initialise the network twice

Let me clarify that. In our code the network is restarted only once. The second thread doing that too is the reconnect job in the ZigBeeCoordinatorHandler which is started when the network state changes to OFFLINE.

cdjackson, 2019-02-26:
> The second thread doing that too is the reconnect job in the ZigBeeCoordinatorHandler which is started when the network state changes to OFFLINE.

That is also a thread in your code isn't it?  It's not part of this library is it? So your application code is calling the initialise method twice as far as I can see?

TomTravaglino, 2019-02-26:
> That is also a thread in your code isn't it? It's not part of this library is it?

It's neither part of our code nor of ZSS, it's in the ZigBee binding [see here](https://github.com/openhab/org.openhab.binding.zigbee/blob/b6d5e6125041493d2db3c29bf7c8c96112225a7a/org.openhab.binding.zigbee/src/main/java/org/openhab/binding/zigbee/handler/ZigBeeCoordinatorHandler.java#L512)

cdjackson, 2019-02-26:
Sure - I understand that, but as I said above, it's not part of this library. I suggested in the other discussion that this should be resolved in the binding.

Maybe I'm missing something - what is your suggestion if you want it to be solved in this library?

cdjackson, 2019-02-28:
@TomTravaglino just wondering if you saw my comment?

TomTravaglino, 2019-03-01:
@cdjackson I'm discussing this with @triller-telekom. We will come back to you soon.

cdjackson, 2019-03-01:
Wouldn’t it be better to continue the discussion we started here?

TomTravaglino, 2019-03-01:
The discussion has to do with https://github.com/openhab/org.openhab.binding.zigbee/issues/398#issuecomment-468583376. That's why I was discussing this with @triller-telekom

cdjackson, 2019-03-01:
Ok, but it's all inter-related, so it just seems better to have the discussion in an open forum. You are trading where you want to put the functionality - in the library, or in the binding and I think this should be discussed openly.

cdjackson, 2019-04-18:
Closed by #590

PR #557 (2019-03-10): Introduce ZigBeeNetworkState enum to separate transport and network states
This separates the state of the transport layer from the network state. It introduces a new ```ZigBeeNetworkState``` enum, and the ```ZigBeeNetworkStateListener``` now receives updates with this new enum and not the ```ZigBeeTransportState```.

When refactoring it is noted that state names remain the same between the two - just the class name changes.

Closes #548 
Closes #554
closed 2019-04-18 by cdjackson