zsmartsystems/com.zsmartsystems.zigbee #553: Nodes cli command issues getting basic information

cdjackson, 2019-02-28:
Following #541 if there are issues getting the basic information, then the nodes cli command gets spread over a long duration making it difficult to use now when we just want to see the list of nodes. I would suggest that the BASIC information that was added in this change should not be requested by default, and we should have an option to request it.

This could be implemented with a Long.MAX_VALUE as the refresh time to allow the information to be displayed if it is already known, but not requested if it isn't.

@hsudbrock WDYT?

hsudbrock, 2019-03-01:
Hm, I think you are right, this is a drawback of my changes that does not really stand out when using the openHAB binding (as this always requests these values at discovery time).

I agree with your suggestion for improvement (let me create a PR later for adding this).

hsudbrock, 2019-03-01:
> This could be implemented with a Long.MAX_VALUE as the refresh time to allow the information to be displayed if it is already known, but not requested if it isn't.

This is actually already implemented like this; it does not help though, because even with a refresh time of `Long.MAX_VALUE` the attribute value is requested from the device if the value is not already known.

I will use a flag for the command to improve the situation.

PR #555 (2019-03-01): CLI nodelist: Do not force read of modelId/manufacturer
This resolves #553.

Unlike what I wrote in #553, I did not introduce a flag for the command; instead, the command now displays the last value of the model ID and the manufacturer (unless these values were never read from the device, in which no model ID / manufacturer is displayed in the node list for the device).

If one wants to have model ID and manufacturer in the nodelist output, and they have not yet been read, one simply has to execute a `read <endpoint> 0 4 5` command to read both attributes once.
closed 2019-03-01 by cdjackson