zsmartsystems/com.zsmartsystems.zigbee #558: Adding support for MOXA nPort Z3150 gateway

TheWizz, 2019-03-03:
I'm looking at adding support for the industrial-grade MOXA ZigBee-to-Ethernet gateway. This gateway (which is available in Ethernet or serial port flavors) appears to be rather low level. Merely acting as a "gridge" between the Ethernet/serial on one side and ZigBee on the other. So it has no internal database of nodes or such. I get the impression that this would be a rather nice fit for this library, as it appears to handle all other aspects needed. Looking at the code, I see a serial port interface being used by some of the implementations (e.g., ZigBeeDongleTiCc2531, ZigBeeDongleXBee, ZigBeeDongleConBee, etc). So adding a "driver" for the MOXA interface seems as it would be quite doable. However, the docs for the MOXA is rather sparse when it comes to details around ZigBee. It seems to set things like the PAN ID, channel, etc using a web based interface, described on page 33 in the manual found here:

http://support.elmark.com.pl/moxa/products/Serwery_portow_szeregowych/NPortZ3150/Manual/NPort_Z3150_Series_Users_Manual_v1.pdf

There's really no description of the data packet sent through the MOXA to the ZigBee network, giving me the impression this is little more than a pass-through.

Is there anything else of what's in the project now that would seem like a good starting point for adding support for the MOXA gateway? Any other "dongle" or serial port based device that works in a similar manner?

-JM

cdjackson, 2019-03-03:
You would clearly need to know the API, and as you say, the documentation is sparse as it's really a user manual for configuring via the web interface and not an API manual.

TheWizz, 2019-03-03:
Well I don't really think there _is_ much of an API here. Looking at the block diagram of the MOXA device, there's a TI CC2530 chip inside. MOXA have another model (nPort Z2150) which is just this same CC2530 chip directly wired to a serial port on the box (with equally sparse documentation). I've used other MOXA nPort devices in the past, and they're just "pass-through" (in those cases, a TCP port passed through in both directions to a serial port). So I believe the "protocol" used to talk to the MOXA Z3150 should be pretty much identical to the ZigBeeDongleTiCc2531 (which talks through a serial port as far as I can tell). The only difference between the CC2531 and the CC2530 is that the former also contains a USB interface (preumably typically used to emulate a serial port), while the latter provides a raw seraial port.

So my plan is to try to get hold of the MOXA box, and then try to adapt the ZigBeeDongleTiCc2531 code to talk through a TCP port instead of a serial port, and possibly strip out some initialization data (e.g., the PAN address), which seems to be handled by the MOXA box as part of its initialization sequence.

-JM

cdjackson, 2019-03-03:
Call it an API or not - you need to know how to set things like keys, PAN, EPAN, install codes, etc. If it’s just passing through the data to the 2351, and you can assume the 2531 API, then great, but somehow you need to know if that’s what it’s doing.

Good luck.

If you create a new driver, then feel free to provide a PR here for it.

TheWizz, 2019-03-03:
Will do!

cdjackson, 2019-05-07:
I understand that this is not being progressed due to hardware availability, so I will close this for now. If this is picked up again, please feel free to reopen.

closed 2019-05-07 by cdjackson