zsmartsystems/com.zsmartsystems.zigbee #562: ConBee options

TheWizz, 2019-03-04:
I'm trying to get the ConBee dongle up and running from the console. I'm on a Mac, using the following options:

-dongle CONBEE -port /dev/cu.usbserial-DM00ZXK0 -baud 38400 -flow hardware -reset

I've also tried 115200 baud as well as no handshake. I've not installed any driver, as this appears to be built in (when plugged in, it appears under "/dev/cu.usbserial-DM00ZXK0"). Running with the above configuration ends with a "CONBEE TX: TIMEOUT" message, indicating that no response is received from the dongle (see below).

Any pointers to how to get the ConBee working would be appreciated.

-JM

Initialising ZigBee console...
16:48:19.331  DEBUG  Adding supported cluster 1280
16:48:19.334  DEBUG  ClusterMatcher starting
16:48:19.356  DEBUG  ClusterMatcher adding cluster IAS_ZONE
16:48:19.357  DEBUG  Adding supported cluster 25
16:48:19.357  DEBUG  ClusterMatcher adding cluster OTA_UPGRADE
16:48:19.392  DEBUG  ConBee transport initialize
16:48:19.392  DEBUG  Network state is updated to INITIALISING
16:48:19.394  DEBUG  Ignoring invalid network state transition from INITIALISING to UNINITIALISED
ZigBee network state updated to INITIALISING
16:48:19.394  DEBUG  Opening port /dev/cu.usbserial-DM00ZXK0 at 38400 baud with FLOWCONTROL_OUT_RTSCTS.
16:48:19.414  DEBUG  ConBeeReceiveHandler thread started
16:48:21.918  DEBUG  TX CONBEE queue: 1
16:48:21.919  DEBUG  TX CONBEE queue: 1 sync
16:48:21.919  DEBUG  TX CONBEE queue: 1 send
16:48:24.205  DEBUG  CONBEE TX: DeviceStateRequest [sequence=1]
16:48:27.454  DEBUG  CONBEE TX: 07 01 00 08 00 00 00 00 F0 FF
16:48:27.457  DEBUG  TX CONBEE queue: 1 exit
16:48:27.711  DEBUG  CONBEE TX: TIMEOUT

cdjackson, 2019-03-04:
As per https://github.com/zsmartsystems/com.zsmartsystems.zigbee/issues/294 this is not completely working - it would be great if you can get it up and running though.

TheWizz, 2019-03-04:
Yes, I'd very much like to too ;-). However, I don't seem to get anything working *at all*. I get the impression from #294 that it could be made to work in some fashion, but perhaps isn't entirely stable. So any pointer as to how to configure the thing in order to make it talk to a HUE system would be very much appreciated. What parameters fo I need to set? Do the parameters I currently use make sense? Any additional parameters that I should add?

-JM

cdjackson, 2019-03-04:
It looks ok I think, and it should work. I had the initial implementation working, but this was a long time ago now so I'm not really familiar enough to provide a lot of help as I now mainly support Telegesis and Ember, with some support for TI.

TheWizz, 2019-03-04:
OK, I made some further progress getting the ConBee to respond using the following options:

`-dongle CONBEE -port /dev/cu.usbserial-DM00ZXK0 -baud 38400 -flow software -reset
`

This starts up the console with the following output:

`Initialising ZigBee console...
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.thoughtworks.xstream.core.util.Fields (file:/Users/mikefahl/.m2/repository/com/thoughtworks/xstream/xstream/1.4.9/xstream-1.4.9.jar) to field java.util.TreeMap.comparator
WARNING: Please consider reporting this to the maintainers of com.thoughtworks.xstream.core.util.Fields
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
ZigBee network state updated to INITIALISING
networkManager.initialize returned SUCCESS
ZigBee network state updated to ONLINE
Node Added ZigBeeNode [state=UNKNOWN, IEEE=00212EFFFF03B184, NWK=0000, Type=COORDINATOR]
PAN ID          = 17955
Node Added ZigBeeNode [state=ONLINE, IEEE=00212EFFFF03B184, NWK=0000, Type=COORDINATOR]
Extended PAN ID = 00212EFFFF03B184
Channel         = CHANNEL_15
*** Resetting network
  * Channel                = 11
  * PAN ID                 = 1
  * Extended PAN ID        = 0000000000000000
  * Link Key               = 5A6967426565416C6C69616E63653039
  * Network Key            = 8D30CB545B2306556D52491805A75D0F
ZigBee console starting up ... [OK]
ZigBee console ready.
PAN ID          = 17955
Extended PAN ID = 00212EFFFF03B184
Channel         = CHANNEL_15
`

I can then use the _nodes_ command which displays:

`
Network  Addr       IEEE Address   Logical Type   EP                    Profile      Device Type     Manufacturer            Model
      0  0000   00212EFFFF03B184    COORDINATOR    1     ZIGBEE_HOME_AUTOMATION  CONFIGURATION_TOOL                                  
                                                  80                       null  LEVEL_CONTROL_SWITCH                                  
`

I have a very simple system consisting of only the ConBee and an Innr mains switch plug. Presumably that's what's listed as the "LEVEL_CONTROL_SWITCH" above.

Im now looking for the commands needed to add this node to the ConBee network "database" and then send commands to it to turn it on and off. This would seem like basic things to do, but the list of commands supported by the console application doesn't really tell me how to proceed here. So what would be my next step in getting this going?

-JM

cdjackson, 2019-03-04:
There is no such “conbee network database” - all the work is done by the Java firmware once the device is accepted by the trust centre.

TheWizz, 2019-03-04:
FYI, the main issue with getting the ConBee to "talk" was probably the baudrate (38400). I don't know whether the ConBee supports other data rate, and if so how to configure this.

I apologize for the confusion. By "conbee network database" I meant the "CONBEE-simple-network.xml" file, created when running the console. Presumably this holds the configuration of all nodes added to the system. So what I'm looking for now is where to find the commands used to discover and add nodes, as well as controlling them (on/off/brightness/temperature, etc). 

-JM

cdjackson, 2019-03-04:
Try the documentation -:
https://github.com/zsmartsystems/com.zsmartsystems.zigbee#console-commands <https://github.com/zsmartsystems/com.zsmartsystems.zigbee#console-commands>

This lists most of the commands - you can also use the help command to get a list of commands.

Note that this is not meant to be a fully featured, user orientated application - it is meant for development of the library such that the library can be used in another product. Don’t expect it to do everything you want.

TheWizz, 2019-03-04:
Yes, I've looked at the documentation (of course), as well as the more up-to-date help. But there are a lot of commands, and it isn't very clear which ones are used to add a few devices to the system and to control them in a basic manner. My goal is to use this as a library in our software product, but before starting to integrate it into our product, I would like to get at least a basic understanding of the procedure here. Experimenting with that using the console is probably easier than doing it with code. So perhaps you could give me a few hints on how I can accomplish this using the console, to get a better understanding of how the library works.

The nodes command currently lists two "endpoints" (with number 1 and 80). I believe one of those is the ConBee itself, an the other one the Innr power outlet switch. So those would be the ones I want to get control over to begin with. I would expect some command to "adopt" those into the system (resulting in them being added to the "CONBEE-simple-network.xml"  file when the node tree is serialized). I would then expect the ability to turn the Innr switch on/off (it appears to behave as a light, using the Light Link part of the protocol).

-JM

cdjackson, 2019-03-04:
Well, while the documentation is light, if you understand ZigBee it should be sufficient. You have the join command to open joining, the leave command to make a device leave. There are commands to read information about the device (the info command). There is an on and an off command - these send the on and off commands to a device. There is a level command that sends the level control command to a device.

I think the documentation should be self descriptive if you understand ZigBee at least a little. If not, you might want to start with a read of some ZigBee docs?

If you want commercial assistance, then we can also provide that.

cdjackson, 2019-04-13:
Closing this following offline discussions. Feel free to reopen if there are further issues.

closed 2019-04-13 by cdjackson