zsmartsystems/com.zsmartsystems.zigbee #582: com.zsmartsystems.zigbee.serialization.DefaultDeserializer.readZigBeeType Nullpointer when using openhab2/aqara cube

schwabe, 2019-03-27:
I opened a [ticket against the OpenHab 2 binding](https://github.com/openhab/org.openhab.binding.zigbee/issues/419) before but was pointed here as it seems to be a bug in this library. The interesting part is the traceback:

```
2019-03-27 12:58:42.659 [ERROR] [e.ember.internal.ash.AshFrameHandler] - AshFrameHandler Exception processing EZSP frame: 
java.lang.NullPointerException: null
	at com.zsmartsystems.zigbee.serialization.DefaultDeserializer.readZigBeeType(DefaultDeserializer.java:72) ~[239:com.zsmartsystems.zigbee:1.1.6]
	at com.zsmartsystems.zigbee.zcl.field.AttributeReport.deserialize(AttributeReport.java:100) ~[239:com.zsmartsystems.zigbee:1.1.6]
	at com.zsmartsystems.zigbee.zcl.ZclFieldDeserializer.deserialize(ZclFieldDeserializer.java:73) ~[239:com.zsmartsystems.zigbee:1.1.6]
	at com.zsmartsystems.zigbee.zcl.clusters.general.ReportAttributesCommand.deserialize(ReportAttributesCommand.java:88) ~[239:com.zsmartsystems.zigbee:1.1.6]
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.receiveZclCommand(ZigBeeNetworkManager.java:808) ~[239:com.zsmartsystems.zigbee:1.1.6]
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.receiveCommand(ZigBeeNetworkManager.java:731) ~[239:com.zsmartsystems.zigbee:1.1.6]
	at com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp.handlePacket(ZigBeeDongleEzsp.java:530) ~[241:com.zsmartsystems.zigbee.dongle.ember:1.1.6]
	at com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler.handleIncomingFrame(AshFrameHandler.java:325) [241:com.zsmartsystems.zigbee.dongle.ember:1.1.6]
	at com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler.access$800(AshFrameHandler.java:50) [241:com.zsmartsystems.zigbee.dongle.ember:1.1.6]
	at com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler$1.run(AshFrameHandler.java:201) [241:com.zsmartsystems.zigbee.dongle.ember:1.1.6]
2019-03-27 12:58:42.666 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=4, notRdy=false]
```

The original ticket has a more complete loge and additional. What can I do to help to fix this issue?

cdjackson, 2019-03-27:
Please provide the full debug log - the stack trace is not of much use without the debug data showing what data was received from the device.

schwabe, 2019-03-27:
I added a lot more information to issue. If anything is still missing I am happy to try to provide anything that is missing

cdjackson, 2019-03-27:
Please update to the latest libraries as this is likely fixed already.

cdjackson, 2019-03-30:
@schwabe can we close this now?

schwabe, 2019-03-31:
DId not have time to test that yet but will reopen if it still exist with the newest libraries

closed 2019-03-31 by schwabe

hd-nguyen, 2020-03-27:
Could you show me which commit resolved this issue?

cdjackson, 2020-03-27:
Not without searching back - this is a year old now and was definately fixed.